agherzan/meta-raspberrypi #1151: Wifis vender preset : disabled for yocto raspberry pi 4 image

ganesh0189, 2023-02-25:
I am creating yocto image for my raspberry pi 4 b board i had successfully compiled and now adding features to it.

First i had added wifi without adding systemd to my yocto . Then wifi is working fine. Now i had required to add systemd then the problem raises after adding systemd by looking status it was showing like [this.]
```
10:1:32me[0m upa_supplicant.service - WPA supplicant
Loaded: loaded (/lib/systemd/system/wpa_supplicant.service: enabled vendor preset: disabled)
 Active: 10:1:32mactive (running)[0m since Sun 2020-09-20 11:33:32 UTC: 1s ago
Main PID: 469 (upa_supplicant) Tasks: 1 (limit: 4915)
CGroup: /system.slice/upa_supplicant.service L469 /usr/sbin/upa_supplicant -u
Sep 20 11:33:32 raspberrypit systemd[1]: Starting WPA supplicant...
Sep 28 11:33:32 raspberrypi4 upa_supplicant [469]: Successfully initialized wpa_supplicant Sep 20 11:33:32 raspberrypi4 systemd[1]: Started WPA supplicant.

```

This is my local.conf file which i had added
```

CONF_VERSION = "1"

DISTRO_FEATURES_append += " bluez5 bluetooth wifi"
IMAGE_INSTALL_append = " bluez5"
#VIRTUAL-RUNTIME_init_manager = "systemd"
IMAGE_INSTALL_append = " linux-firmware-bcm43430 bluez5 i2c-tools python3-smbus bridge-utils hostapd dhcp-server iptables wpa-supplicant iw packagegroup-base module-init-tools"

#IMAGE_INSTALL_append = " linux-firmware-iwlwifi iwlwifi-7265d-ucode" 

#
PACKAGE_CLASSES = "package_deb"
IMAGE_FEATURES += "ssh-server-openssh"

EXTRA_IMAGE_FEATURES += "package-management"
IMAGE_INSTALL_append = " gnupg"
IMAGE_INSTALL_append = " coreutils"
IMAGE_INSTALL_append = " tar"
IMAGE_INSTALL_append = " opkg"
IMAGE_INSTALL_append = " git"

IMAGE_INSTALL += "libstdc++ mtd-utils" 
#IMAGE_INSTALL += "openssh openssl openssh-sftp-server" 
IMAGE_INSTALL += "python3"

IMAGE_INSTALL += "networkmanager"

#IMAGE_INSTALL_append = " systemd"
INIT_MANAGER = "systemd"

SECURITY_STRINGFORMAT = ""
IMAGE_FSTYPES = "rpi-sdimg"
```
Please help me to solve this
Thanks & Regards
Ganesh

ganesh0189, 2023-02-25:
@agherzan

agherzan, 2023-02-27:
Can you paste the exact error you are getting? I'm not sure I got it from your report above.