agherzan/meta-raspberrypi #1155: 32 Bit binary on 64 bit machine

lachlancresswell, 2023-03-03:
I have a 32 bit binary I have compiled on a 32 bit machine. Installing and running the binary on a 64 bit Yocto build returns a 'No such file or directory' error which seems to be an architecture mismatch. What else needs to be done to allow a 32 bit binary to run on a 64 bit build?

agherzan, 2023-03-03:
You are probably facing https://stackoverflow.com/questions/2716702/no-such-file-or-directory-error-when-executing-a-binary. Doesn't sound like a BSP issue so I'll close this for now. I hope it helps.

closed 2023-03-03 by agherzan

kraj, 2023-03-03:
> I have a 32 bit binary I have compiled on a 32 bit machine. Installing and running the binary on a 64 bit Yocto build returns a 'No such file or directory' error which seems to be an architecture mismatch. What else needs to be done to allow a 32 bit binary to run on a 64 bit build?

you need to enable 32bit multilib in yocto and add approprite dependencies to your apps recipe