agherzan/meta-raspberrypi #1157: how to add x11-xserver-utils xautomation unclutter in yocto image

gitgonewithwind, 2023-03-06:
hi all,
I want to add epiphany  in yocto image ,  it can build and can run.
but I need kiosk-screen fullscreen.
according to here: https://github.com/elalemanyo/raspberry-pi-kiosk-screen
but xte not found 
and  it seems need to install x11-xserver-utils xautomation unclutter
Can I do it in yocto ?how to do ? there is no those receipts in yocto 
Thanks

kraj, 2023-03-06:
meta-oe provides `unclutter-xfixes` which can replace unclutter  functionality. Not sure what all is needed from `x11-xserver-utils` but OE provided multiple recipes to provide content of `x11-xserver-utils` which is rgb,xhost,xmodmap,xrandr,xrdb and xset. For `xautomation` there is a replacement called `xdotool`

so try your luck, your milege may vary :)

gitgonewithwind, 2023-03-06:
> meta-oe provides `unclutter-xfixes` which can replace unclutter functionality. Not sure what all is needed from `x11-xserver-utils` but OE provided multiple recipes to provide content of `x11-xserver-utils` which is rgb,xhost,xmodmap,xrandr,xrdb and xset. For `xautomation` there is a replacement called `xdotool`
> 
> so try your luck, your mileage may vary :)

--
@kraj 
thank you so much for reply so soon

I only has a issue:
 xte "key F11" -x:0
xte                                         
-sh: xte: command not found 
do you have idea for  install xte ?

sorry to post here. because for our heavy application  chromium is very slow to show image. so we have to change a light browser to try
Thank you so much again.

kraj, 2023-03-06:
xte comes form `xautomation` so you might want to explore `xdotools` replacing xte, it must have equivalent commands for replacing xte
xte "key F11" -x:0

perhaps can become

"xdotool key F11"
   release - x:0

gitgonewithwind, 2023-03-06:
@kraj,
Thank you so much.
let me try

gitgonewithwind, 2023-03-07:
@kraj 
sorry to bother you again

I use xdotool to similate F11 , it works, but there is 5 seconds time gap to become fullscreen
I see here: 
https://raspberrypi.stackexchange.com/questions/30093/epiphany-browser-in-full-screen-mode
it use  matchbox-window-manager 

but yocto no matchbox-window-manager
only  find matchbox ? is it?
thank you.

kraj, 2023-03-07:
Add `IMAGE_INSTALL += "packagegroup-core-x11-base"` to your image

gitgonewithwind, 2023-03-07:
@kraj  Thank you so much. let me try