agherzan/meta-raspberrypi #1161: [Kirkstone] Unable to set root password for CM4

RajasJoshi, 2023-03-23:
Hi All,

I am trying to build a custom image based on kirkstone branch for my raspberry pi CM4. I am trying to set password for root user by adding the following to the image recipe
```
inherit extrausers

EXTRA_USERS_PARAMS += "\
        usermod -p 'root' root; \
        "
```

And have also removed debug-tweaks from local.conf. When I build image using this configuration and try to enter the mentioned password, I get an error that the password is wrong.
I have used the same method for setting root password for CM3 using the pyro branch and it works. In pyro build I used the following 
```
inherit extrausers

EXTRA_USERS_PARAMS += "\
        usermod -P 'root' root; \
        "
```

Please let me know if I am doing somthing wrong or if there is any alternate method to do this. Any help is greatly appreciated!!

kraj, 2023-03-23:
see this 
https://github.com/YoeDistro/yoe-distro/blob/master/sources/meta-yoe/recipes-core/images/yoe-simple-image.bb

it might help

RajasJoshi, 2023-03-25:
@kraj  Thank you so much!!! That solved the problem!

closed 2023-03-25 by RajasJoshi