agherzan/meta-raspberrypi #1165: Create an image for boot from USB stick

JMStranz, 2023-03-31:
I have successfully built the image "rpi-test-image" and booted my "Raspbery Pi 4" from the SD card.

However, you cannot use this image to boot from a USB stick.
In the file "cmdline.txt" (partition "boot") the SD card is fixed for the parameter "root":

root=/dev/mmcblk0p2

How can I create an image to boot from a USB stick?

Jan.

agherzan, 2023-03-31:
You should play with `CMDLINE_ROOTFS`. With that you would be able to set anything for that part of the arguments.

agherzan, 2023-03-31:
Are you using uboot?

JMStranz, 2023-03-31:
Regarding U-Boot and Linux Kernel the following variables are set in my machine configuration:

RPI_USE_U_BOOT = "1"

UBOOT_MACHINE = "rpi_4_config"

KERNEL_BOOTCMD ?= "booti"

RPI_KERNEL_DEVICETREE = " \
    broadcom/bcm2711-rpi-4-b.dtb \
"

VC4DTBO ?= "vc4-fkms-v3d"

ARMSTUB ?= "armstub8-gic.bin"

SDIMG_KERNELIMAGE ?= "kernel8.img"

KERNEL_IMAGETYPE_UBOOT ?= "Image"

KERNEL_IMAGETYPE_DIRECT ?= "Image"

I've attached my machine configuration "rpi4-64.conf" as "rpi4-64.txt"
[rpi4-64.txt](https://github.com/agherzan/meta-raspberrypi/files/11122069/rpi4-64.txt)

By the way: for "meta-taspberrypi" there are 2 sources:

https://github.com/agherzan/meta-raspberrypi.git
https://git.yoctoproject.org/meta-raspberrypi

Which GIT repository should I use?

EasyIP2023, 2023-03-31:
@JMStranz Can you see the usb drive in u-boot console? May find this link useful
* https://wiki.analog.com/resources/tools-software/linuxdsp/docs/das-u-boot/loading_file_from_usb_storage_in_uboot

Although not u-boot this is how I boot a live usb image for x64 AMD machine with grub.  You may need to create an initrd put that on the usb drive then instruct u-boot to boot off usb drive's initrd image + device tree.

```
menuentry 'Initrd Image' {
    linux /bzImage LABEL=root root=/dev/ram0 console=tty1 console=ttyS0,115200n8 rootfstype=ext4
    initrd /initrd
}
```

Yocto wise to build initrd just need to set these two variables
```
INITRAMFS_IMAGE ?= "<image recipe name>"
INITRAMFS_IMAGE_BUNDLE ?= "0"
```

JMStranz, 2023-04-03:
Since I am still using the older branch "dunfell", I need to set the variable "CMDLINE" in my "local.conf" file.
E.g. for booting from a USB stick:

`CMDLINE = "dwc_otg.lpm_enable=0 ${SERIAL} root=/dev/sda2 rootfstype=ext4 rootwait"`

The possibility to set the variable "CMDLINE_ROOTFS" is only available in one of the newer branches of "meta-raspberrypi".

agherzan, 2023-04-03:
@JMStranz The yocto project one is a mirror of the Gihub project. So this means that it doesn't matter which one you are using.

So the answer is yes, you are using uboot. Check the scr file: https://github.com/agherzan/meta-raspberrypi/blob/acb2f73e58c5a2e105e416c5e8a609c493193005/recipes-bsp/rpi-u-boot-scr/files/boot.cmd.in