agherzan/meta-raspberrypi #1168: Lacks support for Hifiberry Dac+ADC Pro

valentinlageard, 2023-04-18:
**Description**

Hi, I'm trying to build an image containing hifiberry-dacplusadcpro.dtbo.
I've seen that it is not included in RPI_KERNEL_DEVICE_TREE_OVERLAYS `conf/machine/include/rpi-base.inc`. 
I'm in the process of learning yocto and I'm not entirely sure how to append the overlay I need to the previously mentioned variable. I've tried to find information on how to change variables in .inc files to no avail.
I considered cloning and patching your layer but if I understand correctly, it is considered bad practice to clone and modify layers you depend on.
Could you point me to the right direction in order to have my custom layer adding support for the DAC + ADC Pro by appending to your layer ?
Or even, better, add support directly in your layer ?

Thanks and cheers,
Val.

valentinlageard, 2023-04-18:
Hi, I tried simply manually adding `overlays/hifiberry-dacplusadcpro.dtbo` to the `rpi-base.inc` file and the device tree overlay is added to the image and works fine.

Still, I don't understand how if I should do a pull request or find a way to override this in my own layer. If option 2 is better, could you point me in the right direction ?

Val.

JDavidsson, 2026-09-03:
Hi, 

It was added in https://github.com/agherzan/meta-raspberrypi/commit/fc34bc3b86f9eb42296618a3ceef82f276c472f3 (https://github.com/agherzan/meta-raspberrypi/pull/1275) and is present in kernels 6.1, 6.6, 6.12 and 6.18.

closed 2026-10-18 by agherzan