agherzan/meta-raspberrypi #1189: RPi4 does not boot when enabling RPI_USE_U_BOOT

5o5uk3, 2023-05-31:
**Description**

When enabling `RPI_USE_U_BOOT` in `conf/local.conf`, the raspberry pi 4 does not boot *(`langdale` branch)*.

**Steps to reproduce the issue:**
1. Clone the yocto project
2. Add the meta-raspberrypi layer
3. Add `RPI_USE_U_BOOT = "1"` in `conf/local.conf` file
4. build rpi-test-image and copy it to a SD card
5. boot

**Current results:**

* Nothing on HDMI output / only red led is on - green led remains off
* Nothing on UART but that's normal as it is disabled

**Expected results:**

* A normal boot sequence

**Howto fix:**

* A simple workaround is to add `enable_uart=1` in `config.txt` file. The reason is explained in `arch/arm/mach-bcm283x/Kconfig` from u-boot source code:

```
 config TARGET_RPI_4
    bool "Raspberry Pi 4 64-bit build"
    help
      Support for all BCM2711-based Raspberry Pi variants, such as
      the RPi 4 model B, in AArch64 (64-bit) mode.

      This option assumes the VideoCore firmware is configured to use the
      mini UART (rather than PL011) for the serial console. This is the
      default on the RPi 4. To enable the UART console, the following non-
      default option must be present in config.txt: enable_uart=1. This is
      required for U-Boot to operate correctly, even if you only care
      about the HDMI/usbkbd console.
```

Note that this requirement is also valid for other raspberry pi flavors.
IMHO, `recipes-bsp/bootfiles/rpi-config_git.bb` should add a conditional statement like:

```bash
    elif [ "${RPI_USE_U_BOOT}" = "1" ] ; then
        echo "# Enable UART" >>$CONFIG
        echo "enable_uart=1" >>$CONFIG
```

Indeed, it would be a bad idea to set `ENABLE_UART = "1"` when `RPI_USE_U_BOOT` is enabled as it would also add the serial console parameter on the kernel command line which we may not want.

agherzan, 2023-05-31:
That is an excellent catch. I was not aware of this. Your change is reasonable for me. You mention that this is required for all boards. Why is that, exactly? The snippet above only mentions RPi4-64.

5o5uk3, 2023-05-31:
I did only mention RPi4-64 because I did not test on other boards, so I cannot assert the problem exist for other boards. Nonetheless, based on this file: https://source.denx.de/u-boot/u-boot/-/blob/v2023.01/arch/arm/mach-bcm283x/Kconfig?ref_type=tags#L65 I am pretty sure that the following boards also have the `enable_uart` requirement:
* Raspberry Pi Zero W
* Raspberry Pi 3 32-bit
* Raspberry Pi 3 64-bit
* Raspberry Pi 4 32-bit
* Raspberry Pi 4 64-bit
Note that "Raspberry Pi 2" also has its specific requirement...
Considering these facts, I would say that my first proposal was a bit naive and that the `enable_uart` requirement is also dependent of `UBOOT_MACHINE` value.

agherzan, 2023-05-31:
Indeed. Based on that, it looks like pretty much all of them have some requirements on the UART configuration (including the earlier versions). Happy to consider an PR if you are looking into it.

5o5uk3, 2023-05-31:
I made a PR on master branch. If it is accepted, it would be great to if you can cherry-pick it to langdale and mickledore branches.

agherzan, 2023-06-05:
@5o5uk3 feel free to create MRs against those branches once gets into master.