agherzan/meta-raspberrypi #1192: Raspberry pi 4, 7 inc touchscreen not working, no display

karaketir16, 2023-06-12:
**Description**

Raspberry pi 4, 7 inc touchscreen not working. Shows rainbow on boot, and then shows only white screen.

changing `dtoverlay=vc4-kms-v3d` to `dtoverlay=vc4-fkms-v3d` in sdcard/boot fixes my problem.

**Additional information you deem important (e.g. issue happens only occasionally):**

`VC4DTBO ?= "vc4-kms-v3d"` in *rpi-config_git.bb* can be made conditional for raspberrypi4.

**Additional details (revisions used, host distro, etc.):**

mickledore, poky, core-image-sato

abehonest, 2023-11-18:
I had the exact same problem and can confirm that this fix works for Freenove touchscreens.

agherzan, 2023-11-23:
Have we reported this upstream? Do we have any context on why this happens? KMS should be working on RPI4.

karaketir16, 2023-11-24:
https://forums.raspberrypi.com/viewtopic.php?p=1507247#p1507337

```
fkms is adopted as standard on the Pi4.
kms doesn't exist on the Pi4, and requires configuration in significantly different ways.
```

This is a reply from: [6by9](https://forums.raspberrypi.com/memberlist.php?mode=viewprofile&u=92334)

As forums say: he/she is Raspberry Pi Engineer & Forum Moderator

kraj, 2023-11-24:
> https://forums.raspberrypi.com/viewtopic.php?p=1507247#p1507337
>
> ```
> fkms is adopted as standard on the Pi4.
> kms doesn't exist on the Pi4, and requires configuration in significantly different ways. [...]

that was  before vc4 driver was complete. What dtb is used in raspbian today ? Can you check and report back ?

matthiasklein, 2023-11-25:
I used the following in kirkstone a few days ago:

```
RPI_EXTRA_CONFIG = '\n\
# Raspberry Pi 7\" display/touch screen\n\
dtoverlay=vc4-kms-dsi-7inch\n\
'
```

mainmachine, 2023-11-30:
There are discussions of the issue OP is facing [here](https://lists.yoctoproject.org/g/yocto/topic/meta_raspberrypi/99727959?p=,,,20,0,0,0::recentpostdate/sticky,,,20,0,0,99727959,previd%3D1687557894698372585,nextid%3D1687429270883266584&previd=1687557894698372585&nextid=1687429270883266584), however:

> The "legacy firmware" and "FKMS" display modes used in earlier versions of Raspberry Pi OS are no longer supported; KMS (Kernel Mode Setting) is now used instead.
[Reference](https://www.raspberrypi.com/documentation/computers/configuration.html#command-line-options)

As of the Bookworm rebase of RPIOS (if not just before that...?) the preferred choice is `vc4-kms-v3d` - that's the open source driver, which is working great on newer kernels (6.1.0-rpi6-rpi-v8 is current on RPIOS). The default should now be `vc4-kms-v3d`, and in cases like what @karaketir16 is doing, you can change the overlays to suit your specific hardware:

```
dtoverlay=vc4-kms-dsi-7inch
dtoverlay=rpi-ft5406
```

There are discussions of the issue OP is facing [here](https://lists.yoctoproject.org/g/yocto/topic/meta_raspberrypi/99727959?p=,,,20,0,0,0::recentpostdate/sticky,,,20,0,0,99727959,previd%3D1687557894698372585,nextid%3D1687429270883266584&previd=1687557894698372585&nextid=1687429270883266584) and also:

https://forums.raspberrypi.com/viewtopic.php?t=312042
https://forums.raspberrypi.com/viewtopic.php?t=346690

karaketir16, 2023-12-01:
I cannot try it now, because I no longer have a Pi4 to experiment with. I had tried it with the 'kirkstone' branch and the 5.15.34-v7 kernel version.

closed 2023-12-13 by agherzan