agherzan/meta-raspberrypi #1196: ERROR: Required build target 'core-image-sato' has no buildable providers. Missing or unbuildable dependency chain was: ['core-image-sato', 'packagegroup-base-extended', 'linux-firmware-rpidistro-bcm43456']

santomy1999, 2023-06-29:
**Description**

**Steps to reproduce the issue:**
1.git clone https://github.com/openembedded/meta-openembedded.git
2.git clone https://github.com/agherzan/meta-raspberrypi
3.bitbake-layers add-layer ../meta-openembedded/meta-networking
4.bitbake-layers add-layer ../meta-openembedded/meta-python
5.bitbake-layers add-layer ../meta-openembedded/meta-multimedia/
6.Changed the machine to raspberrypi4-64
7. bitbake core-image-sato

**Describe the results you received:**

**Describe the results you expected:**

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

aurelian17, 2023-06-29:
If you check the Quicstart you will understand why core-image-sato doesn't work:

Quick Start
source poky/oe-init-build-env rpi-build
Add this layer to bblayers.conf and the dependencies above
Set MACHINE in local.conf to one of the supported boards
bitbake core-image-base
Use bmaptool to copy the generated .wic.bz2 file to the SD card
Boot your RPI

Can you please try to use the core-image-base image instead?

agherzan, 2023-07-03:
Sadly sato is not an image we test but it looks like you don't have this layer added. Have you added the rpi layer?

agherzan, 2024-03-18:
@khimaros I'm unsure how you can have the same core-image-sato error while building core-image-base. I would suggest you create a new issue and clearly describe the issue with reproducible steps.

I will close this. The initial issue must have been due to the RPI layer not being added. The author didn't confirm.