agherzan/meta-raspberrypi #1206: libcamera-apps doesn't build with tensorflow-lite

vivien, 2023-07-27:
With the meta-tensorflow layer added to bblayers, enabling TFLITE in the libcamera-apps build will result in: 

```
fatal error: tensorflow/lite/builtin_op_data.h: No such file or directory
|    16 | #include "tensorflow/lite/builtin_op_data.h"
|       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| compilation terminated.
```

This happens regardless of having tensorflow-lite listed as a build dependency or not, as fixed by [2cdb97a](https://github.com/agherzan/meta-raspberrypi/pull/1205/commits/2cdb97a3738748b46f042fe5fb9febbfdb715a55).

vivien, 2023-08-03:
As of [76263ba](https://github.com/agherzan/meta-raspberrypi/pull/1205/commits/76263bae950bd37b6d16de0e7f8561674c6ceb96), libcamera-apps builds for raspberrypi3 and raspberrypi4, but tensorflow-lite fails for raspberrypi with:

```
Linking tensorflow/lite/libtensorflowlite.so failed: (Exit 1): arm-poky-linux-gnueabi-gcc failed: error executing command
```

kraj, 2023-08-10:
`Run-time dependency tensorflow-lite found: NO (tried pkgconfig and cmake)`

there will be some .txt files for errors generated by cmake in builddir. So please share those.

vivien, 2023-08-10:
The package is now using meson instead of cmake. Here's the related snippet from `/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libcamera-apps/1.2.1+gitAUTOINC+1c1d1c1a2a-r0/build/meson-logs/meson-log.txt `:

```
Determining dependency 'tensorflow-lite' with pkg-config executable '/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libcamera-apps/1.2.1+gitAUTOINC+1c1d1c1a2a-r0/recipe-sysroot-native/usr/bin/pkg-config'
env[PKG_CONFIG_DISABLE_UNINSTALLED]: yes
env[PKG_CONFIG_SYSTEM_LIBRARY_PATH]: /lib:/usr/lib
env[PKG_CONFIG_PATH]: /build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libcamera-apps/1.2.1+gitAUTOINC+1c1d1c1a2a-r0/recipe-sysroot/usr/lib/pkgconfig:/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libcamera-apps/1.2.1+gitAUTOINC+1c1d1c1a2a-r0/recipe-sysroot/usr/share/pkgconfig
env[PKG_CONFIG_DIR]: /build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libcamera-apps/1.2.1+gitAUTOINC+1c1d1c1a2a-r0/recipe-sysroot/usr/lib/pkgconfig
env[PKG_CONFIG_SYSROOT_DIR]: /build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libcamera-apps/1.2.1+gitAUTOINC+1c1d1c1a2a-r0/recipe-sysroot
env[PKG_CONFIG_LIBDIR]: /build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libcamera-apps/1.2.1+gitAUTOINC+1c1d1c1a2a-r0/recipe-sysroot/usr/lib/pkgconfig
env[PKG_CONFIG_SYSTEM_INCLUDE_PATH]: /usr/include
Called `/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libcamera-apps/1.2.1+gitAUTOINC+1c1d1c1a2a-r0/recipe-sysroot-native/usr/bin/pkg-config --modversion tensorflow-lite` -> 1
stderr:
Package tensorflow-lite was not found in the pkg-config search path.
Perhaps you should add the directory containing `tensorflow-lite.pc'
to the PKG_CONFIG_PATH environment variable
No package 'tensorflow-lite' found
```
Adding `inherit pkgconfig` to the tensorflow-lite recipe results in the same error.

kraj, 2023-08-10:
you might want to look at tensorflow-lite recipe and see why it is not generating and/or packaging the .pc file

vivien, 2023-08-10:
So tensorflow-lite from meta-tensorflow is not installing any tensorflow-lite.pc file, it isn't installed the header neither. While this can be fixed, it doesn't provide the missing flatbuffers files too. I switched to tensorflow-lite from meta-neural-network, which works with a few fixups. I now have libcamera-detect on raspberrypi4-64.

kraj, 2023-08-11:
interesting, I dont know which layer should be considered more official some indicators I use is activity and on that meta-tensorflow is on https://git.yoctoproject.org/meta-tensorflow so it should be the one people usually will default to and therefore should be fixed. I would suggest to inform the meta-tensorflow maintainers about this issue. meta-neural-network is for NNstreamer and they might maintain tensorflow-lite in a narrow sense of just NNstreamer which makes sense for what the layer is about.

agherzan, 2023-08-14:
So this doesn't look RPi-specific, does it?

vivien, 2023-08-14:
Indeed, these are issues in meta-tensorflow and compilation issue on 32-bit ARM. I'm closing this issue for now but I'll push a few tweaks for libcamera-detect integration soon though.

closed 2023-08-14 by vivien