agherzan/meta-raspberrypi #1211: Waveshare-DSI-LCD can not display for yocto image with linux-raspberrypi 6.1.34

gitgonewithwind, 2023-08-08:
Hi 
I post my issue here:

https://github.com/waveshare/Waveshare-DSI-LCD/issues/4#issuecomment-1662683896
in that page shows  if the kernel > 6.1.25  , it can support Waveshare-DSI-LCD,

 I used meta-raspberrypi with yocto
commit dff85b9a9f66002856b9ed3b1aa3a384c0bc43d9
linux-raspberrypi-6.1: Update to 6.1.34 release

but it seems the kernel can not communicate with i2c,
so do I need to modify the default dtb(dts )file to enable i2c?
how can I do ? I want to know if the issue is about i2c communicate  or something else?

Thank you very much.

gitgonewithwind, 2023-08-16:
hi all ,
do you have any idea?
I have updated kernel to 6.1.38 and built the drivers in kernel, but it still can not  work for waveshare-dsi-lcd in yocto image. I do no know how to do now
and I tried to set "okay" in  dsi1 in dtsi ,but it not useful.
I want to know if ycoto image not support it ? but I think if ther kernel support it in raspbian image, it should support it.
I do not know why not work
any idea ?  @agherzan 
Thanks

root@raspberrypi4-64:~# ls /dev/i2c*                                            
/dev/i2c-0  /dev/i2c-1  /dev/i2c-10  /dev/i2c-20  /dev/i2c-21  /dev/i2c-22      
root@raspberrypi4-64:~# modinfo panel-waveshare-dsi                             
name:           panel_waveshare_dsi                                             
filename:       (builtin)                                                       
license:        GPL                                                             
file:           drivers/gpu/drm/panel/panel-waveshare-dsi                       
description:    Waveshare DSI panel driver                                      
author:         Dave Stevenson <dave.stevenson@raspberrypi.com>                 
root@raspberrypi4-64:~# modinfo goodix_ts                                       
name:           goodix_ts                                                       
filename:       (builtin)                                                       
license:        GPL v2                                                          
file:           drivers/input/touchscreen/goodix_ts                             
description:    Goodix touchscreen driver                                       
author:         Bastien Nocera <hadess@hadess.net>                              
author:         Benjamin Tissoires <benjamin.tissoires@gmail.com>               
root@raspberrypi4-64:~# uname -a                                                
Linux raspberrypi4-64 6.1.38-v8 #1 SMP PREEMPT Mon Jul 10 18:26:47 UTC 2023 aarx
root@raspberrypi4-64:~# ls /dev/i2c*                                            
/dev/i2c-0  /dev/i2c-1  /dev/i2c-10  /dev/i2c-20  /dev/i2c-21  /dev/i2c-22 

config.txt 
                                                                   
dtparam=i2c1=on                                                                 
dtparam=i2c_arm=on                                                              
enable_uart=1                                                                   
dtoverlay=vc4-kms-v3d                                                           
dtparam=i2c_arm=on                                                         
dtoverlay=vc4-kms-v3d                                                      
dtoverlay=vc4-kms-dsi-waveshare-panel,8_0_inch                             
 dtparam=i2c_vc=on

agherzan, 2023-08-24:
I am not sure how to guide you from here. I think the right way would be to ask, as you already did, on waveshare.

omarhogni, 2023-08-24:
Here are the steps I used to get this working for me.

You do need to update the kernel version provided in meta-raspberrypi to a version that includes 
the driver [https://github.com/raspberrypi/linux/pull/5433](https://github.com/raspberrypi/linux/pull/5433).

I created a file called `linux-raspberrypi_6.1.bbappend`
```
LINUX_VERSION = "6.1.46"
SRCREV_machine = "e89e7655a197d28df49da2be7e2003436cf52197"
```

Then you need to add the driver dtbo file into `/boot` of the device for that you need to add
```
RPI_KERNEL_DEVICETREE_OVERLAYS += "\
  <... REPLACE THIS WITH ALL THE OVERLAYS YOU NEED ...>
  overlays/vc4-kms-dsi-waveshare-panel.dtbo \
"
```
For rest that you need I just used the [default](https://github.com/agherzan/meta-raspberrypi/blob/master/conf/machine/include/rpi-base.inc#L17) as a template and worked
from there. If you add this line to your local conf the built in one
does not apply anymore so beware.

You also need to enable i2c and configure the driver for you panel
```
KERNEL_MODULE_AUTOLOAD:rpi += " i2c-dev"

RPI_EXTRA_CONFIG = '\n \
# ----- CUSTOM OPTIONS SET IN local.conf ----------\n \
# Raspberry Pi 10_1 dsi \" display/touch screen \n \
dtparam=i2c_arm=on \n \
dtoverlay=vc4-kms-dsi-waveshare-panel,10_1_inch \n \
# ----- END CUSTOM OPTIONS SET IN local.conf ------ \n \
'
```

This was sufficient to get my 10" display working.
Hope this helps!

gitgonewithwind, 2023-08-25:
HI @omarhogni 
thank you so much.

I did not do this step before :

RPI_KERNEL_DEVICETREE_OVERLAYS += "\
  <... REPLACE THIS WITH ALL THE OVERLAYS YOU NEED ...>
  overlays/vc4-kms-dsi-waveshare-panel.dtbo \
"
in my understanding , I am going to add this  line "overlays/vc4-kms-dsi-waveshare-panel.dtbo"  to file  conf/machine/include/rpi-base.inc ,  can  not add this line in build/conf/local.conf , right?
let me  try

gitgonewithwind, 2023-08-25:
@omarhogni  
your solution works well 8" display
Thank you so much.

closed 2023-08-29 by agherzan

bhaggenmiller, 2024-04-15:
> Here are the steps I used to get this working for me.
> 
> You do need to update the kernel version provided in meta-raspberrypi to a version that includes the driver [raspberrypi/linux#5433](https://github.com/raspberrypi/linux/pull/5433).
> 
> I created a file called `linux-raspberrypi_6.1.bbappend`
> 
> ```
> LINUX_VERSION = "6.1.46"
> SRCREV_machine = "e89e7655a197d28df49da2be7e2003436cf52197"
> ```
> 
> Then you need to add the driver dtbo file into `/boot` of the device for that you need to add
> 
> ```
> RPI_KERNEL_DEVICETREE_OVERLAYS += "\
>   <... REPLACE THIS WITH ALL THE OVERLAYS YOU NEED ...>
>   overlays/vc4-kms-dsi-waveshare-panel.dtbo \
> "
> ```
> 
> For rest that you need I just used the [default](https://github.com/agherzan/meta-raspberrypi/blob/master/conf/machine/include/rpi-base.inc#L17) as a template and worked from there. If you add this line to your local conf the built in one does not apply anymore so beware.
> 
> You also need to enable i2c and configure the driver for you panel
> 
> ```
> KERNEL_MODULE_AUTOLOAD:rpi += " i2c-dev"
> 
> RPI_EXTRA_CONFIG = '\n \
> # ----- CUSTOM OPTIONS SET IN local.conf ----------\n \
> # Raspberry Pi 10_1 dsi \" display/touch screen \n \
> dtparam=i2c_arm=on \n \
> dtoverlay=vc4-kms-dsi-waveshare-panel,10_1_inch \n \
> # ----- END CUSTOM OPTIONS SET IN local.conf ------ \n \
> '
> ```
> 
> This was sufficient to get my 10" display working. Hope this helps!
Hey I am new to Yocto and the DTS stuff. Could you help me with explaining this in a little bit more detail. Especially where to edit what. Thanks

gitgonewithwind, 2024-04-15:
@bhaggenmiller 
1. at you build folder conf/local.conf  to add the following:

RPI_EXTRA_CONFIG = '\n \
# ----- CUSTOM OPTIONS SET IN local.conf ----------\n \
# Raspberry Pi 10_1 dsi \" display/touch screen \n \
dtparam=i2c_arm=on \n \
dtoverlay=vc4-kms-dsi-waveshare-panel,8_0_inch \n \
# ----- END CUSTOM OPTIONS SET IN local.conf ------ \n \
'

2.  depend on  what git commit your are using meta-raspberrypi  because the solution is months ago

I am using  Author: Vivien Didelot <vivien.didelot@gmail.com>
Date:   Thu Aug 3 01:36:48 2023 -0400

anyway similarly 
at folder meta-raspberrypi/recipes-kernel/linux  there was a latest linux version (at  a few month ago) .bb file  linux-raspberrypi_6.1.bb
so you 
I created a file called linux-raspberrypi_6.1.bbappend

LINUX_VERSION = "6.1.46"
SRCREV_machine = "e89e7655a197d28df49da2be7e2003436cf52197"

when building, it will download the commit  of source code.   (maybe the currently master of  meta-raspberrypi  updated version already support DSI LCD, so you do not need this step

3, in file   meta-raspberrypi/conf/machine/include/rpi-base.inc to add  overlays/vc4-kms-dsi-waveshare-panel.dtbo 
"
RPI_KERNEL_DEVICETREE_OVERLAYS += "\
  <... REPLACE THIS WITH ALL THE OVERLAYS YOU NEED ...>
  overlays/vc4-kms-dsi-waveshare-panel.dtbo \
"

and then, It will be :  

  overlays/vc4-kms-v3d-pi4.dtbo \
    overlays/vc4-kms-dsi-7inch.dtbo \
    overlays/w1-gpio.dtbo \
    overlays/w1-gpio-pullup.dtbo \
    overlays/wm8960-soundcard.dtbo \
    overlays/vc4-kms-dsi-waveshare-panel.dtbo \
    "

(same, maybe updated  master of  meta-raspberrypi  version  already added it, please have a look)

hope it  can help you

talksik, 2024-04-21:
I have tried to create a custom yocto image on top of boot2qt. I am running into the following problems:
1. blank screen.
2. Qt app won't run as it says "No screens found" or "No DRM device found".

I tried to troubleshoot, but I am out of ideas.

Context:
* I have a yocto custom boot2qt image that enables i2c via meta-raspberrypi settings.
* My config has `dtoverlay=vc4-kms-dsi-waveshare-panel,2_8_inch` as [waveshare's guide says.](https://www.waveshare.com/wiki/2.8inch_DSI_LCD)
* I've tried every different variant of `vc4-fkms-v3d` and `vc4-kms-v3d` and `vc4-kms-v3d-pi4`

Output of `lsmod | grep waveshare`
```
root@b2qt-raspberrypi4-64:~# lsmod | grep waveshare
panel_waveshare_dsi    16384  0
backlight              24576  1 panel_waveshare_dsi
```

Output of `ls /dev/dri`
```
root@b2qt-raspberrypi4-64:~# ls /dev/dri/
by-path     card0       card1       renderD128
```
My entire config.txt file:
```
# Enable SPI bus
dtparam=spi=on
# Enable I2C
dtparam=i2c1=on
dtparam=i2c_arm=on

# Enable VC4 Graphics
# dtoverlay=vc4-kms-v3d # was causing problems with DSI display
# https://bugreports.qt.io/browse/QTBUG-106112
# https://forum.qt.io/topic/152160/official-7-raspberry-touch-screen-lcd-isn-t-working-with-boot2qt-6-6-for-raspberry-pi-4-64-bit/2
dtoverlay=vc4-fkms-v3d
# dtoverlay=vc4-kms-v3d-pi4

# Enable USB peripheral mode
dtoverlay=dwc2,dr_mode=peripheral
# Enable the onboard ALSA audio
dtparam=audio=on
dtoverlay=vc4-kms-dsi-waveshare-panel,2_8_inch

# enable both hdmi outputs
# max_framebuffers=2 # disabled to prevent conflict with DSI display
```

Linux kernel version:
```
root@b2qt-raspberrypi4-64:~# uname -a
Linux b2qt-raspberrypi4-64 6.1.54-v8 #1 SMP PREEMPT Wed Sep 20 14:13:53 UTC 2023 aarch64 GNU/Linux
```

Confirmed that I have waveshare overlay:
```
root@b2qt-raspberrypi4-64:~# ls /boot/overlays/ | grep waveshare
vc4-kms-dsi-waveshare-panel.dtbo
waveshare-can-fd-hat-mode-a.dtbo
waveshare-can-fd-hat-mode-b.dtbo
```
Note: I also tried using a compiled dtbo file from a linux kernel 6.6 `/boot/overlays` into my yocto image with a kernel of 6.1.54. This workaround doesn't load the module when querying: `lsmod | grep waveshare`.

Here is the `dmesg` error that repeats endlessly:
```
...
[  469.247935] Goodix-TS 10-0014: Error writing 1 bytes to 0x814e: -110
[  470.271888] i2c-bcm2835 fe205000.i2c: i2c transfer timed out
[  470.271936] Goodix-TS 10-0014: Error reading 10 bytes from 0x814e: -110
[  471.295900] i2c-bcm2835 fe205000.i2c: i2c transfer timed out
[  471.295949] Goodix-TS 10-0014: Error writing 1 bytes to 0x814e: -110
[  472.319893] i2c-bcm2835 fe205000.i2c: i2c transfer timed out
[  472.319944] Goodix-TS 10-0014: Error reading 10 bytes from 0x814e: -110
...
```
Problem: my screen is blank. My Qt app says "No screens available" or "No DRM device" depending on which vc4-* overlay I load.

Ahh, I'm out of ideas and lost. If anyone can give me any sort of hints that helped them, I'd appreciate it!

Could this be a faulty device that needs to be replaced?

talksik, 2024-04-22:
Long couple days, but fixed. 

1. Made sure to use 6.1.46 as noted in this thread. Don't know if this is absolutely necessary.
2. Another layer `meta-boot2qt` that I was building with enabled raspberry pi camera. From my research, I found other people had this problem of the DSI, GPIO, and camera i2c causing conflicts.
3. I made sure to adjust the `KERNEL_MODULE_AUTOLOAD` as noted in this thread as the [`meta-raspberrypi` docs](https://meta-raspberrypi.readthedocs.io/en/latest/extra-build-config.html#enable-i2c) state a different method that is likely targeted for latest branches.

Thank you for the hints! I wonder if these problems never would have occurred if I updated the kernel to 6.6.x. I avoided this as I am a newbie when it comes to yocto and upgrading all layers seems daunting to me. Regardless, learned a lot of pieces from this debacle 😆

bhaggenmiller, 2024-04-29:
> @bhaggenmiller
> 
>     1. at you build folder conf/local.conf  to add the following:
> 
> 
> RPI_EXTRA_CONFIG = '\n \
> # ----- CUSTOM OPTIONS SET IN local.conf ----------\n \
> # Raspberry Pi 10_1 dsi " display/touch screen \n \
> 
> dtparam=i2c_arm=on \n dtoverlay=vc4-kms-dsi-waveshare-panel,8_0_inch \n \
> # ----- END CUSTOM OPTIONS SET IN local.conf ------ \n \
> 
> '
> 
>     2. depend on  what git commit your are using meta-raspberrypi  because the solution is months ago
> 
> 
> I am using Author: Vivien Didelot [vivien.didelot@gmail.com](mailto:vivien.didelot@gmail.com) Date: Thu Aug 3 01:36:48 2023 -0400
> 
> anyway similarly at folder meta-raspberrypi/recipes-kernel/linux there was a latest linux version (at a few month ago) .bb file linux-raspberrypi_6.1.bb so you I created a file called linux-raspberrypi_6.1.bbappend
> 
> LINUX_VERSION = "6.1.46" SRCREV_machine = "e89e7655a197d28df49da2be7e2003436cf52197"
> 
> when building, it will download the commit of source code. (maybe the currently master of meta-raspberrypi updated version already support DSI LCD, so you do not need this step
> 
> 3, in file meta-raspberrypi/conf/machine/include/rpi-base.inc to add overlays/vc4-kms-dsi-waveshare-panel.dtbo " RPI_KERNEL_DEVICETREE_OVERLAYS += " <... REPLACE THIS WITH ALL THE OVERLAYS YOU NEED ...> overlays/vc4-kms-dsi-waveshare-panel.dtbo "
> 
> and then, It will be :
> 
> overlays/vc4-kms-v3d-pi4.dtbo overlays/vc4-kms-dsi-7inch.dtbo overlays/w1-gpio.dtbo overlays/w1-gpio-pullup.dtbo overlays/wm8960-soundcard.dtbo overlays/vc4-kms-dsi-waveshare-panel.dtbo "
> 
> (same, maybe updated master of meta-raspberrypi version already added it, please have a look)
> 
> hope it can help you
> 
> if master of meta-raspberrypi version is already added it (I forget from what kernel version , it already support DSI), so you do not need step 2 to download "6.1.46" and commit "e89e7655a197d28df49da2be7e2003436cf52197".
> 
> you only need step 1 and step 3 ,I think

Thanks for your help. The display is no working. Unfortunately the application is not working properly. It s a html5/js page which runs on localhost in cog webengine.  It has some problems with the framebuffer:

swiotlb_tbl_map_single: 376 callbacks suppressed
vc4-drm gpu: swiotlb buffer is full (sz: 1441792 bytes), total 32768 (slots), used 466 (slots)

gitgonewithwind, 2024-04-29:
@bhaggenmiller 
Maybe your issue is not at DSI display, just at your APPs. I used chromium to display page. you can use a regular hdmi screen to make sure you app can run well, and then switch to DSI screen. just guess

bhaggenmiller, 2024-05-02:
> @bhaggenmiller Maybe your issue is not at DSI display, just at your APPs. I used chromium to display page. you can use a regular hdmi screen to make sure you app can run well, and then switch to DSI screen. just guess

Mhm on PI OS it works fine. Therefore i don t think the app is the problem.