agherzan/meta-raspberrypi #1212: Raspberry pi 0 2w kernel crashed during startup

ivrabie, 2023-08-14:
I tried to build core-image-base for raspberry pi 0 2w.
After build when the image starts I get kernel panic like in the startup_log.txt

**Steps to reproduce the issue:**
0. Kirkstone revision
1. Use local.conf attached in this issue
2. bitbake core-image-base
3.Flash sd-image on the sdcard

**Describe the results you received:**
Expect the image to start and I should be able to login.

**Describe the results you expected:**
Kernel crashes during startup

[startu_log.txt](https://github.com/agherzan/meta-raspberrypi/files/12333322/startu_log.txt)
[local.conf.txt](https://github.com/agherzan/meta-raspberrypi/files/12333327/local.conf.txt)

agherzan, 2023-08-14:
@ivrabie is this happening on master too?

ivrabie, 2023-08-19:
It works fine also with WiFi I will close this.

closed 2023-08-19 by ivrabie