agherzan/meta-raspberrypi #1216: can't fetch RPI_5.15 Branch

AYAHASSAN287, 2023-08-27:
I'm trying to make image for raspberry pi from branch "Langdale " 
but i have this issue . 

i want to switch to version 5.10 but i don't know which to files shall i change in . 

can any one help please ?

[image]

AYAHASSAN287, 2023-08-28:
update : 

i tried fetching 5.10 and i got the same issue 
i changed the content of 5.15.bb file to be like that 
LINUX_VERSION ?= "5.10.110"
LINUX_RPI_BRANCH ?= "rpi-5.10.y"
LINUX_RPI_KMETA_BRANCH ?= "yocto-5.10"

SRCREV_machine = "427c6bd8835e197693e9b4aedbe45c2c3c84cdce"
SRCREV_meta = "3b1dc2f1fcd869f97901402759b859035984aa7f"

but still same issue 

[image]

AYAHASSAN287, 2023-08-28:
I tried many branches like dunfell // micklorde but got the same issue .

cloning the issue outside yocto got me this 

 git clone https://github.com/raspberrypi/linux.git
Cloning into 'linux'...
remote: Enumerating objects: 10939748, done.
remote: Counting objects: 100% (19413/19413), done.
remote: Compressing objects: 100% (5378/5378), done.
error: 2874 bytes of body are still expected1.70 GiB | 182.00 KiB/s   
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output

kraj, 2023-08-28:
this tells that the issue might be on how your build host is set to access internet. Maybe some firewalls are kicking in.

AYAHASSAN287, 2023-08-29:
i managed to fetch the repo outside yocto using some git workaround . 

now i copied the folder into the downloads folder of yocto but not sure about what exactly the name should be 

can any one guide on what is the naming convention for he folder or where to find it ?
i tried
 linux-raspberrypi-1_6.1.20.tar.gz 
linux-raspberrypi-1_6.1.tar.gz
linux-raspberrypi-1_6.1.20

but still want to fetch

agherzan, 2023-08-29:
What was the git workaround?

Check the downloads directory for examples of fetched packages.

AYAHASSAN287, 2023-08-29:
i will leave the solution here for everyone 
1-increasing buffer 
git config --global http.postBuffer 52428800

2-Turn off the compression
git config --global core.compression 0

closed 2023-08-29 by agherzan