agherzan/meta-raspberrypi #1219: Facing build issue for raspberry4-64 board on mickledore branch

Amg-AmolGore, 2023-08-29:
Hi,

I facing build issue for machine raspberrypi4-64 on mickledore branch. Here is the log. Any solution will be appreciated. 

 /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/gcc/12.3.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: ifcvt.o: in function `noce_emit_cmove(noce_if_info*, rtx_def*, rtx_code, rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*)':
| /usr/src/debug/gcc/12.3.0-r0/gcc/recog.h:273: undefined reference to `recog(rtx_def*, rtx_insn*, int*)'
| /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/gcc/12.3.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: ifcvt.o: in function `end_ifcvt_sequence(noce_if_info*)':
| /usr/src/debug/gcc/12.3.0-r0/gcc/recog.h:273: undefined reference to `recog(rtx_def*, rtx_insn*, int*)'
| /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/gcc/12.3.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: ifcvt.o: in function `noce_emit_store_flag(noce_if_info*, rtx_def*, int, int)':
| /usr/src/debug/gcc/12.3.0-r0/gcc/recog.h:273: undefined reference to `recog(rtx_def*, rtx_insn*, int*)'
| /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/gcc/12.3.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: ifcvt.o: in function `noce_emit_move_insn(rtx_def*, rtx_def*)':
| /usr/src/debug/gcc/12.3.0-r0/gcc/recog.h:273: undefined reference to `recog(rtx_def*, rtx_insn*, int*)'
| /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/gcc/12.3.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: ifcvt.o: in function `noce_emit_bb(rtx_def*, basic_block_def*, bool)':
| /usr/src/debug/gcc/12.3.0-r0/gcc/recog.h:273: undefined reference to `recog(rtx_def*, rtx_insn*, int*)'
| /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/gcc/12.3.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: ifcvt.o:/usr/src/debug/gcc/12.3.0-r0/gcc/recog.h:273: more undefined references to `recog(rtx_def*, rtx_insn*, int*)' follow
| /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/gcc/12.3.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: rtl-ssa/changes.o: in function `recog_level2(rtl_ssa::insn_change&, std::function<bool (rtl_ssa::insn_change&, unsigned int)>)':
| /usr/src/debug/gcc/12.3.0-r0/gcc/rtl-ssa/changes.cc:840: undefined reference to `add_clobbers(rtx_def*, int)'
| /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/gcc/12.3.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: /usr/src/debug/gcc/12.3.0-r0/gcc/rtl-ssa/changes.cc:803: undefined reference to `recog(rtx_def*, rtx_insn*, int*)'
| /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/gcc/12.3.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: rtl-ssa/changes.o: in function `recog_level2(rtl_ssa::insn_change&, std::function<bool (rtl_ssa::insn_change&, unsigned int)>)':
| /usr/src/debug/gcc/12.3.0-r0/gcc/rtl-ssa/changes.cc:840: undefined reference to `add_clobbers(rtx_def*, int)'
| /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/gcc/12.3.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: /usr/src/debug/gcc/12.3.0-r0/gcc/rtl-ssa/changes.cc:803: undefined reference to `recog(rtx_def*, rtx_insn*, int*)'
| /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/gcc/12.3.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: rtl-ssa/changes.o: in function `recog_level2(rtl_ssa::insn_change&, std::function<bool (rtl_ssa::insn_change&, unsigned int)>)':
| /usr/src/debug/gcc/12.3.0-r0/gcc/rtl-ssa/changes.cc:840: undefined reference to `add_clobbers(rtx_def*, int)'
| /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/gcc/12.3.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: /usr/src/debug/gcc/12.3.0-r0/gcc/rtl-ssa/changes.cc:803: undefined reference to `recog(rtx_def*, rtx_insn*, int*)'
| /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/gcc/12.3.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: rtl-ssa/changes.o: in function `recog_level2(rtl_ssa::insn_change&, std::function<bool (rtl_ssa::insn_change&, unsigned int)>)':
| /usr/src/debug/gcc/12.3.0-r0/gcc/rtl-ssa/changes.cc:840: undefined reference to `add_clobbers(rtx_def*, int)'
| /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/gcc/12.3.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: /usr/src/debug/gcc/12.3.0-r0/gcc/rtl-ssa/changes.cc:803: undefined reference to `recog(rtx_def*, rtx_insn*, int*)'
| collect2: error: ld returned 1 exit status
| make[1]: *** [../../../../../../../work-shared/gcc-12.3.0-r0/gcc-12.3.0/gcc/lto/Make-lang.in:102: lto-dump] Error 1
| make[1]: *** Waiting for unfinished jobs....
| collect2: error: ld returned 1 exit status
| make[1]: *** [../../../../../../../work-shared/gcc-12.3.0-r0/gcc-12.3.0/gcc/c/Make-lang.in:87: cc1] Error 1
| collect2: error: ld returned 1 exit status
| make[1]: *** [../../../../../../../work-shared/gcc-12.3.0-r0/gcc-12.3.0/gcc/lto/Make-lang.in:96: lto1] Error 1
| collect2: error: ld returned 1 exit status
| make[1]: *** [../../../../../../../work-shared/gcc-12.3.0-r0/gcc-12.3.0/gcc/cp/Make-lang.in:145: cc1plus] Error 1
| make[1]: Leaving directory '/home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/gcc/12.3.0-r0/gcc-12.3.0/build.aarch64-poky-linux.aarch64-poky-linux/gcc'
| make: *** [Makefile:4619: all-gcc] Error 2
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/amgt/pi3/poky/meta/recipes-devtools/gcc/gcc_12.3.bb:do_compile) failed with exit code '1'
ERROR: ghostscript-10.0.0-r0 do_compile: oe_runmake failed
ERROR: ghostscript-10.0.0-r0 do_compile: ExecutionError('/home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/ghostscript/10.0.0-r0/temp/run.do_compile.116315', 1, None, None)
ERROR: Logfile of failure stored in: /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/ghostscript/10.0.0-r0/temp/log.do_compile.116315

exec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: ./obj/tesseract_lstm_series.o:(.data.rel.ro._ZTVN9tesseract6SeriesE[_ZTVN9tesseract6SeriesE]+0x48): undefined reference to `tesseract::Plumbing::SetNetworkFlags(unsigned int)'
| /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/ghostscript/10.0.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: ./obj/tesseract_lstm_series.o:(.data.rel.ro._ZTVN9tesseract6SeriesE[_ZTVN9tesseract6SeriesE]+0x60): undefined reference to `tesseract::Plumbing::ConvertToInt()'
| /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/ghostscript/10.0.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: ./obj/tesseract_lstm_series.o:(.data.rel.ro._ZTVN9tesseract6SeriesE[_ZTVN9tesseract6SeriesE]+0x68): undefined reference to `tesseract::Plumbing::SetRandomizer(tesseract::TRand*)'
| /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/ghostscript/10.0.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: ./obj/tesseract_lstm_series.o:(.data.rel.ro._ZTVN9tesseract6SeriesE[_ZTVN9tesseract6SeriesE]+0x88): undefined reference to `tesseract::Plumbing::DebugWeights()'
| /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/ghostscript/10.0.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: ./obj/tesseract_lstm_series.o:(.data.rel.ro._ZTVN9tesseract6SeriesE[_ZTVN9tesseract6SeriesE]+0x90): undefined reference to `tesseract::Plumbing::Serialize(tesseract::TFile*) const'
[... cut ...]
| /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/ghostscript/10.0.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: ./obj/tesseract_lstm_series.o:(.data.rel.ro._ZTVN9tesseract6SeriesE[_ZTVN9tesseract6SeriesE]+0xc0): undefined reference to `tesseract::Plumbing::AddToStack(tesseract::Network*)'
| collect2: error: ld returned 1 exit status
| make: *** [base/unixlink.mak:172: bin/gs] Error 1
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/amgt/pi3/poky/meta/recipes-extended/ghostscript/ghostscript_10.0.0.bb:do_compile) failed with exit code '1

Amg-AmolGore, 2023-09-01:
> Hi,
> 
> I facing build issue for machine raspberrypi4-64 on mickledore branch. Here is the log. Any solution will be appreciated.
> 
> Build Configuration: BB_VERSION = "2.4.0" BUILD_SYS = "x86_64-linux" NATIVELSBSTRING = "universal" TARGET_SYS = "aarch64-poky-linux" MACHINE = "raspberrypi4-64" DISTRO = "poky" DISTRO_VERSION = "4.2.3" TUNE_FEATURES = "aarch64 armv8a crc cortexa72" TARGET_FPU = "" meta meta-poky meta-yocto-bsp = "mickledore:aa63b25cbe25d89ab07ca11ee72c17cab68df8de" meta-raspberrypi = "mickledore:aa0aed9a08d6578a18c4eeb3b44ed8354a57ebee" meta-oe meta-python meta-networking meta-multimedia = "mickledore:b0ba472191c2b7709ef04ae9c5a23bea040518f9"
> 
> Issue Log:
> 
> ```
>  /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/gcc/12.3.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: ifcvt.o: in function `noce_emit_cmove(noce_if_info*, rtx_def*, rtx_code, rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*)':
> | /usr/src/debug/gcc/12.3.0-r0/gcc/recog.h:273: undefined reference to `recog(rtx_def*, rtx_insn*, int*)'
> | /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/gcc/12.3.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: ifcvt.o: in function `end_ifcvt_sequence(noce_if_info*)':
> | /usr/src/debug/gcc/12.3.0-r0/gcc/recog.h:273: undefined reference to `recog(rtx_def*, rtx_insn*, int*)'
> | /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/gcc/12.3.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: ifcvt.o: in function `noce_emit_store_flag(noce_if_info*, rtx_def*, int, int)':
> | /usr/src/debug/gcc/12.3.0-r0/gcc/recog.h:273: undefined reference to `recog(rtx_def*, rtx_insn*, int*)'
> | /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/gcc/12.3.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: ifcvt.o: in function `noce_emit_move_insn(rtx_def*, rtx_def*)':
> | /usr/src/debug/gcc/12.3.0-r0/gcc/recog.h:273: undefined reference to `recog(rtx_def*, rtx_insn*, int*)'
> | /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/gcc/12.3.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: ifcvt.o: in function `noce_emit_bb(rtx_def*, basic_block_def*, bool)':
> | /usr/src/debug/gcc/12.3.0-r0/gcc/recog.h:273: undefined reference to `recog(rtx_def*, rtx_insn*, int*)'
> | /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/gcc/12.3.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: ifcvt.o:/usr/src/debug/gcc/12.3.0-r0/gcc/recog.h:273: more undefined references to `recog(rtx_def*, rtx_insn*, int*)' follow
> | /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/gcc/12.3.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: rtl-ssa/changes.o: in function `recog_level2(rtl_ssa::insn_change&, std::function<bool (rtl_ssa::insn_change&, unsigned int)>)':
> | /usr/src/debug/gcc/12.3.0-r0/gcc/rtl-ssa/changes.cc:840: undefined reference to `add_clobbers(rtx_def*, int)'
> | /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/gcc/12.3.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: /usr/src/debug/gcc/12.3.0-r0/gcc/rtl-ssa/changes.cc:803: undefined reference to `recog(rtx_def*, rtx_insn*, int*)'
> | /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/gcc/12.3.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: rtl-ssa/changes.o: in function `recog_level2(rtl_ssa::insn_change&, std::function<bool (rtl_ssa::insn_change&, unsigned int)>)':
> | /usr/src/debug/gcc/12.3.0-r0/gcc/rtl-ssa/changes.cc:840: undefined reference to `add_clobbers(rtx_def*, int)'
> | /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/gcc/12.3.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: /usr/src/debug/gcc/12.3.0-r0/gcc/rtl-ssa/changes.cc:803: undefined reference to `recog(rtx_def*, rtx_insn*, int*)'
> | /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/gcc/12.3.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: rtl-ssa/changes.o: in function `recog_level2(rtl_ssa::insn_change&, std::function<bool (rtl_ssa::insn_change&, unsigned int)>)':
> | /usr/src/debug/gcc/12.3.0-r0/gcc/rtl-ssa/changes.cc:840: undefined reference to `add_clobbers(rtx_def*, int)'
> | /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/gcc/12.3.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: /usr/src/debug/gcc/12.3.0-r0/gcc/rtl-ssa/changes.cc:803: undefined reference to `recog(rtx_def*, rtx_insn*, int*)'
> | /home/amgt/pi3/poky/rpi-build/tmp/work/cortexa72-poky-linux/gcc/12.3.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/12.3.0/ld: rtl-ssa/changes.o: in function `recog_level2(rtl_ssa::insn_change&, std::function<bool (rtl_ssa::insn_change&, unsigned int)>)':
[... cut ...] TVN9tesseract6SeriesE]+0xc0): undefined reference to `tesseract::Plumbing::AddToStack(tesseract::Network*)'
> | collect2: error: ld returned 1 exit status
> | make: *** [base/unixlink.mak:172: bin/gs] Error 1
> | ERROR: oe_runmake failed
> | WARNING: exit code 1 from a shell command.
> ERROR: Task (/home/amgt/pi3/poky/meta/recipes-extended/ghostscript/ghostscript_10.0.0.bb:do_compile) failed with exit code '1
> ```
@agherzan  can you please suggest what steps i can fallow to resolve this issue ?

kraj, 2023-09-01:
I think you might be running out of memory on your host, gcc is not particularly informing about these issues. Maybe clamp on PARALLEL_MAKE and BB_NUMBER_THREADS to a smaller number than NCPU value which is what is default. Mayeb select NCPU/2 or somesuch value.

agherzan, 2023-09-01:
I don't see anything obvious in the logs, so @kraj is a good check.

AmolGoreExtc, 2023-09-15:
Hello @kraj  and @agherzan 

I changed the host system, Unfortunately i have faced new issue.

ERROR: ExpansionError during parsing /home/amgt-ws0005l-ind-pun/Projetcs/RaspberryPi/Pi4B/meta-openembedded/meta-oe/recipes-graphics/libsdl/libsdl2-ttf_2.20.2.bb                            | ETA:  0:00:03
Traceback (most recent call last):
  File "Var <fetcher_hashes_dummyfunc[vardepvalue]>", line 1, in <module>
  File "/home/amgt-ws0005l-ind-pun/Projetcs/RaspberryPi/Pi4B/poky/bitbake/lib/bb/fetch2/__init__.py", line 834, in get_hashvalue(d=<bb.data_smart.DataSmart object at 0x7fc9a3b76b30>, method_name='sortable_revision'):
     def get_hashvalue(d, method_name='sortable_revision'):
    >    pkgv, revs = _get_srcrev(d, method_name=method_name)
         return " ".join(revs)
  File "/home/amgt-ws0005l-ind-pun/Projetcs/RaspberryPi/Pi4B/poky/bitbake/lib/bb/fetch2/__init__.py", line 804, in _get_srcrev(d=<bb.data_smart.DataSmart object at 0x7fc9a3b76b30>, method_name='sortable_revision'):
         if not format:
    >        raise FetchError("The SRCREV_FORMAT variable must be set when multiple SCMs are used.\n"\
                              "The SCMs are:\n%s" % '\n'.join(scms))
bb.data_smart.ExpansionError: Failure expanding variable fetcher_hashes_dummyfunc[vardepvalue], expression was ${@bb.fetch.get_hashvalue(d)} which triggered exception FetchError: Fetcher failure: The SRCREV_FORMAT variable must be set when multiple SCMs are used.
The SCMs are:
git://github.com/libsdl-org/SDL_ttf.git;branch=release-2.20.x;protocol=https
git://github.com/libsdl-org/freetype.git;branch=VER-2-12-1-SDL;destsuffix=git/external/freetype;name=freetype;protocol=https
git://github.com/libsdl-org/harfbuzz.git;branch=2.9.1-SDL;destsuffix=git/external/harfbuzz;name=harfbuzz;protocol=https
The variable dependency chain for the failure is: fetcher_hashes_dummyfunc[vardepvalue]

Amg-AmolGore, 2023-09-16:
Above mentioned issue  resolved after host system upgrade and changing branch to dunfell. @kraj  and @agherzan for support.

closed 2023-09-16 by Amg-AmolGore