agherzan/meta-raspberrypi #1222: Backport 6.1 to kirkstone -> no serial console

nandra, 2023-09-14:
Hi, I'm using kirkstone release of this layer in my project but would like to use new kernel 6.1. I've backported it to my layer (basically took it from master branch). Kernel seems to boot but I cannot see any output on serial console. In kirkstone I've set `console=ttyAMA0,115200` in cmdline.txt file. Does something changed between kernel releases? I've tried to change to `console=ttyS0,155200` but witout success. Any ideas what else I can check please? Thanks.

nandra, 2023-09-19:
@agherzan I've checked and rpi-firmware is same on kirkstone & master if you mean by "bootloader update"

mainmachine, 2023-12-05:
Looks like `/etc/udev/rules.d/99-com.rules` in a kirkstone build is different than versus bookworm. Kirkstone has this section which is absent from what I see on RPIOS:
```
KERNEL=="ttyAMA0", PROGRAM="/bin/sh -c '\
	ALIASES=/proc/device-tree/aliases; \
	if cmp -s $$ALIASES/uart0 $$ALIASES/serial0; then \
		echo 0;\
	elif cmp -s $$ALIASES/uart0 $$ALIASES/serial1; then \
		echo 1; \
	else \
		exit 1; \
	fi\
'", SYMLINK+="serial%c"
```
So udev on kirkstone is setting up a serial on uart, presumably the master branch is not doing this. About to attempt porting 6.1 to kirkstone myself so I can check this, but you can compare `/etc/udev/rules.d/99-com.rules` between kirkstone and master builds to verify.

[kirkstone_99-com.rules.txt](https://github.com/agherzan/meta-raspberrypi/files/13571999/kirkstone_99-com.rules.txt)
[bookworm_99-com.rules.txt](https://github.com/agherzan/meta-raspberrypi/files/13572000/bookworm_99-com.rules.txt)

yayay, 2024-04-10:
@nandra were you able to boot a 6.1 Linux kernel from the kirkstone branch?

I'm not having much success with that. I don't get any kernel log output on the serial console.

nandra, 2024-04-13:
@yayay Please try to update also rpi firmware (bootfiles). It should work then.

yayay, 2024-04-26:
In my case, the kernel was panicking early in the boot process during a USB controller init and I wasn't getting the printout on serial console because tty numbering has changed for some reason. 
Now,
5.15        -> 6.1
ttyAMA0 -> ttyAMA1
ttyAMA1 -> ttyAMA5