agherzan/meta-raspberrypi #1256: RPI4 cannot boot on using u-boot on nanbield

andcor, 2023-12-20:
**Description**
We have some issue with our raspberry pi build where we originally designed it around honister, but now would like to upgrade to nanbield as the other yocto platforms we are using. When we upgrade to nanbield, u-boot doesn't seem to want to boot from the sd-card, but instead default to trying to boot on ethernet phy. 

**Steps to reproduce the issue:**
In order to isolate the issue I have tried using the kas file in the root of meta-rasberrypi and just add the following two lines to the local.conf section of the image definition:
```
LICENSE_FLAGS_ACCEPTED += "synaptics-killswitch" 
RPI_USE_U_BOOT = "1"
```
and then just load the resulting image to the sd-card. However, this doesn't boot

**Describe the results you received:**
U-boot doesn't find my configuration and does not build the kernel. The trace below is what happens on the terminal:
```
U-Boot 2023.10 (Oct 02 2023 - 14:39:59 +0000)

DRAM:  948 MiB (effective 3.9 GiB)
RPI 4 Model B (0xc03115)
Core:  211 devices, 16 uclasses, devicetree: board
MMC:   mmcnr@7e300000: 1, mmc@7e340000: 0
Loading Environment from FAT... Unable to read "uboot.env" from mmc0:1...
In:    serial,usbkbd
Out:   serial,vidconsole
Err:   serial,vidconsole
Net:   eth0: ethernet@7d580000
PCIe BRCM: link up, 5.0 Gbps x1 (SSC)
PCI: Failed autoconfig bar 10
starting USB...
Bus xhci_pci: pci_hose_bus_to_phys: invalid physical address
xhci-pci init cannot map PCI mem bar
probe failed, error -5
No working controllers found
Hit any key to stop autoboot:  0
Card did not respond to voltage select! : -110
Bus xhci_pci: pci_hose_bus_to_phys: invalid physical address
xhci-pci init cannot map PCI mem bar
probe failed, error -5
No working controllers found
ethernet@7d580000 Waiting for PHY auto negotiation to complete......... TIMEOUT !
bcmgenet: PHY startup failed: -110
ethernet@7d580000 Waiting for PHY auto negotiation to complete......... TIMEOUT !
bcmgenet: PHY startup failed: -110
U-Boot>
```

**Describe the results you expected:**
I expected u-boot to boot my kernel and rootfs

**Additional information you deem important (e.g. issue happens only occasionally):**
- When building the kas image on the honister branch of meta-raspberrypi with the same two changes (and changing all refspecs from master to honister), the image boots as expected
```
U-Boot 2021.07 (Jul 05 2021 - 15:11:28 +0000)

DRAM:  3.9 GiB
RPI 4 Model B (0xc03115)
MMC:   mmcnr@7e300000: 1, mmc@7e340000: 0
Loading Environment from FAT... OK
In:    serial
Out:   serial
Err:   serial
Net:   eth0: ethernet@7d580000
PCIe BRCM: link up, 5.0 Gbps x1 (SSC)
starting USB...
Bus xhci_pci: Register 5000420 NbrPorts 5
Starting the controller
USB XHCI 1.00
scanning bus xhci_pci for devices... 2 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
263 bytes read in 5 ms (50.8 KiB/s)
## Executing script at 02400000
7393560 bytes read in 329 ms (21.4 MiB/s)
## Booting kernel from Legacy Image at 00080000 ...
   Image Name:   Linux-5.10.83-v7l
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    7393496 Bytes = 7.1 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 2eff3400
   Booting using the fdt blob at 0x2eff3400
   Loading Kernel Image
   Using Device Tree in place at 2eff3400, end 2f002fa3

Starting kernel ...
```
- If we copy over the kernel7l.img from the boot partition of the honister build to the nanbield build, it seems the image boots correctly. 
- the md5sum of the kernel7l.img of the boot partition of the nanbield build is not the same as that from the image downloaded by the bootfiles recipe (I have downloaded and unpacked the firmware package manually). I haven't however, been able to figure out which recipes changes the file after downloading
- if just copying the kernel7l.img from the file downloaded by the bootfiles recipe (again manually downloaded and unpacked), it seems to skip u-boot entirely in the boot process
- the kernel7l.img of the boot partition of the nanbield image is much smaller (~600kB) than the kernel7l.img file from the firmware image (6,8 MB). I might have fully misunderstood where the kernel7l.img is coming from. I just can't find it anywhere else in my build folder than in the build folder for bootfiles.bb 

**Additional details (revisions used, host distro, etc.):**
We have tried to build on both ubuntu 22.04 and WSL ubuntu 22.04 with the exact same results.

muehle28, 2024-01-31:
We have the same problem. Were you able to resolve it?

andcor, 2024-01-31:
For now, sadly no. As I had other more pressing tasks on my desk, I had to resort to use the honister branch in stead. I hope I will be able to find time for further debug later, if no one else finds a fix before me.

brobusta, 2024-02-08:
I am able to boot the rpi4b with U-Boot 2024.1
I'm using meta-rauc-raspberrypi so you may want to look into that layer to see if there is any new fixes.

nknuth, 2024-03-12:
This issue is not a kernel issue but a u-boot issue.  It was caused by a bug submitted to the u-boot code in the fall of 2023.  I have submitted a patch to the u-boot dev team.  

To fix this issue, the variables mmc0, mmc1, and mmc2 need to be added to the boot_targets variable.  If you attach a serial port or video console, halt the boot at the U-Boot prompt and type the following commands.  You can see that the boot_targets is set to `mmc usb pxe dhcp.`  It should be set to `mmc0 mmc1 mmc2 usb pxe dhcp.`

```
U-Boot> echo $boot_targets
mmc usb pxe dhcp
U-Boot> setenv boot_targets mmc0 mmc1 mmc2 usb pxe dhcp 
U-Boot> saveenv 
Saving Environment to FAT... OK
```

Then reset the Raspberrypi4 and it should boot the kernel.

andcor, 2024-03-13:
It is fixed in the u-boot recipe on poky master. I have just copied over that recipe to my own local layer and uses this newer version of u-boot which works fine