agherzan/meta-raspberrypi #1260: Add support for vc4-kms-dsi-waveshare-panel overlay

Romain-Donze, 2024-01-05:
Since 1a6b218d786990f461b3598250906a83fbb93aaa, https://github.com/raspberrypi/linux has add an overlay for the waveshare dsi displays. how can we add it to a yocto build? and will it work with vc4-fkms-v3d-overlay?

Because due to https://github.com/agherzan/meta-raspberrypi/commit/a6fa6b3aec36b92c3750852dd6788af9d2ce08f6. vc4-kms-v3d only works with HDMI, for the rpi display we need to change it back to vc4-fkms-v3d in /boot/config.txt

Romain-Donze, 2024-01-13:
For those wondering what you have to do for now to support a waveshare panel is add to local.conf 

`
RPI_KERNEL_DEVICETREE_OVERLAYS:append = " \
    overlays/vc4-kms-dsi-waveshare-panel.dtbo \
"

RPI_EXTRA_CONFIG:append = " \n\
# Enable waveshare panel \n\
dtoverlay=vc4-kms-dsi-waveshare-panel,7_0_inchC,i2c1 \n\ # choose the right display
"
`

agherzan, 2024-01-23:
I am happy to accept a change to add the respective dtb. Could you prepare a PR?

shaunmulligan, 2024-04-12:
What if we ask really nicely Mr @agherzan 😄

agherzan, 2024-04-18:
@shaunmulligan That I can't refuse :) But what is the question again?

shaunmulligan, 2024-04-19:
To get the `vc4-kms-dsi-waveshare-panel` dtb added to meta-raspberrypi otherwise I will have to try PR it myself and you know how badly that could go 😅

agherzan, 2024-04-29:
As a person without the respective panel, I'd be happier to get that from someone else who can also test it.