agherzan/meta-raspberrypi #1270: kirkstone: ERROR: gstreamer1.0-plugins-bad-1.20.7-r0 do_configure: QA Issue: gstreamer1.0-plugins-bad: invalid PACKAGECONFIG: gpl [invalid-packageconfig]

Herrie82, 2024-01-26:
**Description**

gstreamer1.0-plugins-bad fails to build due to removed gpl packageconfig option.

**Steps to reproduce the issue:**
1. Build with kirkstone branch
2.
3.

**Describe the results you received:**
ERROR: gstreamer1.0-plugins-bad-1.20.7-r0 do_configure: QA Issue: gstreamer1.0-plugins-bad: invalid PACKAGECONFIG: gpl [invalid-packageconfig]

**Describe the results you expected:**
Successful build.

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

kluszon, 2024-02-16:
I met the same during the building, but in my case, bitbake classified it as a warning and finished normally the building process. Unfortunately gstreamer1.0 works only with specific settings and without hardware acceleration.

kluszon, 2024-02-16:
I still see information about it, but in my case, it warning and not an error (branch kirkstone uptodate):

`WARNING: gstreamer1.0-plugins-bad-1.20.7-r0 do_configure: QA Issue: gstreamer1.0-plugins-bad: invalid PACKAGECONFIG: gpl [invalid-packageconfig]`