agherzan/meta-raspberrypi #1274: Can't build rpi5 image with deb packages

mast, 2024-01-31:
**Description**

Thanks for amazing job bringing rpi5 support. I tried to build it yesterday and I'm having issue when deb package manager is on. Rpm works ok.
raspberrypi4 and raspberrypi4-64 machines build well even with deb package manager turned on.

```
Build Configuration:
BB_VERSION           = "2.7.1"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-20.04"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "raspberrypi5"
DISTRO               = "poky"
DISTRO_VERSION       = "4.3+snapshot-7af374c90c349af2c7ae9cf7d4fa14eeeb23b108"
TUNE_FEATURES        = "aarch64 armv8-2a crypto cortexa76"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "master:7af374c90c349af2c7ae9cf7d4fa14eeeb23b108"
meta-raspberrypi     = "master:9c901bf170960069576b9323f69d43a8b6c4bd71"
meta-oe              
meta-multimedia      
meta-networking      
meta-python          = "master:0f90c7dc981263de7b2ab9e9719f462c1c48c990"

Sstate summary: Wanted 2260 Local 0 Mirrors 0 Missed 2260 Current 0 (0% match, 0% complete)########################################################              | ETA:  0:00:00
Initialising tasks: 100% |#######################################################################################################################################| Time: 0:00:03
NOTE: Executing Tasks
ERROR: linux-raspberrypi-1_6.1.61+git-r0 do_package_write_deb: Fatal errors occurred in subprocesses:
Command 'PATH="/mast-home/yocto-pi5/build/tmp/sysroots-uninative/x86_64-linux/usr/bin:/mast-home/yocto-pi5/poky/scripts:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot-native/usr/bin/aarch64-poky-linux:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot/usr/bin/crossscripts:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot-native/usr/sbin:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot-native/usr/bin:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot-native/sbin:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot-native/bin:/mast-home/yocto-pi5/poky/bitbake/bin:/mast-home/yocto-pi5/build/tmp/hosttools" dpkg-deb -b /mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/packages-split/kernel-module-snd-usb-hiface-6.1.61-v8_16k /mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/deploy-debs/raspberrypi5' returned non-zero exit status 2.
Subprocess output:dpkg-deb: error: package name has characters that aren't lowercase alphanums or '-+.'
Command 'PATH="/mast-home/yocto-pi5/build/tmp/sysroots-uninative/x86_64-linux/usr/bin:/mast-home/yocto-pi5/poky/scripts:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot-native/usr/bin/aarch64-poky-linux:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot/usr/bin/crossscripts:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot-native/usr/sbin:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot-native/usr/bin:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot-native/sbin:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot-native/bin:/mast-home/yocto-pi5/poky/bitbake/bin:/mast-home/yocto-pi5/build/tmp/hosttools" dpkg-deb -b /mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/packages-split/kernel-module-snd-usb-audio-6.1.61-v8_16k /mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/deploy-debs/raspberrypi5' returned non-zero exit status 2.
Subprocess output:dpkg-deb: error: package name has characters that aren't lowercase alphanums or '-+.'
Command 'PATH="/mast-home/yocto-pi5/build/tmp/sysroots-uninative/x86_64-linux/usr/bin:/mast-home/yocto-pi5/poky/scripts:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot-native/usr/bin/aarch64-poky-linux:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot/usr/bin/crossscripts:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot-native/usr/sbin:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot-native/usr/bin:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot-native/sbin:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot-native/bin:/mast-home/yocto-pi5/poky/bitbake/bin:/mast-home/yocto-pi5/build/tmp/hosttools" dpkg-deb -b /mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/packages-split/kernel-module-snd-usb-toneport-6.1.61-v8_16k /mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/deploy-debs/raspberrypi5' returned non-zero exit status 2.
Subprocess output:dpkg-deb: error: package name has characters that aren't lowercase alphanums or '-+.'
Command 'PATH="/mast-home/yocto-pi5/build/tmp/sysroots-uninative/x86_64-linux/usr/bin:/mast-home/yocto-pi5/poky/scripts:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot-native/usr/bin/aarch64-poky-linux:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot/usr/bin/crossscripts:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot-native/usr/sbin:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot-native/usr/bin:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot-native/sbin:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot-native/bin:/mast-home/yocto-pi5/poky/bitbake/bin:/mast-home/yocto-pi5/build/tmp/hosttools" dpkg-deb -b /mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/packages-split/kernel-module-snd-usb-line6-6.1.61-v8_16k /mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/deploy-debs/raspberrypi5' returned non-zero exit status 2.
Subprocess output:dpkg-deb: error: package name has characters that aren't lowercase alphanums or '-+.'
Command 'PATH="/mast-home/yocto-pi5/build/tmp/sysroots-uninative/x86_64-linux/usr/bin:/mast-home/yocto-pi5/poky/scripts:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot-native/usr/bin/aarch64-poky-linux:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot/usr/bin/crossscripts:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot-native/usr/sbin:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot-native/usr/bin:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot-native/sbin:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot-native/bin:/mast-home/yocto-pi5/poky/bitbake/bin:/mast-home/yocto-pi5/build/tmp/hosttools" dpkg-deb -b /mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/packages-split/kernel-module-snd-ua101-6.1.61-v8_16k /mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/deploy-debs/raspberrypi5' returned non-zero exit status 2.
Subprocess output:dpkg-deb: error: package name has characters that aren't lowercase alphanums or '-+.'
Command 'PATH="/mast-home/yocto-pi5/build/tmp/sysroots-uninative/x86_64-linux/usr/bin:/mast-home/yocto-pi5/poky/scripts:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.1.61+git/recipe-sysroot-native/usr/bin/aarch64-poky-linux:/mast-home/yocto-pi5/build/tmp/work/raspberrypi5-poky-li [... cut ...]
MACHINE = "raspberrypi5"
ENABLE_UART = "1"
ENABLE_SPI_BUS = "1"
ENABLE_I2C = "1"
ENABLE_DWC2_PERIPHERAL = "1"
KERNEL_MODULE_AUTOLOAD:rpi += "i2c-dev i2c-bcm2708"
LICENSE_FLAGS_ACCEPTED = "synaptics-killswitch"

PACKAGE_CLASSES = "package_deb"
INIT_MANAGER = "systemd"
EXTRA_IMAGE_FEATURES += " package-management "
DISTRO_FEATURES += "usbgadget x11 pam"
IMAGE_INSTALL:append = " nano openssh apt dpkg bash alsa-utils python3 python3-pip python3-setuptools curl kernel-dev procps kernel-devsrc v4l-utils "
```

mast, 2024-01-31:
It might be fixed here:
https://github.com/raspberrypi/linux/commit/526b986030c2ef61e2dd61641838290810da61f9

kraj, 2024-01-31:
can you test this kernel update and see if this helps - https://github.com/agherzan/meta-raspberrypi/pull/1272

mast, 2024-01-31:
@kraj thanks, I'm already building a little different one but I guess it's only couple of days difference.

SRCREV_machine = "8fd7f44624538675abadc73f5a44e95016964d22"
SRCREV_meta = "23ce79d8ef2e20f27aa87a9b2fdfedfff9d7836d"

I will update you.

agherzan, 2024-02-01:
> It might be fixed here: [raspberrypi/linux@526b986](https://github.com/raspberrypi/linux/commit/526b986030c2ef61e2dd61641838290810da61f9)

This looks like the culprit here.

mast, 2024-02-01:
I confirm https://github.com/agherzan/meta-raspberrypi/pull/1272 fixes the issue.

closed 2024-02-08 by agherzan