agherzan/meta-raspberrypi #1279: No machine configuration for raspberrypi 5

rogueWookie, 2024-02-15:
## BACKGROUND

I see there's a [pull request](https://github.com/agherzan/meta-raspberrypi/pull/1237) out there on adding rpi5 support. In that pull request are comments saying to build off master. I have tried building off master for meta-raspberrypi and poky but kept receiving errors about machine variable being invalid.

## ISSUE

I followed the [Yocto Quick Start Guide](https://docs.yoctoproject.org/brief-yoctoprojectqs/index.html). I checked out master on both meta-raspberry layer and poky layer, set machine to "raspberrypi5" and then built "core-image-minimal". I hit the following error when doing so. I have also tried building "rpi-test-image" and "core-image-base" all of which result in the same error message. 

```bash
~/yocto/rpi5build took 57s 
☕ ❯ bitbake core-image-minimal
ERROR:  OE-core's config sanity checker detected a potential misconfiguration.
    Either fix the cause of this error or at your own risk disable the checker (see sanity.conf).
    Following is the list of potential problems / advisories:
    MACHINE=raspberrypi5 is invalid. Please set a valid MACHINE in your local.conf, environment or other configuration file.
Summary: There was 1 ERROR message, returning a non-zero exit code.
```

## WHAT I'VE TRIED

I have also tried disabling the checker in my `local.conf` but that doesn't work either. I have tried searching the meta-raspberrypi documentation but there's no build instructions for the raspberrypi5. I presume this is still a WIP but according to the pull request I linked above, I should be able to build off master. 

## STEPS TO REPRODUCE

1. Pull down poky and meta-raspberrypi layers
2. Checkout master on meta-raspberrypi layer `cd ./meta-raspberrypi/ && git checkout -t origin/master -b local-master`
3. Checkout master on poky layer `cd ./poky && git checkout -t origin/master -b local-master`
4. Create build workspace `source src/poky/oe-init-build-env rpi5build`
5. Set machine variable `MACHINE = raspberrypi5" in `local.conf` (not sure why there's no "raspberrypi5-64)
6. Add license exception `LICENSE_FLAGS_ACCEPTED = "synaptics-killswitch"` to `local.conf`
7. Build an image `bitbake rpi-test-image` or `bitbake core-image-minimal`

## EXPECTATION

Expectation here is that I should be able to just build my image, flash it to sd card (via bzip and dd) and be on my merry way. I understand uboot support isn't added but my understanding is that I should still be able to boot with the above steps.

agherzan, 2024-02-19:
Have you added the meta-raspberrypy layer in the layers configuration file?

nknuth, 2024-03-10:
I can confirm that your expectations are correct. I built from the master branch (ref 92a9b7a), flashed it to my SD card, and was able to boot and run successfully on my Raspberry Pi5. Building 'core-image-weston' also works well and provides a desktop and shell. Finally, the latest build from 'master' uses the new Raspberry Pi5 UART on /dev/AMA10.  

In looking at the error you reported, you might not have the meta-raspberry pi layers in your bilayers.conf file.  My bblayers.conf file is as follows:
BBLAYERS += " \
    /home/nknuth/raspberrypi5-nknuth-build-latest/sources/poky/meta \
    /home/nknuth/raspberrypi5-nknuth-build-latest/sources/poky/meta-poky \
    /home/nknuth/raspberrypi5-nknuth-build-latest/sources/poky/meta-yocto-bsp \
    /home/nknuth/raspberrypi5-nknuth-build-latest/sources/meta-raspberrypi \
    /home/nknuth/raspberrypi5-nknuth-build-latest/sources/meta-openembedded/meta-python \
    /home/nknuth/raspberrypi5-nknuth-build-latest/sources/meta-openembedded/meta-oe \
"

closed 2024-03-18 by agherzan