agherzan/meta-raspberrypi #1282: Using external kernel source

emilm, 2024-02-17:
Is it feasible to combine meta-raspberrypi with kernel source from another repository? Like if a 3rd party has a special kernel for their hardware with everything readily patched in, it is more tempting to use that than re-applying patches and keeping up.

Thanks!

agherzan, 2024-02-19:
There are ways of achieving that. Either as a bbappend to the existing kernel recipes or (cleaner) using another provider and overriding the default kernel one. Have you tried any of these venues?

emilm, 2024-02-20:
Well I haven't tried yet because things like LINUX_RPI_KMETA_BRANCH and KMETA makes me wonder if I am tied to anything special for raspberry pi

agherzan, 2024-03-04:
`LINUX_RPI_KMETA_BRANCH` is only used in the SRC_URI definition. I'm sure you can handle that as per your custom kernel fork requirements.