agherzan/meta-raspberrypi #1285: raspberrypi4-64 kirkstone linux 6.1 image stuck on boot logos

cveilleux, 2024-02-19:
**Description**

Thanks to the [backport of raspberrypi5 to kirkstone](https://github.com/agherzan/meta-raspberrypi/pull/1271), we now have a 6.1 kernel available.

I tried to upgrade a raspberrypi4-64 image to use linux 6.1.

I am using u-boot. I can see the u-boot output up to Starting kernel ..., then the screen gets stuck on the pi logos and there is no kernel logs. Also, the green LED activity stops immediately.

- [u-boot output](https://github.com/agherzan/meta-raspberrypi/assets/418168/93b513b1-f7fe-4c7d-a139-55cdd74526e0)
- [kernel stuck on logos](https://github.com/agherzan/meta-raspberrypi/assets/418168/3f572417-7d2f-449a-a473-b84cf43dfd27)

I am guessing the kernel boots but stops very early in the process.

Any suggestions on how to diagnose this further?

shr-project, 2024-02-22:
It might be unrelated to your issue, but today I've noticed that the default /boot partition is only 20MB:

wic/sdimage-raspberrypi.wks:part /boot --source bootimg-partition --ondisk mmcblk0 --fstype=vfat --label boot --active --align 4096 --size 20

and the kernel image is bigger than that:
24M     tmp-glibc/deploy/images/raspberrypi4-64/Image-raspberrypi4-64-1-6.1.77+git0+43d1723dbe_77fc1fbcb5-r0.1-20110405230000.bin

with webOS OSE we're using own .wks file with 80MB /boot, but I wonder how it works for other people who might be using the default .wks file from this layer, is it extended so that it fits during the boot and then you don't need to ever write to /boot?

kraj, 2024-02-22:
@shr-project regardless, this should be fixed. Can you send a patch to raise it to 100M?

shr-project, 2024-02-22:
Looks like it is increased automatically:
```
tmp-glibc/deploy/images/raspberrypi4-64 $ sfdisk -l core-image-minimal-raspberrypi4-64.rootfs--1.0-r0-20110405230000.wic
Disk core-image-minimal-raspberrypi4-64.rootfs--1.0-r0-20110405230000.wic: 241.2 MiB, 252915712 bytes, 493976 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x076c4a2a

Device                                                                Boot  Start    End Sectors   Size Id Type
core-image-minimal-raspberrypi4-64.rootfs--1.0-r0-20110405230000.wic1 *      8192 157815  149624  73.1M  c W95 FAT32 (LBA)
core-image-minimal-raspberrypi4-64.rootfs--1.0-r0-20110405230000.wic2      163840 493975  330136 161.2M 83 Linux
```

"--size 20M" is just minimum (our wks uses "--fixed-size 80M"):
https://github.com/webosose/meta-webosose/blob/master/meta-webos-raspberrypi/wic/sdimage-raspberrypi-webos.wks

https://docs.yoctoproject.org/ref-manual/kickstart.html
says:

> --size: The minimum partition size. Specify as an integer value optionally followed by one of the units “k” / “K” for kibibyte, “M” for mebibyte and “G” for gibibyte. The default unit if none is given is “M”. You do not need this option if you use --source.
> --fixed-size: The exact partition size. Specify as an integer value optionally followed by one of the units “k” / “K” for kibibyte, “M” for mebibyte and “G” for gibibyte. The default unit if none is given is “M”. Cannot be specify together with --size. An error occurs when assembling the disk image if the partition data is larger than --fixed-size.

cveilleux, 2024-02-23:
I have gotten the 6.1 kernel to boot using a fresh yocto build.

I can confirm 6.1.61 running well on raspberrypi4-64 board. Thanks!

closed 2024-02-23 by cveilleux