agherzan/meta-raspberrypi #1299: Fetching raspberrypi-firmware git requires lots of gigabytes

jdapena, 2024-03-20:
**Description**

`do_fetch` stage of `rpi-bootfiles` takes ages because of the huge size of the repository, even when using `BB_GIT_SHALLOW`. Here, at 37% of the fetch, it already had to download more than 14 GB.

**Steps to reproduce the issue:**
1. Using https://github.com/Igalia/meta-chromium and building with `KAS_MACHINE=raspberrypi4-64 kas build kas:chromium.yml:kas:commercial.yml`

**Describe the results you received:**

Fetching fails and/or takes ages while downloading `raspberrypi-firmware` repository.

**Describe the results you expected:**

Fetching should be faster, and not fetch the whole repository. Ideally by fetching tagged tarballs as needed. An intermediate solution could be fetching the generated zip from github. I.e. https://github.com/raspberrypi/firmware/archive/5689dd184872b809af91d6636a832b1484d90fef.zip (but that tarball is AFAIK generated at navigation. Tagged versions are usually pregenerated.

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

Host: Ubuntu.
Local bandwitdh: around 10-15 MB per second fetching from github.

kraj, 2024-03-20:
https://github.com/agherzan/meta-raspberrypi/pull/1298 should fix it

jdapena, 2024-03-20:
> #1298 should fix it

Yes, it looks like that should fix it. I will close this one when landed.

agherzan, 2024-03-27:
The fix should be merged. Thanks, @kraj!

closed 2024-03-27 by agherzan