agherzan/meta-raspberrypi #1303: Raspberry 5 camera doesn't work

al1img, 2024-03-30:
**Camera doesn't work on Raspberry 5**

**Steps to reproduce the issue:**
1.  Set MACHINE = "raspberrypi5"
2. Add libcamera-apps to the target image: IMAGE_INSTALL:append = "libcamera-apps"

**Describe the results you received:**

`libcamera-hello` displays image from camera.

**Describe the results you expected:**

```bash
[0:17:43.474887809] [833]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+46-075b54d5-dirty (2024-03-30T18:04:07+00:00)
Preview window unavailable
ERROR: *** no cameras available ***
````

**Additional information you deem important (e.g. issue happens only occasionally):**

Camera is detected by kernel:

```bash
[    2.130214] platform 1f00110000.csi: Fixed dependency cycle(s) with /axi/pcie@120000/rp1/i2c@88000/ov5647@36
[    5.397222] rp1-cfe 1f00110000.csi: found subdevice /axi/pcie@120000/rp1/i2c@88000/ov5647@36
[    5.463006] rp1-cfe 1f00110000.csi: Using sensor ov5647 6-0036 for capture
```

**Additional details (revisions used, host distro, etc.):**

I've also tried switching `libcamera` recipe to `https://github.com/raspberrypi/libcamera` repo. Result is the same: no cameras available.

AshishShrikhande2408, 2025-01-11:
I was able to run libcamera-apps successfully but I am currently struggling for preview window. What is missing the libcamera recipe is libpisp which is required for pi5 and these changes are not yet upstreamed.

gizahNL, 2025-03-14:
> I was able to run libcamera-apps successfully but I am currently struggling for preview window. What is missing in the libcamera recipe is libpisp which is required for pi5 and these changes are not yet up-streamed.

What did you do to get libcamera-apps to detect your cam? I'm in the same boat...

gizahNL, 2025-03-14:
I'll write up a draft MR where libcamera can be pulled from rpi-distro sources as an option and be build with libpisp

gizahNL, 2025-03-14:
> I had success using that version: https://github.com/openembedded/meta-openembedded/blob/scarthgap/meta-multimedia%2Frecipes-multimedia%2Flibcamera%2Flibcamera_0.4.0.bb

I tried upgrading libcamera to upstream 0.4.0 before commenting here, it didn't seem to fix it for me, I've just created a PR which does fix it for me.

jwinarske, 2025-08-25:
@agherzan I have cameras working on a raspberrypi5 with meta-raspberrypi master.

* If you attempt to use upstream libcamera on `master` with rpi5  it will fail with [Unable to acquire a CFE instance](https://github.com/raspberrypi/libcamera/issues/298#top)
* Use of the raspberrypi linux kernel requires using the raspberrypi5 libcamera fork, `next` branch.  It will not work with the oe-core libcamera recipe as-is in `master`; not tested else where but pretty sure this is the case for all other branches.
* Adding a libpisp recipe was required.

I propose the following changes to meta-raspberrypi `master`:
1. rpi-libcamera recipe.  Uses the raspberrypi libcamera fork, `next` branch.
2. libpisp recipe.
3. Set `PREFERRED_PROVIDER_libcamera=rpi-libcamera`

Let me know if that works, if so I will submit the PR.

gizahNL, 2025-08-26:
@jwinarske look at: https://github.com/agherzan/meta-raspberrypi/pull/1436 ;)  

Haven't had time yet to make the requested changes (holidays + OSS Europe now), somewhere next week most likely.