agherzan/meta-raspberrypi #1309: Can't find vclog

ajlennon, 2024-04-22:
Hi all,

We need to investigate some issues with DeviceTree and have been advised that we need vclog.

I can't spot support for this anywhere here. Does anybody know if there's a Yocto recipe around for this?

Thanks!

Alex

nknuth, 2024-04-23:
vclog is source code is located here https://github.com/raspberrypi/utils/tree/master/vclog

nknuth, 2024-04-23:
I produced a quick recipe in the past to compile raspberrypi-utils.  You could start with this recipe. 
# Recipe created by recipetool
# This is the basis of a recipe and may need further editing in order to be fully functional.
# (Feel free to remove these comments when editing.)

# WARNING: the following LICENSE and LIC_FILES_CHKSUM values are best guesses - it is
# your responsibility to verify that the values are complete and correct.
#
# The following license files were not able to be identified and are
# represented as "Unknown" below, you will need to check them yourself:
#   LICENCE
LICENSE = "BSD-3-Clause"
LIC_FILES_CHKSUM = "file://LICENCE;md5=4c01239e5c3a3d133858dedacdbca63c"

SRC_URI = "git://github.com/raspberrypi/utils.git;protocol=https;branch=master"

# Modify these as desired
PV = "1.0+git"
SRCREV = "cf46fe84b2a345146e067a9a3529ba6e28517c8e"

RDEPENDS:${PN} += "perl bash"

DEPENDS = "dtc"

S = "${WORKDIR}/git"

inherit cmake

# Specify any options you want to pass to cmake using EXTRA_OECMAKE:
EXTRA_OECMAKE = ""

FILES:${PN} = "${datadir}/bash-completion/completions/* ${bindir}/*"

ajlennon, 2024-04-23:
Ah great - thanks @nknuth !

nknuth, 2024-04-23:
@kraj -- where do you want it committed?

kraj, 2024-04-24:
Master branch is good here

agherzan, 2024-04-29:
Thanks @nknuth - see https://meta-raspberrypi.readthedocs.io/en/latest/contributing.html