agherzan/meta-raspberrypi #1312: Rpi5 core-image-sato requires /etc/X11/xorg.conf.d/99-vc4.conf file for GUI to appear (not needed for Rpi4...)

PSkeberdis, 2024-04-26:
Used yocto/poky and meta-raspberrypi "nanbield" branch to build core-image-sato for MACHINE?= "raspberrypi5".

(as an aside, had to increase boot partition size by adding BOOT_SPACE = "61440" in local.conf to create sd_image...)

Made an SD card and booted Rpi5 - the xserver fails, and the GUI does not appear as expected, and boots to console to login.

After searching for solutions, found the following:
Add below configuration to a file named 99-vc4.conf and save to /etc/X11/xorg.conf.d/ directory:
```
Section "OutputClass"
  Identifier "vc4"
  MatchDriver "vc4"
  Driver "modesetting"
  Option "PrimaryGPU" "true"
EndSection
```
Following reboot, the xserver succeeds and the GUI appears as expected.

References:
https://forums.raspberrypi.com/viewtopic.php?t=361902
https://forums.raspberrypi.com/viewtopic.php?p=2159161#p2159161

Seems like this /etc/X11/xorg.conf.d/99-vc4.conf should be added to the recipe.

kraj, 2024-04-27:
@PSkeberdis thanks for reporting the issue and also possible patch. Would be great if you can push it for a PR

gitgonewithwind, 2024-05-15:
I  think I just met the same issue, X11 failed, before at commit meta-raspberry :Author: Vivien Didelot <vivien.didelot@gmail.com>
Date:   Thu Aug 3 01:36:48 2023 -0400
I use almost the same local.conf
my X11 worked well in pi4,
but in pi5 , X sever is failed. let me try this solution.
directly run X, output 
---

root@raspberrypi5:/etc/init.d# X

X.Org X Server 1.21.1.12
X Protocol Version 11, Revision 0
Current Operating System: Linux raspberrypi5 6.6.22-v8-16k #1 SMP PREEMPT Tue Mar 19 17:41:59 UTC 2024 aarch64
Kernel command line: reboot=w coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=D8:3A:DD:DE:27:D6 vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyAMA10,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait
 
Current version of pixman: 0.42.2
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Fri Mar  9 12:45:15 2018
(==) Using config file: "/etc/X11/xorg.conf"
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(EE) 
Fatal server error:
(EE) Cannot run in framebuffer mode. Please specify busIDs        for all framebuffer devices
(EE) 
(EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
(EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
(EE) 
(EE) Server terminated with error (1). Closing log file.

root@raspberrypi5:~# export DISPLAY=:0
root@raspberrypi5:~# chromium  -no-sandbox
[852:852:0515/125839.814911:ERROR:ozone_platform_x11.cc(243)] Missing X server or $DISPLAY
[852:852:0515/125839.814954:ERROR:env.cc(257)] The platform failed to initialize.  Exiting.

gitgonewithwind, 2024-05-15:
right!
the solution works for me
Thanks

carpajr, 2024-10-08:
Greetings everyone!

Was someone able to configure rotation of the screen? I can't find a way to do it over x11.

Here I'm having issues with xrandr returning `xrandr: Failed to get size of gamma for output default`.

Saicharanguda, 2025-03-31:
while building core-image-sato 

ERROR: Nothing RPROVIDES 'matchbox-wm' (but /media/saicharan/99edc95e-5b2a-4d9d-884b-37af2b9162a1/home/charan/Yocto_RaspberryPi/Yocto_raspberrypi5/sources/poky/meta/recipes-graphics/packagegroups/packagegroup-core-x11-base.bb RDEPENDS on or otherwise requires it)
matchbox-wm was skipped: missing required distro feature 'x11' (not in DISTRO_FEATURES)
NOTE: Runtime target 'matchbox-wm' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['matchbox-wm']
NOTE: Runtime target 'packagegroup-core-x11-base' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-core-x11-base', 'matchbox-wm']
ERROR: Required build target 'core-image-sato' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-sato', 'packagegroup-core-x11-base', 'matchbox-wm']