agherzan/meta-raspberrypi #1313: psplash issues with systemd

leon-anavi, 2024-04-30:
Hi,

I noticed that psplash doesn't work as expected when using it with systemd despite [the fix for race condition](98349bb367418866e008ba22ca75f4ab2984561a) with /dev/fb0:

```
Apr 30 11:04:09 raspberrypi4-64 systemd[1]: sys-devices-platform-gpu-graphics-fb0.device: Job sys-devices-platform-gpu-graphics-fb0.device/start timed out.
Apr 30 11:04:09 raspberrypi4-64 systemd[1]: Timed out waiting for device /sys/devices/platform/gpu/graphics/fb0.
Apr 30 11:04:09 raspberrypi4-64 systemd[1]: sys-devices-platform-gpu-graphics-fb0.device: Job sys-devices-platform-gpu-graphics-fb0.device/start failed with result 'timeout'.
```

**Steps to reproduce the issue:**

1. Include psplash: `IMAGE_INSTALL:append = " psplash"`
2. Use systemd: `INIT_MANAGER = "systemd"`

**Describe the results you received:**

No splash screen. Systemd service psplash-start dependency failed:

```
Apr 30 11:04:09 raspberrypi4-64 systemd[1]: sys-devices-platform-gpu-graphics-fb0.device: Job sys-devices-platform-gpu-graphics-fb0.device/start timed out.
Apr 30 11:04:09 raspberrypi4-64 systemd[1]: Timed out waiting for device /sys/devices/platform/gpu/graphics/fb0.
Apr 30 11:04:09 raspberrypi4-64 systemd[1]: sys-devices-platform-gpu-graphics-fb0.device: Job sys-devices-platform-gpu-graphics-fb0.device/start failed with result 'timeout'.
```

**Describe the results you expected:**

`psplash-start.service` to be OK and to display the splash screen while booting the system

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

Tested with core-image-base from branches master and kirkstone on Raspberry Pi 4 (machine `raspberrypi4-64`).

BuriKizilkaya, 2024-07-05:
Have you found a solution?

gitgonewithwind, 2024-07-09:
@BuriKizilkaya 
please try my solution:
https://github.com/agherzan/meta-raspberrypi/issues/1330#issuecomment-2142679662

BuriKizilkaya, 2024-07-09:
@gitgonewithwind 

I found this [PR](https://github.com/agherzan/meta-raspberrypi/pull/1268). I add a new bbappend file for the udev rules and then the boot screen appears.

If I connect the raspi with a HDMI screen and I boot up the device, then the progressbar is already 100%. Have you noticed this issue also?

gitgonewithwind, 2024-07-10:
@BuriKizilkaya 
is already 100%.
No this issue for mine. I do not use systemv

BuriKizilkaya, 2024-07-10:
@gitgonewithwind 
what is 100%?