agherzan/meta-raspberrypi #1315: linux-raspberrypi-1_6.6.22+git-r0 do_fetch: Failed to fetch URL git://github.com/raspberrypi/linux.git;name=machine;branch=rpi-6.6.y;protocol=https, attempting MIRRORS if available

VVAQARAHMED, 2024-05-08:
Hi,
I am trying to build  yocto image for raspberry pi3, and following the instruction on README file meta-raspberrypi layer.

  1. source poky/oe-init-build-env rpi-build
2. Add this layer to bblayers.conf and the dependencies above
3. Set MACHINE in local.conf to one of the supported boards
4. bitbake core-image-base

Build Configuration:
BB_VERSION           = "2.9.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-22.04"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi3"
DISTRO               = "poky"
DISTRO_VERSION       = "5.0+snapshot-7eebbdb6c45149e8bf7abbf6a9f0c8068f7668f7"
TUNE_FEATURES        = "arm vfp cortexa7 neon vfpv4 thumb callconvention-hard"
TARGET_FPU           = "hard"
meta
meta-poky
meta-yocto-bsp       = "master:7eebbdb6c45149e8bf7abbf6a9f0c8068f7668f7"
meta-raspberrypi     = "master:1879cb831f4ea3e532cb5ce9fa0f32be917e8fa3"
**Description**

ERROR: linux-raspberrypi-1_6.6.22+git-r0 do_fetch: Bitbake Fetcher Error: FetchE                                                                                                                                                             rror('Unable to fetch URL from any source.', 'git://github.com/raspberrypi/linux                                                                                                                                                             .git;name=machine;branch=rpi-6.6.y;protocol=https')
ERROR: 
Task (/home/xx/source/meta-raspberrypi/recipes-kernel/                                                                                                                                                             linux/linux-raspberrypi_6.6.bb:do_fetch) failed with exit code '1'

Regards,
Waqar Ahmed

kraj, 2024-05-08:
@VVAQARAHMED it works fine here, so I would suggest that you try
```
git clone https://github.com/raspberrypi/linux.git
```
locally on this machine to ensure that you machine is able to access and connect to this repository URI

in logs I see unwanted space in SRC_URI 'git://github.com/raspberrypi/linux .git' just before .git, I wonder if its a typo during pasting here or you see it in build logs too.

VVAQARAHMED, 2024-05-09:
I tried to clone 
git clone git://github.com/raspberrypi/linux.git
Cloning into 'linux'...
fatal: unable to connect to github.com:
github.com[0: 4.208.26.197]: errno=Connection timed out

VVAQARAHMED, 2024-05-09:
Ping response for 
ping github.com
PING github.com (140.82.121.4) 56(84) bytes of data.
64 bytes from lb-140-82-121-4-fra.github.com (140.82.121.4): icmp_seq=1 ttl=51 time=35.9 ms
64 bytes from lb-140-82-121-4-fra.github.com (140.82.121.4): icmp_seq=2 ttl=51 time=31.6 ms
64 bytes from lb-140-82-121-4-fra.github.com (140.82.121.4): icmp_seq=3 ttl=51 time=29.2 ms
64 bytes from lb-140-82-121-4-fra.github.com (140.82.121.4): icmp_seq=4 ttl=51 time=34.2 ms
64 bytes from lb-140-82-121-4-fra.github.com (140.82.121.4): icmp_seq=5 ttl=51 time=26.9 ms

VVAQARAHMED, 2024-05-09:
I try to clone locally pc and successfully cloned git clone https://github.com/raspberrypi/linux.git
However i am trying build  bitbake core-image-base, i am getting the same. they error appear at 74%.

WARNING: linux-raspberrypi-1_6.6.22+git-r0 do_fetch: Failed to fetch URL git://github.com/raspberrypi/linux.git;name=machine;branch=rpi-6.6.y;protocol=https, attempting MIRRORS if available
ERROR: linux-raspberrypi-1_6.6.22+git-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/1001/bus"; export PATH="/home/wahmed/Yocto-training/rpi-build/tmp/sysroots-uninative/x86_64-linux/usr/bin:/home/wahmed/Yocto-training/source/poky/scripts:/home/wahmed/Yocto-training/rpi-build/tmp/work/raspberrypi3-poky-linux-gnueabi/linux-raspberrypi/6.6.22+git/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi:/home/wahmed/Yocto-training/rpi-build/tmp/work/raspberrypi3-poky-linux-gnueabi/linux-raspberrypi/6.6.22+git/recipe-sysroot/usr/bin/crossscripts:/home/wahmed/Yocto-training/rpi-build/tmp/work/raspberrypi3-poky-linux-gnueabi/linux-raspberrypi/6.6.22+git/recipe-sysroot-native/usr/sbin:/home/wahmed/Yocto-training/rpi-build/tmp/work/raspberrypi3-poky-linux-gnueabi/linux-raspberrypi/6.6.22+git/recipe-sysroot-native/usr/bin:/home/wahmed/Yocto-training/rpi-build/tmp/work/raspberrypi3-poky-linux-gnueabi/linux-raspberrypi/6.6.22+git/recipe-sysroot-native/sbin:/home/wahmed/Yocto-training/rpi-build/tmp/work/raspberrypi3-poky-linux-gnueabi/linux-raspberrypi/6.6.22+git/recipe-sysroot-native/bin:/home/wahmed/Yocto-training/source/poky/bitbake/bin:/home/wahmed/Yocto-training/rpi-build/tmp/hosttools"; export HOME="/home/wahmed"; LANG=C git -c gc.autoDetach=false -c core.pager=cat -c safe.bareRepository=all clone --bare --mirror https://github.com/raspberrypi/linux.git /home/wahmed/Yocto-training/rpi-build/downloads/git2/github.com.raspberrypi.linux.git --progress failed with exit code 128, see logfile for output
ERROR: linux-raspberrypi-1_6.6.22+git-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'git://github.com/raspberrypi/linux.git;name=machine;branch=rpi-6.6.y;protocol=https')
ERROR: Logfile of failure stored in: /home/wahmed/Yocto-training/rpi-build/tmp/work/raspberrypi3-poky-linux-gnueabi/linux-raspberrypi/6.6.22+git/temp/log.do_fetch.1416696
ERROR: Task (/home/wahmed/Yocto-training/source/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_6.6.bb:do_fetch) failed with exit code '1'

VVAQARAHMED, 2024-05-15:
Hi,
The issue is due to bitbake sever that failed some time after alot of try iI am able to build image successfully.

closed 2024-05-15 by VVAQARAHMED