agherzan/meta-raspberrypi #1324: scarthgap: wlan0 not enabled (for raspberry pi zero 2w)

alexodus, 2024-05-20:
**kirkstone**: with minimal configuration and bitbake core-image-minimal or bitbake core-image-base I've:
ifconfig -a : **localhost** and **wlan0** enabled

**scarthgap**: with minimal configuration and bitbake core-image-minimal or bitbake core-image-base I've:
ifconfig -a : only **localhost** enabled

matthiasklein, 2024-05-29:
I have the same problem.

Hardware: Raspberry Pi Zero 2 W Rev 1.0
Yocto machine: raspberrypi0-2w-64

Under kirkstone I have a wlan0 device, under scarthgap not anymore.

With kirkstone kernel v5.15 I got the following kernel messages:
```
dmesg | grep brc
[    7.526423] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436-sdio for chip BCM43430/2
[    7.527203] usbcore: registered new interface driver brcmfmac
[    7.544774] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43436-sdio.raspberrypi,model-zero-2-w.bin failed with error -2
[    7.700931] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436-sdio for chip BCM43430/2
[    7.710145] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436-sdio for chip BCM43430/2
[    7.749189] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/2 wl0: Oct  9 2020 14:44:32 version 9.88.4.65 (test) (f149b32@shgit)  (r679549) FWID 01-f40f3270
[   13.829930] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled

```
With scartgap kernel v6.6.22-v8 I get the following messages:
```
dmesg | grep brc
[    8.418991] Bluetooth: hci0: BCM43430B0 'brcm/BCM43430B0.hcd' Patch
[    8.429626] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430b0-sdio for chip BCM43430/2
[    8.438926] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430b0-sdio.raspberrypi,model-zero-2-w.bin failed with error -2
[    8.439612] usbcore: registered new interface driver brcmfmac
[    8.451595] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430b0-sdio.bin failed with error -2
[    9.471299] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
```

Does anyone have an idea what the problem could be?

kraj, 2024-05-29:
do you have files with "brcmfmac43430b0-sdio" in their name under /lib/firmware directory on rootfs

matthiasklein, 2024-05-29:
No, I only have the following files:

```
root@raspberrypi0-2w-64-ocean:~# ls -l  /lib/firmware/brcm/*sdio*
lrwxrwxrwx    1 root     root            31 Mar  9  2018 /lib/firmware/brcm/brcmfmac43430-sdio.bin -> ../cypress/cyfmac43430-sdio.bin
lrwxrwxrwx    1 root     root            36 Mar  9  2018 /lib/firmware/brcm/brcmfmac43430-sdio.clm_blob -> ../cypress/cyfmac43430-sdio.clm_blob
lrwxrwxrwx    1 root     root            31 Mar  9  2018 /lib/firmware/brcm/brcmfmac43430-sdio.raspberrypi,3-model-b.bin -> ../cypress/cyfmac43430-sdio.bin
lrwxrwxrwx    1 root     root            36 Mar  9  2018 /lib/firmware/brcm/brcmfmac43430-sdio.raspberrypi,3-model-b.clm_blob -> ../cypress/cyfmac43430-sdio.clm_blob
lrwxrwxrwx    1 root     root            22 Mar  9  2018 /lib/firmware/brcm/brcmfmac43430-sdio.raspberrypi,3-model-b.txt -> brcmfmac43430-sdio.txt
lrwxrwxrwx    1 root     root            23 Mar  9  2018 /lib/firmware/brcm/brcmfmac43430-sdio.raspberrypi,model-zero-2-w.bin -> brcmfmac43436s-sdio.bin
lrwxrwxrwx    1 root     root            23 Mar  9  2018 /lib/firmware/brcm/brcmfmac43430-sdio.raspberrypi,model-zero-2-w.txt -> brcmfmac43436s-sdio.txt
lrwxrwxrwx    1 root     root            31 Mar  9  2018 /lib/firmware/brcm/brcmfmac43430-sdio.raspberrypi,model-zero-w.bin -> ../cypress/cyfmac43430-sdio.bin
lrwxrwxrwx    1 root     root            36 Mar  9  2018 /lib/firmware/brcm/brcmfmac43430-sdio.raspberrypi,model-zero-w.clm_blob -> ../cypress/cyfmac43430-sdio.clm_blob
lrwxrwxrwx    1 root     root            22 Mar  9  2018 /lib/firmware/brcm/brcmfmac43430-sdio.raspberrypi,model-zero-w.txt -> brcmfmac43430-sdio.txt
-rw-r--r--    1 root     root          1121 Mar  9  2018 /lib/firmware/brcm/brcmfmac43430-sdio.txt
-rw-r--r--    1 root     root        416101 Mar  9  2018 /lib/firmware/brcm/brcmfmac43436-sdio.bin
-rw-r--r--    1 root     root         11209 Mar  9  2018 /lib/firmware/brcm/brcmfmac43436-sdio.clm_blob
lrwxrwxrwx    1 root     root            22 Mar  9  2018 /lib/firmware/brcm/brcmfmac43436-sdio.raspberrypi,model-zero-2-w.bin -> brcmfmac43436-sdio.bin
lrwxrwxrwx    1 root     root            27 Mar  9  2018 /lib/firmware/brcm/brcmfmac43436-sdio.raspberrypi,model-zero-2-w.clm_blob -> brcmfmac43436-sdio.clm_blob
lrwxrwxrwx    1 root     root            22 Mar  9  2018 /lib/firmware/brcm/brcmfmac43436-sdio.raspberrypi,model-zero-2-w.txt -> brcmfmac43436-sdio.txt
-rw-r--r--    1 root     root          1706 Mar  9  2018 /lib/firmware/brcm/brcmfmac43436-sdio.txt
-rw-r--r--    1 root     root        442211 Mar  9  2018 /lib/firmware/brcm/brcmfmac43436s-sdio.bin
lrwxrwxrwx    1 root     root            23 Mar  9  2018 /lib/firmware/brcm/brcmfmac43436s-sdio.raspberrypi,model-zero-2-w.bin -> brcmfmac43436s-sdio.bin
lrwxrwxrwx    1 root     root            23 Mar  9  2018 /lib/firmware/brcm/brcmfmac43436s-sdio.raspberrypi,model-zero-2-w.txt -> brcmfmac43436s-sdio.txt
-rw-r--r--    1 root     root          1185 Mar  9  2018 /lib/firmware/brcm/brcmfmac43436s-sdio.txt
lrwxrwxrwx    1 root     root            40 Mar  9  2018 /lib/firmware/brcm/brcmfmac43455-sdio.bin -> ../cypress/cyfmac43455-sdio-standard.bin
lrwxrwxrwx    1 root     root            36 Mar  9  2018 /lib/firmware/brcm/brcmfmac43455-sdio.clm_blob -> ../cypress/cyfmac43455-sdio.clm_blob
lrwxrwxrwx    1 root     root            31 Mar  9  2018 /lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,3-model-a-plus.bin -> ../cypress/cyfmac43455-sdio.bin
lrwxrwxrwx    1 root     root            36 Mar  9  2018 /lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,3-model-a-plus.clm_blob -> ../cypress/cyfmac43455-sdio.clm_blob
lrwxrwxrwx    1 root     root            22 Mar  9  2018 /lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,3-model-a-plus.txt -> brcmfmac43455-sdio.txt
lrwxrwxrwx    1 root     root            31 Mar  9  2018 /lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,3-model-b-plus.bin -> ../cypress/cyfmac43455-sdio.bin
lrwxrwxrwx    1 root     root            36 Mar  9  2018 /lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,3-model-b-plus.clm_blob -> ../cypress/cyfmac43455-sdio.clm_blob
lrwxrwxrwx    1 root     root            22 Mar  9  2018 /lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,3-model-b-plus.txt -> brcmfmac43455-sdio.txt
lrwxrwxrwx    1 root     root            31 Mar  9  2018 /lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,4-compute-module.bin -> ../cypress/cyfmac43455-sdio.bin
lrwxrwxrwx    1 root     root            36 Mar  9  2018 /lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,4-compute-module.clm_blob -> ../cypress/cyfmac43455-sdio.clm_blob
lrwxrwxrwx    1 root     root            22 Mar  9  2018 /lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,4-compute-module.txt -> brcmfmac43455-sdio.txt
lrwxrwxrwx    1 root     root            31 Mar  9  2018 /lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.bin -> ../cypress/cyfmac43455-sdio.bin
lrwxrwxrwx    1 root     root            36 Mar  9  2018 /lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.clm_blob -> ../cypress/cyfmac43455-sdio.clm_blob
lrwxrwxrwx    1 root     root            22 Mar  9  2018 /lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt -> brcmfmac43455-sdio.txt
lrwxrwxrwx    1 root     root            31 Mar  9  2018 /lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,5-model-b.bin -> ../cypress/cyfmac43455-sdio.bin
lrwxrwxrwx    1 root     root            36 Mar  9  2018 /lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,5-model-b.clm_blob -> ../cypress/cyfmac43455-sdio.clm_blob
lrwxrwxrwx    1 root     root            22 Mar  9  2018 /lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,5-model-b.txt -> brcmfmac43455-sdio.txt
-rw-r--r--    1 root     root          2074 Mar  9  2018 /lib/firmware/brcm/brcmfmac43455-sdio.txt

```

kraj, 2024-05-29:
is that same with working `kirkstone` version ? and secondly are you using systemd ? thirdly whats the output of `ifconfig -a` or `ip a`

matthiasklein, 2024-05-29:
Yes, I am using systemd (in kirkstone and scarthgap).

The firmware folder in kirkstone:
```
root@raspberrypi0-2w-64-ocean:~# ls -l  /lib/firmware/brcm/*sdio*
lrwxrwxrwx    1 root     root            31 Mar  9  2018 /lib/firmware/brcm/brcmfmac43430-sdio.bin -> ../cypress/cyfmac43430-sdio.bin
lrwxrwxrwx    1 root     root            36 Mar  9  2018 /lib/firmware/brcm/brcmfmac43430-sdio.clm_blob -> ../cypress/cyfmac43430-sdio.clm_blob
lrwxrwxrwx    1 root     root            22 Mar  9  2018 /lib/firmware/brcm/brcmfmac43430-sdio.raspberrypi,3-model-b.txt -> brcmfmac43430-sdio.txt
lrwxrwxrwx    1 root     root            22 Mar  9  2018 /lib/firmware/brcm/brcmfmac43430-sdio.raspberrypi,model-zero-w.txt -> brcmfmac43430-sdio.txt
-rw-r--r--    1 root     root          1121 Mar  9  2018 /lib/firmware/brcm/brcmfmac43430-sdio.txt
-rw-r--r--    1 root     root        413934 Mar  9  2018 /lib/firmware/brcm/brcmfmac43436-sdio.bin
-rw-r--r--    1 root     root         11209 Mar  9  2018 /lib/firmware/brcm/brcmfmac43436-sdio.clm_blob
-rw-r--r--    1 root     root          1613 Mar  9  2018 /lib/firmware/brcm/brcmfmac43436-sdio.txt
-rw-r--r--    1 root     root        428103 Mar  9  2018 /lib/firmware/brcm/brcmfmac43436s-sdio.bin
-rw-r--r--    1 root     root          1092 Mar  9  2018 /lib/firmware/brcm/brcmfmac43436s-sdio.txt
lrwxrwxrwx    1 root     root            31 Mar  9  2018 /lib/firmware/brcm/brcmfmac43455-sdio.bin -> ../cypress/cyfmac43455-sdio.bin
lrwxrwxrwx    1 root     root            36 Mar  9  2018 /lib/firmware/brcm/brcmfmac43455-sdio.clm_blob -> ../cypress/cyfmac43455-sdio.clm_blob
lrwxrwxrwx    1 root     root            22 Mar  9  2018 /lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,3-model-a-plus.txt -> brcmfmac43455-sdio.txt
lrwxrwxrwx    1 root     root            22 Mar  9  2018 /lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,3-model-b-plus.txt -> brcmfmac43455-sdio.txt
lrwxrwxrwx    1 root     root            22 Mar  9  2018 /lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,4-compute-module.txt -> brcmfmac43455-sdio.txt
lrwxrwxrwx    1 root     root            22 Mar  9  2018 /lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt -> brcmfmac43455-sdio.txt
-rw-r--r--    1 root     root          2074 Mar  9  2018 /lib/firmware/brcm/brcmfmac43455-sdio.txt
```

ifconfig (kirkstone):
```
root@raspberrypi0-2w-64-ocean:~# ifconfig -a
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 172.16.32.40  netmask 255.255.255.0  broadcast 0.0.0.0
        inet6 2a02:908:4c15:3f80:21b:c5ff:fe05:8fff  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::21b:c5ff:fe05:8fff  prefixlen 64  scopeid 0x20<link>
        ether 00:1b:c5:05:8f:ff  txqueuelen 1000  (Ethernet)
        RX packets 218  bytes 20123 (19.6 KiB)
        RX errors 0  dropped 109  overruns 0  frame 0
        TX packets 7  bytes 630 (630.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=4169<UP,LOOPBACK,RUNNING,MULTICAST>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 59  bytes 4087 (3.9 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 59  bytes 4087 (3.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether e4:5f:01:53:97:3f  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

```

ifconfig (scarthgap):
```
root@raspberrypi0-2w-64-ocean:~# ifconfig -a
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.101.143  netmask 255.255.255.0  broadcast 0.0.0.0
        inet6 fe80::21b:c5ff:fe05:8fff  prefixlen 64  scopeid 0x20<link>
        inet6 2a02:908:4c15:3f80:21b:c5ff:fe05:8fff  prefixlen 64  scopeid 0x0<global>
        ether 00:1b:c5:05:8f:ff  txqueuelen 1000  (Ethernet)
        RX packets 66  bytes 7021 (6.8 KiB)
        RX errors 0  dropped 28  overruns 0  frame 0
        TX packets 9  bytes 1322 (1.2 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=4169<UP,LOOPBACK,RUNNING,MULTICAST>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 38  bytes 2817 (2.7 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 38  bytes 2817 (2.7 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

```

closed 2024-06-04 by agherzan