agherzan/meta-raspberrypi #1327: udev-rules-rpi error during build

ste1maria, 2024-05-23:
When I am building and image and/or the udev recipe, I get the following error:
`bitbake core-image-base
NOTE: Reconnecting to bitbake server...
NOTE: Retrying server connection (#1)... (14:55:57.732232)
Loading cache: 100% |#######################################################################################################################################################################| Time: 0:00:00
Loaded 1917 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "2.9.1"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "raspberrypi5"
DISTRO               = "poky"
DISTRO_VERSION       = "5.0+snapshot-<unknown>"
TUNE_FEATURES        = "aarch64 crypto cortexa76"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "<unknown>:<unknown>"
meta-raspberrypi     = "master:1879cb831f4ea3e532cb5ce9fa0f32be917e8fa3"

Sstate summary: Wanted 1835 Local 0 Mirrors 0 Missed 1835 Current 710 (0% match, 27% complete)#############################################################################                 | ETA:  0:00:00
Initialising tasks: 100% |##################################################################################################################################################################| Time: 0:00:02
NOTE: Executing Tasks
ERROR: udev-rules-rpi-1.0-r0 do_install: ExecutionError('/home/user/Desktop/poky/rpi-build/tmp/work/cortexa76-poky-linux/udev-rules-rpi/1.0/temp/run.do_install.49054', 1, None, None)
ERROR: Logfile of failure stored in: /home/user/Desktop/poky/rpi-build/tmp/work/cortexa76-poky-linux/udev-rules-rpi/1.0/temp/log.do_install.49054
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/home/user/Desktop/poky/meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populate_sysroot', 'virtual:native:/home/user/Desktop/poky/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', 'virtual:native:/home/user/Desktop/poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['gettext-minimal-native', 'libtool-native', 'quilt-native', 'texinfo-dummy-native', 'patch-native', 'pseudo-native', 'xz-native', 'attr-native', 'sqlite3-native']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_install
| install: cannot stat '/home/user/Desktop/poky/rpi-build/tmp/work/cortexa76-poky-linux/udev-rules-rpi/1.0/can.rules': No such file or directory
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/user/Desktop/poky/meta-raspberrypi/recipes-core/udev/udev-rules-rpi.bb:do_install) failed with exit code '1'
`

Contents of the folder:
`~/Desktop/poky/meta-raspberrypi/recipes-core/udev$ ls
udev-rules-rpi     udev-rules-udisks-rpi
udev-rules-rpi.bb  udev-rules-udisks-rpi_1.0.bb
`
I have tried to rebuild the recipe, but get the same error.

ste1maria, 2024-05-23:
contents of the file udev-rules-rpi.bb:
```

DESCRIPTION = "udev rules for Raspberry Pi Boards"
LICENSE = "MIT"
LIC_FILES_CHKSUM = "file://${COMMON_LICENSE_DIR}/MIT;md5=0835ade698e0bcf8506ecda2f7b4f302"

SRC_URI = " \
	git://github.com/RPi-Distro/raspberrypi-sys-mods;protocol=https;branch=master \
	file://can.rules \
	"
SRCREV = "5ce3ef2b7f377c23fea440ca9df0e30f3f8447cf"

S = "${WORKDIR}/git"

INHIBIT_DEFAULT_DEPS = "1"

do_install () {
    install -d ${D}${sysconfdir}/udev/rules.d
    install -m 0644 ${S}/etc.armhf/udev/rules.d/99-com.rules ${D}${sysconfdir}/udev/rules.d/
    install -m 0644 ${WORKDIR}/can.rules ${D}${sysconfdir}/udev/rules.d/
}
```

kraj, 2024-05-23:
Please cherry pick the patch from https://github.com/agherzan/meta-raspberrypi/pull/1326

agherzan, 2024-06-03:
Should be fixed in master.

closed 2024-06-03 by agherzan