agherzan/meta-raspberrypi #1328: No way to modify existing configuration present in config.txt

adabool, 2024-05-23:
**Description**
No way to modify existing configuration present in config.txt

Currently in order to add any extra configuration in config.txt one can use RPI_EXTRA_CONFIG.
But there is no existing way to modify or remove any existing configuration present in config.txt.
If unknowingly done by someone can cause serious undefined behavior (like no bootup) with the RPI device. 
**Steps to reproduce the issue:**
1. Add following in your local.conf or layer.conf file:

RPI_EXTRA_CONFIG = ' \n \
    dpms_timeout=0 \n \
	dtoverlay=vc4-fkms-v3d \n \
	hdmi_force_hotplug=1 \n \
	config_hdmi_boost=10 \n \
	hdmi_group=2 \n \
	hdmi_mode=87 \n \
	hdmi_cvt 1024 600 60 6 0 0 0 \n \
    '

**Describe the results you received:**
Device is not booting as there is already **dtoverlay=vc4-kms-v3d** defined.

**Describe the results you expected:**
Device should work properly

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

farias-automac, 2024-05-29:
Maybe you can overwrite rpi-config_git.bb file

adabool, 2024-05-30:
@farias-automac thanks for this suggestion, do you mean by changing VC4DTBO ?= "vc4-kms-v3d" to VC4DTBO ?= "vc4-fkms-v3d" in rpi-config_git.bb file ?
Though this should work, isn't this a hacky of doing this ?

farias-automac, 2024-05-30:
Yes, that's what I was referring to. Otherwise I think you can create a bbappend file so you don't have to edit the original recipe directly. (although I'm not sure it works in this case, which is a do_deploy function)

Sorry my knowledge is limited, I'm a newbie. And this layer seems to not have much community support. Good luck !

adabool, 2024-05-30:
I think we should have a support to remove configuration parameters defined in config.txt similar to what we have for adding (RPI_EXTRA_CONFIG).
That will be much better and generic approach to do so!

AHSauge, 2024-12-03:
> Yes, that's what I was referring to. Otherwise I think you can create a bbappend file so you don't have to edit the original recipe directly. (although I'm not sure it works in this case, which is a [...]

`.bbappend` is probably the most appropriate solution here. I needed to add some extra config to a RPI target at some point, and all I did was create `recipes-bsp/bootfiles/rpi-config_git.bbappend` which contained

```
do_deploy:append:raspberrypi4-64() {
    CONFIG=${DEPLOYDIR}/${BOOTFILES_DIR_NAME}/config.txt

    #Here you can echo in which ever settings you like that's not covered by rpi-config_git.bb
    echo "foobar=value" >>$CONFIG
}
```

> I think we should have a support to remove configuration parameters defined in config.txt similar to what we have for adding (RPI_EXTRA_CONFIG). That will be much better and generic approach to do so!

Maybe this has improved since then, but the current `rpi-config_git.bb` seems pretty configurable. It doesn't seem to place or uncomment any settings in `config.txt` without something enabling it, so I guess the question here is what you expect.

In your case, you want to override a default value, so in your `rpi-config_git.bbappend`, you want something like `VC4DTBO = "vc4-fkms-v3d"`. Note the use of `=` over `?=` since the latter is just assigning a default value (see https://docs.yoctoproject.org/bitbake/bitbake-user-manual/bitbake-user-manual-metadata.html#setting-a-default-value)