agherzan/meta-raspberrypi #1330: pi5 yocto image, my own psplash pictiure file Not display

gitgonewithwind, 2024-05-29:
Hi all,

I modified meta-raspberrypi/recipes-core/psplash/psplash_%.bbappend almsot same as in pi4  as bebow

FILESEXTRAPATHS:prepend := "${THISDIR}/files:"
SPLASH_IMAGES:rpi = "file://flash_1024_576.png;outsuffix=raspberrypi"

or
FILESEXTRAPATHS:prepend := "${THISDIR}/files:"
SPLASH_IMAGES:rpi = "file://flash_1024_576.png;outsuffix=raspberrypi"

SRC_URI:append:rpi = " file://framebuf.conf"

do_install:append:rpi() {
    if [ "${@bb.utils.filter('DISTRO_FEATURES', 'systemd', d)}" ]; then
        install -Dm 0644 ${WORKDIR}/framebuf.conf ${D}${systemd_system_unitdir}/psplash-start.service.d/framebuf.conf
    fi
}

and DISABLE_SPLASH = "1" in local.con same as pi4 
and dwc_otg.lpm_enable=0 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait silent quiet  splash loglevel=0 logo.nologo
in /boot/cmdline.txt  same as pi4 
but  

but  my own picture flash_1024_576.png No display. same settings in pi4 image work with older commit

any help?

farias-automac, 2024-05-30:
Hi, I have the same problem, it works fine if I use sysvinit but after switching to systemd no more psplash is shown on boot.

gitgonewithwind, 2024-05-30:
@farias-automac 
I saw here https://github.com/agherzan/meta-raspberrypi/issues/1313
I just switched to systemd and built. but if I switch to systemd , I will change my APP startup script to service. I want to keep the older one. I am still building the image

gitgonewithwind, 2024-05-30:
Not work for me 
using solution https://github.com/agherzan/meta-raspberrypi/issues/1313
Include psplash: IMAGE_INSTALL:append = " psplash"
Use systemd: INIT_MANAGER = "system"

after using   systemd , X windows  completely NO display, no  psplash, no serial  console,
quit to  command login with keyboard

so I have not  found the solutions, so I have to restore to NO systemd 

any help?

gitgonewithwind, 2024-05-30:
root@raspberrypi5:/var/log# grep -r plash .
./boot:Thu May 30 12:42:50 2024: /etc/rc5.d/S00psplash.sh: line 39: /usr/bin/psplash: No such file or directory

no executable file   /usr/bin/psplash generated in image.

gitgonewithwind, 2024-05-31:
> Hi, I have the same problem, it works fine if I use sysvinit but after switching to systemd no more psplash is shown on boot.

you mean you use meta-respberry default .bbapend and sysvinit. it will show raspberry ?
but I can not. there is no /usr/bin/psplash generated.

I remember in pi4 there is a /usr/bin/psplash. I directly run it ,it can show in pi4. in yocto, it generated a executable file  psplash-write. no psplash , I do not know how to use psplash-write and search it why no psplash.

now I try to use psplash source code in pi4 board to directly compile it to try

gitgonewithwind, 2024-05-31:
I found  build/tmp/work/cortexa76-poky-linux/psplash/0.1+git/build/psplash, it was not put into image.
let me copy in pi5 board to run

gitgonewithwind, 2024-05-31:
I find the reason,  for my settingssame  as pi4 , psplash actually generated in build/tmp/work/cortexa76-poky-linux/psplash/0.1+git/build, but  it was not put into image rootfs

gitgonewithwind, 2024-05-31:
finally, I find  poky/meta/recipes-core/psplash/psplash_git.bb , there is a line :   rm -f ${D}${bindir}/psplash 
I not am sure why remove here.  pi4 (older commit pork has the same  line ,but pi4 works). I commented it, it works. the psplash was put into image rootfs

gitgonewithwind, 2024-05-31:
solved!

finally fixed,  I  add 1 line 
   install -m 0755 ${B}/psplash ${D}${bindir}/psplash
to install psplash again in
 file meta-raspberrypi/recipes-core/psplash/psplash_%.bbappend , the file is changed to:

FILESEXTRAPATHS:prepend := "${THISDIR}/files:"
SPLASH_IMAGES:rpi = "file://flash_1024_576.png;outsuffix=raspberrypi"

SRC_URI:append:rpi = " file://framebuf.conf"

do_install:append:rpi() {
    if [ "${@bb.utils.filter('DISTRO_FEATURES', 'systemd', d)}" ]; then
        install -Dm 0644 ${WORKDIR}/framebuf.conf ${D}${systemd_system_unitdir}/psplash-start.service.d/framebuf.conf
    fi
    install -m 0755 ${B}/psplash ${D}${bindir}/psplash.   
}

FILES:${PN}:append:rpi = " ${systemd_system_unitdir}/psplash-start.service.d"

farias-automac, 2024-05-31:
> finally, I find poky/meta/recipes-core/psplash/psplash_git.bb , there is a line : rm -f ${D}${bindir}/psplash I not am sure why remove here. pi4 (older commit pork has the same line ,but pi4 works). I commented it, it works. the psplash was put into image rootfs

Out of curiosity, what branch are you using?

gitgonewithwind, 2024-05-31:
@farias-automac 
master branch

meta-raspberrypi

commit 1879cb831f4ea3e532cb5ce9fa0f32be917e8fa3 (HEAD -> master, origin/master, origin/HEAD)
Author: Max Stepanov <mast@imast.ru>
Date:   Thu Mar 21 13:50:55 2024 -0400

closed 2024-06-03 by agherzan

azsde, 2025-08-26:
> solved!
> 
> finally fixed, I add 1 line install -m 0755 ${B}/psplash ${D}${bindir}/psplash to install psplash again in file meta-raspberrypi/recipes-core/psplash/psplash_%.bbappend , the file is changed to:
> 
> FILESEXTRAPATHS:prepend := "${THISDIR}/files:" SPLASH_IMAGES:rpi = "file://flash_1024_576.png;outsuffix=raspberrypi"
> 
> SRC_URI:append:rpi = " file://framebuf.conf"
> 
> do_install:append:rpi() { if [ "${@bb.utils.filter('DISTRO_FEATURES', 'systemd', d)}" ]; then install -Dm 0644 ${WORKDIR}/framebuf.conf ${D}${systemd_system_unitdir}/psplash-start.service.d/framebuf.conf fi install -m 0755 ${B}/psplash ${D}${bindir}/psplash. }
> 
> FILES:${PN}:append:rpi = " ${systemd_system_unitdir}/psplash-start.service.d"
> 
> DISABLE_SPLASH = "1" in local.conf do not matter if add it and do not add it. finally ,I remove this line

I don't understand what is going on with the missing psplash binary when adding meta-raspberrypi but I have the same issue with current scarthgap.