agherzan/meta-raspberrypi #1337: ffmpeg does not compile

nmaas87, 2024-06-09:
Trying to compile rpidisto-ffmpeg on the current master release of Yocto for CM4 does not work, build breaking errors:

| make: *** [/home/user/yocto_manual_cm4-2/build/tmp/work/cortexa72-poky-linux/rpidistro-ffmpeg/4.3.4/git/ffbuild/common.mak:59: libavdevice/opengl_enc.o] Error 1
| make: *** Waiting for unfinished jobs....
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/user/yocto_manual_cm4-2/sources/meta-raspberrypi/recipes-multimedia/rpidistro-ffmpeg/rpidistro-ffmpeg_4.3.4.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 6073 tasks of which 6066 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 2 seconds

Summary: 1 task failed:
  /home/user/yocto_manual_cm4-2/sources/meta-raspberrypi/recipes-multimedia/rpidistro-ffmpeg/rpidistro-ffmpeg_4.3.4.bb:do_compile
Summary: There was 1 WARNING message.
Summary: There were 2 ERROR messages, returning a non-zero exit code.

EasyIP2023, 2024-06-09:
@nmaas87

Thanks, been fairly distance from regularly checking rpidistro-ffmpeg & rpidistro-vlc.

Recommend using OE-core & meta-openembedded version of ffmpeg and vlc for the time being.

```sh
PREFERRED_PROVIDER_ffmpeg = "ffmpeg"
PREFERRED_PROVIDER_vlc = "vlc"
```

nmaas87, 2024-06-09:
Many thanks, that worked! 👍🏻

closed 2024-06-09 by nmaas87