agherzan/meta-raspberrypi #1352: how to build firefox in pi5 yocto image

gitgonewithwind, 2024-07-09:
Hi all,
sorry to bother you again.
before I used chromium-x11 in pi5 image. my docker build ENV works well.
 for some reasons, I want to build firefox.
but after I added  meta-browser/meta-firefox in  bblayers.conf 
and added (or not added ) fIMAGE_INSTALL_append = " firefox" in local.cong to build
there is an error output:
ERROR: The following required tools (as specified by HOSTTOOLS) appear to be unavailable in PATH, please install them in order to proceed:
  python python2.7

I added python 2.7 in Dockerfile (default python3 I think) , but it was output:
ERROR: The following required tools (as specified by HOSTTOOLS) appear to be unavailable in PATH, please install them in order to proceed:
  python

I did not how to add python ？？？？？  I added python in Dockerfile, but can not find the source

I use it  commit 1879cb831f4ea3e532cb5ce9fa0f32be917e8fa3 (HEAD -> master, origin/master, origin/HEAD)
Author: Max Stepanov <mast@imast.ru>
Date:   Thu Mar 21 13:50:55 2024 -0400

I what  to know how to build firefox or can I build firefox in yocto image?

gitgonewithwind, 2024-07-09:
I added RUN ln -s /usr/bin/python3 /usr/bin/python  in Dockerfile to solve the issue.
now It is build firefox, hope it can work well.

gitgonewithwind, 2024-07-10:
can not pass compiling firefox
8 times 
2:47.30 error: proc-macro derive panicked.  

..

2:49.16 error: could not compile `cssparser` (lib) due to 14 previous errors
..
ERROR: Task (/home/yocto/yocto/source/layers/meta-browser/meta-firefox/recipes-browser/firefox/firefox_68.9.0esr.bb:do_compile) failed with exit code '1'

any help?

gitgonewithwind, 2024-07-12:
I found the first error output:
1:02.34 /home/yocto/yocto/source/build/tmp/work/cortexa76-poky-linux/firefox/68.9.0esr/firefox-68.9.0/media/libopus/celt/arm/celt_neon_intr.c:208:18: error: call to undeclared function 'celt_inner_prod_neon'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]

it looks like  use wrong compiler C version?
how to fix ?
any idea?
Thanks

mihaistoica97, 2024-10-29:
Take a look at https://github.com/OldManYellsAtCloud/meta-browser if you haven't already found it

Original meta-browser repo hasn't updated Firefox in 4 years. Not sure on compatibility with RPI 5 though

gitgonewithwind, 2024-10-29:
@mihaistoica97  thanks. I did not use firefox. I still use chrome