agherzan/meta-raspberrypi #1354: how set pi5 hdmi output as pixel to pixel mode?

gitgonewithwind, 2024-07-15:
Hi all,
sorry to bother you again

I never use hdmi output as pixel to pixel mode, when our touchscreen is set to pixel to pixel mode,the hdmi output is not FULL display.
It seems I need to set pi5 hdmi as  pixel to pixel mode.

I just searched online , it seems I need to change dts  source code? it is right?
what is default output mode for hdmi?
any ideas ?

I am using commit 
meta-raspberrypi

commit 1879cb831f4ea3e532cb5ce9fa0f32be917e8fa3 (HEAD -> master, origin/master, origin/HEAD)
Author: Max Stepanov <mast@imast.ru>
Date:   Thu Mar 21 13:50:55 2024 -0400

Thank you so much.

kraj, 2024-07-15:
if say display is 800x480 then add something like below to `/boot/config.txt` on the SD card and reboot

```
hdmi_group=2
hdmi_mode=87
hdmi_cvt=800 480 60 6 0 0 0
hdmi_force_hotplug=1
```

hdmi_group=2: Sets the HDMI output group to DMT (Display Monitor Timings)
hdmi_mode=87: Enables a custom HDMI mode

closed 2024-07-16 by gitgonewithwind

gitgonewithwind, 2024-07-16:
@kraj 
sorry, I think I have to open this issue .
after I and my colleague tested. it is not useful to change config.txt.
it seems config.txt not work as expection

we only change   /etc/X11/xorg.conf.d/99-monitor.conf  will can fix it and keep config.txt as default.
Section "Monitor"
    Identifier "HDMI-1"
    Option "PreferredMode" "1920x1080"
EndSection

Section "Monitor"
    Identifier "HDMI-2"
    Option "PreferredMode" "1920x1080"
EndSection

refs to this issue:  https://github.com/agherzan/meta-raspberrypi/issues/1312   
and this  : https://github.com/agherzan/meta-raspberrypi/issues/1322

Alanfromgard, 2025-02-26:
for raspberry 5I  do 99-monitor.conf,

Alanfromgard, 2025-02-26:
sorry, rp5, i do all thinks  you say. Using image bitbake core-image-x11. X11 start, cursor appeared. But if X-terminal is a black screen. It is reactiv , if i enter reboot as a blind man, it reboot. xrandr give me good resolution. is i enter via ssh xclock it start, and i see cursor reacting, but no clock.  NO error on Xorg.log.

gitgonewithwind, 2025-02-26:
@Alanfromgard 

1. in /etc/X11/xorg.conf.d folder, create the first file :  10-vc4.conf

Section "OutputClass"
  Identifier "vc4"
  MatchDriver "vc4"
  Driver "modesetting"
  Option "PrimaryGPU" "true"
EndSection

2.  create  the second file: 99-monitor.conf
Section "Monitor"
    Identifier "HDMI-1"
    Option "PreferredMode" "1920x1080"
EndSection

Section "Monitor"
    Identifier "HDMI-2"
    Option "PreferredMode" "1920x1080"
EndSection

it works for me:
please try.