agherzan/meta-raspberrypi #1357: terrible GUI performance on Raspberry Pi 4 and Raspberry Pi 5

bartiszosti, 2024-08-08:
**Description**

I have terrible GUI performance on Raspberry Pi 4 and Raspberry Pi 5 with Wayland and Weston.
Everything lagging independently on screen resolution. I use screen connected by HDMI. Only fkms works.

**Steps to reproduce the issue:**

MACHINE = "raspberrypi4-64"
ENABLE_UART = "1"
DISABLE_SPLASH = "1"
DISABLE_RPI_BOOT_LOGO = "1"
VC4DTBO = "vc4-fkms-v3d"
IMAGE_FEATURES:append = " splash debug-tweaks"
IMAGE_INSTALL = "vim htop wayland weston weston-xwayland"
INIT_MANAGER = "systemd"
SYSTEMD_DEFAULT_TARGET = "graphical.target"

airjason13, 2024-08-23:
What's the version of Mesa?
I have hit the image lagging problem on Pi 5 due to the older version of Mesa.

joshturge, 2024-08-29:
I'm getting a similar issue using the vc4-kms-v3d driver, vc4-fkms-v3d works fine for me. Here's the dmesg output: [dmesg.txt](https://github.com/user-attachments/files/16790452/dmesg.txt)
You'll notice I'm getting a bunch of warnings related to the kms driver.
I'm using a Raspberry Pi Compute Module 4S

carpajr, 2024-10-12:
I'm also getting the same poor results with wayland.

bartiszosti, did you find a solution?