agherzan/meta-raspberrypi #1358: Access i2c bus from U-Boot

Spark1114, 2024-08-16:
Hi all, 

I am trying to access a TPM connected to the i2c bus 0 from U-boot without any success. I have a custom board with a CM4 and I am using Scarthgap release. The i2c bus used is the 0 connected to gpio0 and gpio1. The config.txt with the dedicated overlays seems correct because everything is working fine in Linux.

I have all the commands available but I always get the error -19 when trying to select bus 0. 

From the dm tree command I see that the i2c bus is not probed at all. 

I tried to modify in U-Boot the device tree which is specified in the rpi_arm64_defconfig file to enable the i2c and set the gpio0 and 1 without any success. I also tried to modify the default dtb provided in /boot by the kernel (bcm2711-rpi-cm4.dtb) and load it manually into u-boot. The fdt print i2c0 shows the node but I no bus are displayed when I do i2c dev or i2c bus. 

I wasn't able to find any ressources online on how to enable this i2c bus. I am sure a small thing is missing but I can't figure it out. 

Thanks a lot for your help.

Spark1114, 2024-08-20:
For anyone having this issue the problem is that u-boot does not have the bcm i2c driver. In order to use a i2c bus, one has to use the i2c-gpio driver.

tandresen77, 2025-10-20:
Can you please advice how to add the i2c-gpio driver? I could enable it using u-boot_%.bbappend but i have still an issue overwriting the device tree which is used so the i2c-bus isn't enabled.