agherzan/meta-raspberrypi #1362: Pulseaudio stack smashing error when starting

farias-automac, 2024-08-26:
Hello, I have an error when launching pulseaudio service. I'm building the image (32 bits) for a CM4, Scarthgap branch. It's a fresh test image with just Alsa, Pulseaudio, Systemd and Dbus. This is the error message:

`
D: [pulseaudio] alsa-util.c:   start_threshold  : -1
D: [pulseaudio] alsa-util.c:   stop_threshold   : 1073741824
D: [pulseaudio] alsa-util.c:   silence_threshold: 0
D: [pulseaudio] alsa-util.c:   silence_size : 0
D: [pulseaudio] alsa-util.c:   boundary     : 1073741824
D: [pulseaudio] alsa-util.c:   appl_ptr     : 0
D: [pulseaudio] alsa-util.c:   hw_ptr       : 0
D: [alsa-sink-MAI PCM i2s-hifi-0] alsa-sink.c: Thread starting up
D: [alsa-sink-MAI PCM i2s-hifi-0] util.c: SCHED_RR|SCHED_RESET_ON_FORK worked.
I: [alsa-sink-MAI PCM i2s-hifi-0] util.c: Successfully enabled SCHED_RR scheduling for thread, with priority 5.
D: [pulseaudio] sink.c: alsa_output.platform-fef00700.hdmi.hdmi-stereo: state: INIT -> IDLE
I: [alsa-sink-MAI PCM i2s-hifi-0] alsa-sink.c: Starting playback.
*** stack smashing detected ***: terminated
Aborted (core dumped)
root@raspberrypi4:~# stack smashing detected
`

Please help, I have been struggling with this for several days.

kraj, 2024-08-26:
Can you try `/etc/asound.conf` from say https://learn.adafruit.com/adafruit-max98357-i2s-class-d-mono-amp/raspberry-pi-usage#create-asound-dot-conf-file-3170879 and see if it helps ?

farias-automac, 2024-08-26:
Hi kraj, thanks for your reply

I tried with your suggested conf file, but without luck. I'm  getting the same error. 

It is worth clarifying that things were going well with the Kirkstone branch. But when I switched to scarthgap pulseaudio no longer wants to start.

kraj, 2024-08-26:
Perhaps try to compile pulseaudio with `-fsanitize=address` added to cflags and then run it. It should give move information during runtime about memory errors.

farias-automac, 2024-08-27:
Sorry, I don't know how to do that.

deadmasterog, 2024-10-12:
i had the same issue and worked through it recently. this issue is caused by alsa-lib and pulseaudio using different resolutions for the timestamp structs. in this case pulseaudio uses 32-bit timestamps while alsa-lib uses 64-bit timestamps. the stack corruption occurs when pulseaudio makes a call to alsa-lib to store a 64-bit timestamp into a 32-bit variable.

i encountered this issue with a yocto project i had setup on scarthgap. the issue has been fixed in yocto styhead and beyond.

here's the commit where pulseaudio's configuration was overridden to use 32-bit timestamps:
https://github.com/yoctoproject/poky/commit/e0ffea74cf3f4a492f440acc9e31cdbe7c805192

here's the commits where this was reverted:
https://github.com/yoctoproject/poky/commit/ba4dd8c51af81fbf87a5ecaa73c2c5b26b2fdeae
https://github.com/yoctoproject/poky/commit/763c6dab335968ac2367a9c4f372b1f64f50c212

the commits contain info on why this was done (and what changes were made to undo it). unfortunately this had the consequence of breaking compatibility between pulseaudio and alsa-lib.

my suggestion to fix this is to update to use the newer poky layer with these fixes, or manually apply the fixes shown in those commits above to configure pulseaudio for 64-bit timestamps and disable OSS support.

closed 2025-10-07 by farias-automac