agherzan/meta-raspberrypi #1364: RPI5 with Qt5 Webengine Webview Failure

sefram, 2024-09-05:
**Description**
Hello,
We try using RPI5 with Yocto and Qt5 with QtWebengine, QtWebview.
Building the Yocto image is ok, same for the SDK.
But once we launch a Qt example like "https://doc.qt.io/qt-5/qtwebengine-webengine-recipebrowser-example.html", we always get an errors when displaying Web:
FATAL:page_allocator_internals_posix.h(169) Check failed: . : Invalid argument (22)
FATAL:page_allocator_internals_posix.h(169) Check failed: . : Invalid argument (22)
FATAL:page_allocator_internals_posix.h(169) Check failed: . : Invalid argument (22)

**Steps to reproduce the issue:**
1. Build a X11 (Same behaviour with Wayland) image using Yocto with Qt and module webengine, webview.
2. Install the SDK generated on a PC
3. Try to compile and launch the example "Recipebrowser" (available in the example)
4. It lead to the error above.

**Additional details (revisions used, host distro, etc.):**
- Raspberry PI5
- Yocto Scarthgap
- Qt 5.15.8

Thanks in advance for your help.

kraj, 2024-09-05:
is there more information you can dig e.g. are there any crash reported or some sort of backtrace ?
we did have issue with rpi4 where it would assume hw crypto which was not part of ISA implemented on
the SOC and it would cause illegal instruction traps on qt6 but perhaps might happen on qt5 as well, depends on what version of chromium is used

https://codereview.qt-project.org/c/yocto/meta-qt6/+/511153

sefram, 2024-09-05:
> is there more information you can dig e.g. are there any crash reported or some sort of backtrace ? we did have issue with rpi4 where it would assume hw crypto which was not part of ISA implemented [...]

Hello,
This error occurs during execution of the example (not during the build).
How i can add trace to give you more details for the webengine plugin ?

learmj, 2024-09-23:
Hi. Qt5 seems incompatible with a 16kB page size.
```recipes-kernel/linux/linux-raspberrypi.inc``` will currently generate a 16kB page size kernel because of
```KBUILD_DEFCONFIG:raspberrypi5 ?= "bcm2712_defconfig"```
The error does not occur with a 4kB page size kernel (ie kernel8.img in Raspberry Pi OS terms).

learmj, 2024-09-23:
ref: https://github.com/raspberrypi/bookworm-feedback/issues/107

sefram, 2024-10-08:
Thanks a lot matthew for your help.