agherzan/meta-raspberrypi #1365: Scarthgap and Kernel 6.1 compilation error

farias-automac, 2024-09-11:
Hi, i switched from the default 6.6 kernel to 6.1, on Scarthgap branch, and it doesn't compile anymore. (core-base image)

CM4 board, 32bit image

Can someone test/confirm that there is a problem with that kernel version?

AHSauge, 2024-12-03:
I haven't tried a 32 bit build, but I regularly build 64 bit images for a CM4 board, and that is working out-of-the-box. If you still have issues compiling the kernel, I suggest sharing a log showing the error. That would make it easier for people to help you :)

UmonDaisuke, 2025-01-20:
Hello,

please fin below my current status when i try to build  a core-image-base from scrthgap with different kernel and architecture for a pi zero v2:
64 bit + kernel 6.6 : ok
32 bit + kernel 6.6 : ok
64 bit + kernel  6.1: ok
32 bit + kernel 6.1: fail

the problem is that the last one is my target and i m not able to find any workaround to "No rule to make target 'arch/arm/boot/dts/broadcom/bcm2708-rpi-zero.dtb"

please find below log of the error  that avoid image creation.
woul you have any  solution?
Kind regards
Michael

DEBUG: Executing shell function do_compile
NOTE: KBUILD_BUILD_TIMESTAMP: Thu Feb  8 15:07:10 UTC 2024
NOTE: make -j 16 CC=arm-poky-linux-gnueabi-gcc  -mno-thumb-interwork -marm -fuse-ld=bfd -fcanon-prefix-map  -fmacro-prefix-map=/home/dingo/poky/32/tmp/work-shared/raspberrypi0-2w/kernel-source=/usr/src/debug/linux-raspberrypi/6.1.77+git  -fdebug-prefix-map=/home/dingo/poky/32/tmp/work-shared/raspberrypi0-2w/kernel-source=/usr/src/debug/linux-raspberrypi/6.1.77+git  -fmacro-prefix-map=/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/linux-raspberrypi0_2w-standard-build=/usr/src/debug/linux-raspberrypi/6.1.77+git  -fdebug-prefix-map=/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/linux-raspberrypi0_2w-standard-build=/usr/src/debug/linux-raspberrypi/6.1.77+git  -fdebug-prefix-map=/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot=  -fmacro-prefix-map=/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot=  -fdebug-prefix-map=/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot-native=  -fdebug-prefix-map=/home/dingo/poky/32/tmp/work-shared/raspberrypi0-2w/kernel-source=/usr/src/kernel -fdebug-prefix-map=/home/dingo/poky/32/tmp/work-shared/raspberrypi0-2w/kernel-build-artifacts=/usr/src/kernel LD=arm-poky-linux-gnueabi-ld.bfd   OBJCOPY=arm-poky-linux-gnueabi-objcopy  STRIP=arm-poky-linux-gnueabi-strip HOSTCC=gcc  HOSTCFLAGS=-isystem/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot-native/usr/include -O2 -pipe HOSTLDFLAGS=-L/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot-native/usr/lib                         -L/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot-native/lib                         -Wl,-rpath,/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot-native/lib                         -Wl,-O1 -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/home/dingo/poky/32/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2 -pthread HOSTCPP=gcc  -E HOSTCXX=g++  HOSTCXXFLAGS=-isystem/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot-native/usr/include -O2 -pipe HOSTPKG_CONFIG=pkg-config-native PAHOLE=false -j 16 zImage LOADADDR=0x00008000
  SYNC    include/config/auto.conf
  GEN     Makefile
  GEN     Makefile
  CALL    /home/dingo/poky/32/tmp/work-shared/raspberrypi0-2w/kernel-source/scripts/checksyscalls.sh
  Kernel: arch/arm/boot/Image is ready
  Kernel: arch/arm/boot/zImage is ready
NOTE: make -j 16 CC=arm-poky-linux-gnueabi-gcc  -mno-thumb-interwork -marm -fuse-ld=bfd -fcanon-prefix-map  -fmacro-prefix-map=/home/dingo/poky/32/tmp/work-shared/raspberrypi0-2w/kernel-source=/usr/src/debug/linux-raspberrypi/6.1.77+git  -fdebug-prefix-map=/home/dingo/poky/32/tmp/work-shared/raspberrypi0-2w/kernel-source=/usr/src/debug/linux-raspberrypi/6.1.77+git  -fmacro-prefix-map=/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/linux-raspberrypi0_2w-standard-build=/usr/src/debug/linux-raspberrypi/6.1.77+git  -fdebug-prefix-map=/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/linux-raspberrypi0_2w-standard-build=/usr/src/debug/linux-raspberrypi/6.1.77+git  -fdebug-prefix-map=/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot=  -fmacro-prefix-map=/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot=  -fdebug-prefix-map=/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot-native=  -fdebug-prefix-map=/home/dingo/poky/32/tmp/work-shared/raspberrypi0-2w/kernel-source=/usr/src/kernel -fdebug-prefix-map=/home/dingo/poky/32/tmp/work-shared/raspberrypi0-2w/kernel-build-artifacts=/usr/src/kernel LD=arm-poky-linux-gnueabi-ld.bfd   OBJCOPY=arm-poky-linux-gnueabi-objcopy  STRIP=arm-poky-linux-gnueabi-strip HOSTCC=gcc  HOSTCFLAGS=-isystem/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-r [... cut ...] facts=/usr/src/kernel   LD=arm-poky-linux-gnueabi-ld.bfd   OBJCOPY=arm-poky-linux-gnueabi-objcopy  STRIP=arm-poky-linux-gnueabi-strip LOADADDR=0x00008000
  SYNC    include/config/auto.conf
  GEN     Makefile
make[1]: *** No rule to make target 'arch/arm/boot/dts/broadcom/bcm2708-rpi-zero.dtb'.  Stop.
make: *** [/home/dingo/poky/32/tmp/work-shared/raspberrypi0-2w/kernel-source/Makefile:1471: broadcom/bcm2708-rpi-zero.dtb] Error 2
ERROR: oe_runmake failed
WARNING: exit code 1 from a shell command.

AllianceTrooper, 2025-02-18:
Hello. I get the same result, 32bit image RPi4B, scarthgap, kernel 6.1.77. I made temporary workaround for my project by adding `linux-raspberrypi_6.1.bbappend` file with the next contents:

```
do_configure:append(){
    mkdir ${STAGING_KERNEL_DIR}/arch/arm/boot/dts/broadcom
    cp ${STAGING_KERNEL_DIR}/arch/arm/boot/dts/bcm* ${STAGING_KERNEL_DIR}/arch/arm/boot/dts/broadcom/
    cp ${STAGING_KERNEL_DIR}/arch/arm/boot/dts/rp1.dtsi ${STAGING_KERNEL_DIR}/arch/arm/boot/dts/broadcom/
}
```
This will work for broadcom devices only. I believe that the reason is different source code structure for 6.1 and 6.6. Kernel 6.6 has vendor-based directory hierarchy with `dts` files sorted inside while kernel 6.1 has all the `dts` files in one directory