agherzan/meta-raspberrypi #1368: Official RPi 7inch display sometimes goes white (or black) upon reboot

perezmeyer, 2024-09-20:
**Description**

RPi4 with official 7inch display, with our without HDMI output connected sometimes goes totally white (or black [1]) upon reboot, and can only be brought back to life by completely powering off the device.

[1] On another device the behaviour is the same as I'll describe, except the screen goes black, with backlight always on.

Display is connected to DSI with ribbon cable and powered using cables connected to the RPi's pins 4 and 6.

**Steps to reproduce the issue:**
1. Boot Yocto image, let Weston start.
2. Reboot
3. Colored square and initial framebuffer always appears.
4. Sometimes the DSI display starts ok, sometimes it becomes all white and can not be restored without a full power cycle.

The issue can not be reproduced with RaspberryPi OS.

`dmesg` shows no change between working and non-working boots.

**Describe the results you received:**

White-blocked display

**Describe the results you expected:**

Display showing Weston

**Additional information you deem important (e.g. issue happens only occasionally):**

Only rebooting. A quick power cycle might not be enough a solution.

**Additional details (revisions used, host distro, etc.):**

Currently using at scarthgap branch at commit 1918a27419dcd5e79954c0dc0edddcde91057a7e

`/boot/config.txt` contents:

```
root@rpi4:~# grep -v ^# /boot/config.txt | sed '/^[[:blank:]]*$/d'
hdmi_group=2
hdmi_mode=39
gpu_mem=256
dtparam=i2c1=on
dtparam=i2c_arm=on
dtoverlay=vc4-kms-v3d
dtoverlay=vc4-kms-dsi-7inch
dtoverlay=i2c-rtc,ds1307
``

drkhsh, 2024-09-22:
Same issue here, happens every few reboots. Pi 4 with official 7" inch display as well.

perezmeyer, 2024-09-23:
@agherzan if you happen to have any pointers to where I should look I'll give it a try