agherzan/meta-raspberrypi #1371: Raspberry Pi 7inch Display + KMS are displaying black/white bars

ykuzma1, 2024-09-24:
**Description**
The official Raspberry Pi 7 inch display consistently displays black and white bars after boot when using the KMS graphics driver. The HDMI display does show the correct output however. Pictures attached.

**Steps to reproduce the issue:**
`
MACHINE = "raspberrypi4"
DISTRO = "poky"
EXTRA_IMAGE_FEATURES = "debug-tweaks"
ENABLE_UART = "1"
IMAGE_INSTALL:append = " kmscube"
LICENSE_FLAGS_ACCEPTED += "synaptics-killswitch"
`

`
bitbake core-image-base
`

**Describe the results you received:**
The RPi 7in display does initially show the rainbow screen and four raspberry graphics, but then immediately fades out to black and white bars. The HDMI display shows the rainbow and raspberry graphics like the RPi display but boots to a visible console and can launch kmscube successfully.

**Describe the results you expected:**
console and kmscube working on the RPi 7in display.

**Additional information you deem important (e.g. issue happens only occasionally):**
Attempted using RPi OS - does work correctly!
Attempted using FKMS driver - does display correctly on 7inch display but does *not* respond to touch input.
Attempted using 32bit vs 64bit - no effect.
Attempted using Linux kernel 6.1 - no effect.
Attempted replacing touchscreen, Pi, cable, reseating everything - no effect.

**Additional details (revisions used, host distro, etc.):**
poky - scarthgap - latest
meta-raspberrypi - scarthgap - latest

Let me know if I can provide any logs or can attempt any other debugging steps because I'm running out of ideas.

[image]
[image]

ykuzma1, 2024-09-25:
I decided to try running on different models of the Pi this morning. The Raspberry Pi 3 and 4 both reflect the issue as described previously. The Raspberry Pi 5 is even worse though; it does not show anything on the RPi 7in display or even illuminate the backlight. (Trying again with RPi OS solves all issues on all hardware versions with the KMS driver and the RPi display.)

ykuzma1, 2024-09-27:
I just realized this may be a continuation of closed issue https://github.com/agherzan/meta-raspberrypi/issues/802. Let me know if you'd rather move this discussion there and reopen that issue.

bojanmilevski, 2024-10-24:
I am experiencing the exact same issue.

Upgrading kernel to 6.6.51 does not help - neither does downgrading to 6.1.77.

bojanmilevski, 2024-10-24:
After many hours of debugging, I have came to the conclusion that setting `INIT_MANAGER = "mdev-busybox"` was the root of the problem. Once I changed it to `systemd` and added `kernel-modules` to `IMAGE_INSTALL` in my image recipe, the problem was gone.

I installed `weston` and was surprised for it to auto-start.

I hope this information is relevant and helpful.

bojanmilevski, 2024-10-24:
It's important to note that I am on each layer's master branch (`poky` and `meta-raspberrypi`).

ykuzma1, 2024-10-24:
@bojanmilevski I was unable to get the problem resolved by adding `INIT_MANAGER = systemd` and `IMAGE_INSTALL:append = " kernel-modules". Would you mind sharing the rest of your `local.conf`?

bojanmilevski, 2024-10-24:
After having spent a couple more hours, I can confirm that adding `dtoverlay=vc4-kms-dsi-7inch` to `config.txt`, as well as the steps above, works.

Another note to mention - I tried switching back to the `scarthgap` branch on both layers, and can confirm that everything works as normal, so no need to switch to `master`

ykuzma1, 2024-10-24:
@bojanmilevski I can confirm that worked on my end as well! Thank you!!

Here is a minimal working example `local.conf`:

```
MACHINE = "raspberrypi4"
DISTRO = "poky"
EXTRA_IMAGE_FEATURES = "debug-tweaks"
ENABLE_UART = "1"
DISTRO_FEATURES += "opengl pam wayland"
INIT_MANAGER = "systemd"
IMAGE_INSTALL:append = " kmscube kernel-modules"
LICENSE_FLAGS_ACCEPTED += "synaptics-killswitch"
RPI_EXTRA_CONFIG = "dtoverlay=vc4-kms-dsi-7inch\n"
```

ykuzma1, 2024-10-24:
Initial testing on a Raspberry Pi 5 continues to not show anything on the RPi display or illuminate the backlight with both KMS and FKMS drivers. Which indicates that may be a different issue altogether. I will do more testing and follow-up with a new issue if necessary.

bojanmilevski, 2024-10-27:
Sorry, I forgot to mention that I was testing this on a RPi4B with 8GB of memory. I am unable to assist you with the RPi5.

ykuzma1, 2024-10-28:
No problem @bojanmilevski. I moved discussion regarding the RPi5 to its own issue tracker #1382. Thanks again for your help on this issue.