agherzan/meta-raspberrypi #1381: Rust support in the kernel

simon-rechermann, 2024-10-27:
**Description**

I would like to enable rust support in the kernel. My MACHINE = "raspberrypi5" and I'm on scarthgap, so my LINUX_VERSION ?= "6.6.22".
It seems like it's not possible to enable it as HAVE_RUST and RUST_IS_AVAILABLE are set to n.
However, I saw that the rust sources build/tmp/work-shared/raspberrypi5/kernel-source/rust are available.
Why is HAVE_RUST=n and RUST_IS_AVAILABLE=n? Is it not supported to use rust in the kernel for raspi5?
Is there a safe way to enable the rust support?

`bitbake linux-raspberrypi -c menuconfig`
```
.config - Linux/arm64 6.6.22 Kernel Configuration
> Search (rust)

┌──────────────────────────── Search Results ─────────────────────────────┐
│ Symbol: RUST [=n]                                                      │  
│ Type  : bool                                                           │  
│ Defined at: init/Kconfig:1893                                          │  
│ Prompt: Rust support                                                   │  
│ Depends on: HAVE_RUST [=n] && RUST_IS_AVAILABLE [=n] && !MODVERSIONS   │  
│ Location:                                                              │  
│    (1) -> General setup                                                │  
│        -> Rust support (RUST [=n])                                     │  
│ Selects: CONSTRUCTORS [=n]                                             │  
│                                                                        │  
│                                                                        │  
│ Symbol: HAVE_RUST [=n]                                                 │  
│ Type  : bool                                                           │  
│ Defined at: arch/Kconfig:374                                           │  
└────────────────────────────────────────────────────────────────────────┘
```

**Steps to reproduce the issue:**
1. Run `bitbake linux-raspberrypi -c menuconfig`
2. Search for "rust"

lsahn-gh, 2024-11-05:
Perhaps arm64 arch for kernel 6.6.22 may not support enabling rust yet. (not sure)
You can dive into HAVE_RUST [=n] why it is n.

simon-rechermann, 2024-11-09:
I found out that HAVE_RUST came into arm64 with version 6.9 https://github.com/raspberrypi/linux/blob/rpi-6.9.y/arch/arm64/Kconfig
Therefore I created a linux-raspberrypi_6.10.bb custom recipe and changed the kernel version like this:
PREFERRED_VERSION_linux-raspberrypi = "6.10%"
I took 6.10 instead of 6.9 because there was no LINUX_RPI_KMETA_BRANCH for 6.9.
It worked but now I found the next issue.
Here it is defined the required rustc version for the 6.10 kernel is 1.78.0 https://github.com/raspberrypi/linux/blob/rpi-6.10.y/scripts/min-tool-version.sh.
However, the the scarthgap poky/meta/recipes-devtools/rust (https://github.com/openembedded/openembedded-core/tree/scarthgap/meta/recipes-devtools/rust )contains rustc version 1.75.0. I guess that's why still RUST_IS_AVAILABLE [=n]
I home the next yocto LTS version will make my life a bit easier.
I guess one of my options is now to use a recipe that provides a newer rust toolchain which turns out to be not that easy as meta-rust seems to be not maintained anymore as it has been merged to openembedded-core https://github.com/meta-rust/meta-rust/issues/451. Is looks like https://git.yoctoproject.org/meta-lts-mixins/ might make this possible.
Another option might be to use a local rust toolchain and somehow provide this toolchain to the bitbake environment instead of using a rust yocto recipe. However i'm not sure now to configure the a rust toolchain on my own for cross compiling by setting the appropriate environment variables or whatever.
If anyone knows a simpler solution I would appreciate it