agherzan/meta-raspberrypi #1382: Raspberry Pi 5 + Raspberry Pi 7inch Display is not working

ykuzma1, 2024-10-28:
**Description**
The official Raspberry Pi 7 inch display does not show anything or illuminate the blacklight when building for and deploying on a Raspberry Pi 5. The HDMI display does show the correct output however. Pictures attached.

**Steps to reproduce the issue:**
```
MACHINE = "raspberrypi5"
DISTRO = "poky"
EXTRA_IMAGE_FEATURES = "debug-tweaks"
ENABLE_UART = "1"
IMAGE_INSTALL:append = " kmscube"
LICENSE_FLAGS_ACCEPTED += "synaptics-killswitch"
```
```
bitbake core-image-base
```

**Describe the results you received:**
The Raspberry Pi 5 never shows anything or illuminates the official RPi 7inch display including during the boot process.

**Describe the results you expected:**
During the boot process, the display should illuminate the backlight, show the rainbow screen, and then show four raspberry graphics. Then after boot, it should show the console and kmscube when executed.

**Additional information you deem important (e.g. issue happens only occasionally):**
- Attempted using RPi OS - does work correctly.
- Attempted using FKMS driver - no change.
- Attempted replacing touchscreen, Pi, cable, reseating everything - no change.
- Attempted RPi display independently of HDMI - no change.

**Additional details (revisions used, host distro, etc.):**
- poky - scarthgap/kirkstone - latest
- meta-raspberrypi - scarthgap/kirkstone - latest

[image]
[image]
[image]

leon-anavi, 2024-11-04:
@ykuzma1  what settings for VC4DTBO are you using? Have you tried with `vc4-kms-dsi-7inch`?

ykuzma1, 2024-11-05:
@leon-anavi I am using the default VC4DTBO value. But I have also tried setting `VC4DTBO = vc4-fkms-v3d` in local.conf with the same results.

Setting `dtoverlay=vc4-kms-dsi-7inch` in config.txt also has the same issue.