agherzan/meta-raspberrypi #1383: Raspberry pi 5 boot problems vc4 graphics and networking

davidlekve, 2024-10-29:
**Description**

The raspberry pi 5 does not boot properly when having enabled vc4graphics in combination with some meta-openembedded/meta-networking.

I have not had a succesfull boot with image-core-full-cmdline without removing vc4graphics. I have not had a succesfull boot with image-core-base and meta-openemvedded/meta-networking added as bblayer.

This seems to be related to other issues [issue #1079 ](https://github.com/agherzan/meta-raspberrypi/issues/1079) and therefore maybe being a problem with the 64-based machines.

**Steps to reproduce the issue:**
1. Build core-image-base for rpi5 and have succesfull boot
2. Run bitbake-layers add-layer /path/to/meta-openembedded/meta-networking (and therefore also add meta-python)
3.Boot now stops when loading vc4graphics:
[image]

**Describe the results you received:**
Not succesful boot

**Describe the results you expected:**
Succesful boot

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**
Using the scarthgap branch.
I have mostly tested in combination with kas from [meta-mender-community](https://github.com/theyoctojester/meta-mender-community)

davidlekve, 2024-11-01:
I have noticed that deleting the boot partition and then reflashing the sd-card (I use dd) often fixes this issue. I have still not made it work with core-image-full-cmdline

Woojin-Crive, 2024-11-25:
I am currently facing the same issue described here. I was wondering if there has been any progress or updates on this matter? Any information or workaround would be greatly appreciated.

trieu1162000, 2025-03-22:
Hi @davidlekve , did you use a HDMI screen? could you please share the steps to enable logging console to the screen.
I've modified in /boot/config.txt and /boot/cmdline.txt but wasn't successful though.

davidlekve, 2025-03-22:
Yes it was HDMI.

I do not remember completely how to set this up, but [source](https://www.raspberrypi.com/documentation/computers/config_txt.html#enable_uart) writes about:

`console=serial0,115200` in cmdline.txt

If I remember correctly that is all that is needed

davidlekve, 2025-03-22:
I can really recommend working with [Rugix](https://oss.silitics.com/rugix/) to set up a custom raspberry pi 5 (and other) image for people facing this issue @Woojin-Crive. It is an easy choice if yocto isnt a hard requirement.