agherzan/meta-raspberrypi #1386: wget not work in pi4 yocto image.how can I do?

gitgonewithwind, 2024-11-06:
wget -O Mambaforge.sh "https://github.com/conda-forge/miniforge/releases/latest/download/Miniforge3-$(uname)-$(uname -m).sh"
Connecting to github.com (140.82.114.3:443)
wget: note: TLS certificate validation not implemented
wget: TLS error from peer (alert code 80): 80
wget: error getting response: Connection reset by peer
--
I add --no-check-certificate  , but same issue.
I can ping www.google.ca 
I tried 
 IMAGE_INSTALL_append = " ca-certificates" in local.conf,
but same issue

any help?

matthiasklein, 2024-11-07:
Two years ago I also had problems with the certificate check with wget. But only on 32 bit ARM machines. It helped me to link against openssl instead of gnutls.

Since then I use the following bbappend ( `recipes-extended/wget/wget_%.bbappend` ) : 

```
# Currently the combination of wget and gnutls seems to be broken on 32 bit ARM devices.
# HTTPS downloads fail with the following error message:
#
# root@qemuarm:~# wget -4 https://speed.hetzner.de/100MB.bin
# --2022-02-04 07:46:37--  https://speed.hetzner.de/100MB.bin
# SSL_INIT
# Resolving speed.hetzner.de... 88.198.248.254
# Connecting to speed.hetzner.de|88.198.248.254|:443... connected.
# The certificate has not yet been activated
#
# Therefore, we are currently building wget against openSSL.
PACKAGECONFIG:remove = "gnutls"
PACKAGECONFIG:append = " openssl"
```
Here is the link to the discussion on the mailing list at the time: https://lists.yoctoproject.org/g/yocto/topic/88879516#msg56053

gitgonewithwind, 2024-11-07:
@matthiasklein  Thank you so much. let me try

gitgonewithwind, 2024-11-08:
@matthiasklein 
can not solve the issue. I use 64bit.
my meta-raspberrypi commit is (a few year ago commit)
e1014d41ef459af807f95cb7ff9105f077bc6daa