agherzan/meta-raspberrypi #1393: sacrthgap : pi zero v2 : licamera-apps build fails with rpicam_app.so.1.4.2 error

UmonDaisuke, 2024-12-09:
Hello,

I'm trying to add libcamera-apps to pi zero v2 64 bit  but i didn't succeed to find a solution for the problem mentioned below. It seems close from this one reported and solved  https://github.com/agherzan/meta-raspberrypi/pull/1273  

here is my setup:
ubuntu 22.04

**bblayers.conf**
BLAYERS ?= " \
  /home/dingo/scarthgap/poky/meta \
  /home/dingo/scarthgap/poky/meta-poky \
  /home/dingo/scarthgap/poky/meta-yocto-bsp \
  /home/dingo/scarthgap/poky/meta-raspberrypi \
  /home/dingo/scarthgap/poky/meta-openembedded/meta-oe \
  /home/dingo/scarthgap/poky/meta-openembedded/meta-python \
  /home/dingo/scarthgap/poky/meta-openembedded/meta-networking \
  /home/dingo/scarthgap/poky/meta-openembedded/meta-multimedia \
  "
**Local.conf**
MACHINE = "raspberrypi0-2w-64"
IMAGE_ROOTFS_EXTRA_SPACE = "10000"
EXTRA_IMAGE_FEATURES += " package-management ssh-server-dropbear debug-tweaks"
PACKAGE_CLASSES = "package_ipk"
IMAGE_INSTALL:append = " libcamera-apps"

**Here is the error message**
ERROR: libcamera-apps-1.4.2+git-r0 do_package: QA Issue: libcamera-apps: Files/directories were installed but not shipped in any package:
  /usr/lib/rpicam_app.so.1.4.2
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
libcamera-apps: 1 installed and not shipped files. [installed-vs-shipped]
ERROR: libcamera-apps-1.4.2+git-r0 do_package: Fatal QA errors were found, failing task.
ERROR: Logfile of failure stored in: /home/dingo/scarthgap/poky/build/tmp/work/cortexa53-poky-linux/libcamera-apps/1.4.2+git/temp/log.do_package.5878
ERROR: Task (/home/dingo/scarthgap/poky/meta-raspberrypi/dynamic-layers/multimedia-layer/recipes-multimedia/libcamera-apps/libcamera-apps_git.bb:do_package) failed with exit code '1'

**Attempts**
befor creating a bbappend, I tried several options by updating  libcamera-apps_git.bb but it fails everytime. Eithet i didn't compil  or libcamera-apps didn't work

d```
o_install:append() {
    rm -v ${D}/${bindir}/camera-bug-report
    rm -v ${D}${libdir}/rpicam_app.so.1.4.2
}

# not picked automatically, because it's missing common 'lib' prefix

FILES:${PN}-dev += "${libdir}/rpicam_app.so"
#FILES:${PN} += "${libdir}/rpicam_app.so.1.4.2"

#FILES_${PN} += "${libdir}/*"
#FILES_${PN}-dev = "${libdir}/* ${includedir}"

#FILES:${PN}-dev += "${libdir}/rpicam_app.so*"
#FILES:${PN} += "${libdir}/rpicam_app.so*"

```
Would you have ay idea how to solve this?
Many thanks for your support
Michael

OldManYellsAtCloud, 2024-12-11:
I would expect that `FILES:${PN} += "${libdir}/rpicam_app.so.1.4.2"` would allow it work, at least to create the package successfully. What do you mean by "libcamera-apps didn't work"? Do you get any errors?

UmonDaisuke, 2024-12-16:
Hello,
yes this new line enable compilation but when i run the following commznd, here is what i get
```
root@raspberrypi0-2w-64:~# libcamera-vid -o  test.h264                          
imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected unicam form
at: 1536x864-pBAA                                                              
munmap_chunk(): invalid pointer                                                
Aborted   
```           
do you have any idea of the solution?
kind regards.

OldManYellsAtCloud, 2024-12-16:
Unfortunately my knowledge stops at this point without taking a deep dive...
But I can confirm that see exactly the same behavior also using Rpi3-64 with IMX219, even if I downgrade libcamera-apps and libcamera, and even with kernel 5.15, and after playing with the packageconfig too.

OldManYellsAtCloud, 2024-12-17:
I couldn't help myself, and installed gdb, and made at least some partial progress. I found at least one error: [this](https://github.com/openembedded/meta-openembedded/blob/scarthgap/meta-multimedia/recipes-multimedia/libcamera/libcamera/0001-rpi-Use-alloca-instead-of-variable-length-arrays.patch) patch in libcamera is buggy, from meta-oe - this seems to be causing this crash.

You can try to remove this patch (`SRC_URI:remove:pn-libcamera = "file://0001-rpi-Use-alloca-instead-of-variable-length-arrays.patch"` in local.conf or similar), and try to build it. If it succeeds, then that's it (the patch only tries to add support for more compilers, otherwise doesn't change anything).

If it fails, then revert it, and try to apply the attached patch on libcamera from a bbappend. This patch tries to fix the bug from that other patch. However after this I faced another problem with displaying the video, that I think comes from my own image config, though I'm not sure. Any case, it might be different on your setup - if it isn't, then sorry, I think I'm really  running out of further immediate ideas.

[9999-Fix-incorrectly-sized-malloc.patch.txt](https://github.com/user-attachments/files/18161516/9999-Fix-incorrectly-sized-malloc.patch.txt)

UmonDaisuke, 2024-12-18:
thanks. i will try and let you know