agherzan/meta-raspberrypi #1394: Pi Compute Module 5, kernel panic.

gizahNL, 2024-12-12:
**Description**
The kernel panics on a CM5 

```
[    0.886917] Serial: 8250/16550 driver, 1 ports, IRQ sharing enabled
[    0.893526] SError Interrupt on CPU3, code 0x00000000be000011 -- SError
[    0.893529] CPU: 3 PID: 1 Comm: swapper/0 Not tainted 6.6.36-v8-16k #1
[    0.893531] Hardware name: Raspberry Pi Compute Module 5 Rev 1.0 (DT)
[    0.893532] pstate: 804000c9 (Nzcv daIF +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    0.893534] pc : bcm2712_pull_config_set+0x68/0xe8
[    0.893540] lr : bcm2712_pull_config_set+0x4c/0xe8
[    0.893541] sp : ffffc00080043ab0
[    0.893542] x29: ffffc00080043ab0 x28: ffffd000853cd068 x27: ffffd00085454008
[    0.893545] x26: ffffd00085300118 x25: ffff800016d58140 x24: 0000000000000000
[    0.893547] x23: 0000000000000018 x22: ffff800016aa7740 x21: 0000000000000122
[    0.893549] x20: ffff800016aa7680 x19: 0000000000000000 x18: ffffffffffffffff
[    0.893551] x17: 0000000000000000 x16: 0000000000000000 x15: ffff800016d58148
[    0.893553] x14: ffffc000800437f0 x13: ffff800016d5814a x12: 3030306330356437
[    0.893555] x11: 0000000000000040 x10: ffff8000165db6e0 x9 : ffffd000847900d4
[    0.893557] x8 : ffff80001803a938 x7 : 0000000000000000 x6 : 0000000000000000
[    0.893559] x5 : ffff80001803a910 x4 : 0000000000000024 x3 : 0000000000000018
[    0.893561] x2 : 0000000000000000 x1 : 0000000000000000 x0 : 0000000000000000
[    0.893563] Kernel panic - not syncing: Asynchronous SError Interrupt
[    0.893565] CPU: 3 PID: 1 Comm: swapper/0 Not tainted 6.6.36-v8-16k #1
[    0.893566] Hardware name: Raspberry Pi Compute Module 5 Rev 1.0 (DT)
[    0.893567] Call trace:
[    0.893568]  dump_backtrace+0x9c/0x100
[    0.893572]  show_stack+0x20/0x38
[    0.893573]  dump_stack_lvl+0x48/0x60
[    0.893577]  dump_stack+0x18/0x28
[    0.893579]  panic+0x318/0x378
[    0.893583]  nmi_panic+0x94/0xa0
[    0.893585]  arm64_serror_panic+0x78/0x98
[    0.893588]  arm64_is_fatal_ras_serror+0x44/0xc0
[    0.893589]  do_serror+0x64/0x80
[    0.893591]  el1h_64_error_handler+0x30/0x48
[    0.893594]  el1h_64_error+0x64/0x68
[    0.893596]  bcm2712_pull_config_set+0x68/0xe8
[    0.893597]  bcm2712_pinconf_set+0x6c/0xe0
[    0.893599]  pinconf_apply_setting+0xbc/0x138
[    0.893603]  pinctrl_commit_state+0xe8/0x220
[    0.893605]  pinctrl_select_state+0x24/0x40
[    0.893607]  pinctrl_bind_pins+0x144/0x160
[    0.893611]  really_probe+0x5c/0x2c0
[    0.893613]  __driver_probe_device+0x80/0x140
[    0.893615]  driver_probe_device+0x44/0x168
[    0.893617]  __driver_attach+0x98/0x1b0
[    0.893618]  bus_for_each_dev+0x84/0xf0
[    0.893620]  driver_attach+0x2c/0x40
[    0.893621]  bus_add_driver+0xec/0x218
[    0.893623]  driver_register+0x64/0x138
[    0.893625]  __platform_driver_register+0x30/0x48
[    0.893627]  brcmuart_init+0x44/0x88
[    0.893630]  do_one_initcall+0x60/0x2c8
[    0.893632]  kernel_init_freeable+0x1ec/0x3e8
[    0.893636]  kernel_init+0x2c/0x1f8
[    0.893639]  ret_from_fork+0x10/0x20
[    0.893642] SMP: stopping secondary CPUs
[    0.893645] Kernel Offset: 0x100004000000 from 0xffffc00080000000
[    0.893646] PHYS_OFFSET: 0x0
[    0.893646] CPU features: 0x0,00000001,70028143,0000720b
[    0.893648] Memory Limit: none
[    1.174851] ---[ end Kernel panic - not syncing: Asynchronous SError Interrupt ]---

```

**Steps to reproduce the issue:**
1. Build RPi5 machine config
2. Enable uart console (panics regardless)
config.txt:
[... 262 lines cut ...]
[    0.893599]  pinconf_apply_setting+0xbc/0x138
[    0.893603]  pinctrl_commit_state+0xe8/0x220
[    0.893605]  pinctrl_select_state+0x24/0x40
[    0.893607]  pinctrl_bind_pins+0x144/0x160
[    0.893611]  really_probe+0x5c/0x2c0
[    0.893613]  __driver_probe_device+0x80/0x140
[    0.893615]  driver_probe_device+0x44/0x168
[    0.893617]  __driver_attach+0x98/0x1b0
[    0.893618]  bus_for_each_dev+0x84/0xf0
[    0.893620]  driver_attach+0x2c/0x40
[    0.893621]  bus_add_driver+0xec/0x218
[    0.893623]  driver_register+0x64/0x138
[    0.893625]  __platform_driver_register+0x30/0x48
[    0.893627]  brcmuart_init+0x44/0x88
[    0.893630]  do_one_initcall+0x60/0x2c8
[    0.893632]  kernel_init_freeable+0x1ec/0x3e8
[    0.893636]  kernel_init+0x2c/0x1f8
[    0.893639]  ret_from_fork+0x10/0x20
[    0.893642] SMP: stopping secondary CPUs
[    0.893645] Kernel Offset: 0x100004000000 from 0xffffc00080000000
[    0.893646] PHYS_OFFSET: 0x0
[    0.893646] CPU features: 0x0,00000001,70028143,0000720b
[    0.893648] Memory Limit: none
[    1.174851] ---[ end Kernel panic - not syncing: Asynchronous SError Interrupt ]---
```

stu-spp, 2024-12-14:
Are you testing with the cm5 plugged into the IO Board?

It looks like the official Raspberry Pi OS images are shipping with:

```
x@xps:~/Downloads$ tar -tf bcm2712-official.tar.gz
home/x/Downloads/cm5-official-boot/bcm2712d0-rpi-5-b.dtb
home/x/Downloads/cm5-official-boot/bcm2712-rpi-5-b.dtb
home/x/Downloads/cm5-official-boot/bcm2712-rpi-cm5-cm4io.dtb
home/x/Downloads/cm5-official-boot/bcm2712-rpi-cm5-cm5io.dtb
home/x/Downloads/cm5-official-boot/bcm2712-rpi-cm5l-cm4io.dtb
home/x/Downloads/cm5-official-boot/bcm2712-rpi-cm5l-cm5io.dtb
home/x/Downloads/cm5-official-boot/initramfs_2712
home/x/Downloads/cm5-official-boot/kernel_2712.img
x@xps:~/Downloads$
```

[bcm2712-official.tar.gz](https://github.com/user-attachments/files/18133967/bcm2712-official.tar.gz)

I'll try to reproduce the panic you show, but while also using these extra sets of .dtb files.

stu-spp, 2024-12-15:
I'm not sure what's wrong. I can't even get console uart out on my builds. I tried to go with a newer kernel than your 6.6.36 (because I wanted to match the recent rpi firmware). Here are my patches:

```diff
diff --git a/conf/machine/include/rpi-base.inc b/conf/machine/include/rpi-base.inc
index 0b5909b..0aeb5d0 100644
--- a/conf/machine/include/rpi-base.inc
+++ b/conf/machine/include/rpi-base.inc
@@ -100,6 +100,7 @@ RPI_KERNEL_DEVICETREE ?= " \
     broadcom/bcm2711-rpi-cm4.dtb \
     broadcom/bcm2711-rpi-cm4s.dtb \
     broadcom/bcm2712-rpi-5-b.dtb \
+    broadcom/bcm2712-rpi-cm5-cm5io.dtb \
     "

 KERNEL_DEVICETREE ??= " \
diff --git a/conf/machine/raspberrypi5.conf b/conf/machine/raspberrypi5.conf
index 03f15c3..c649ebf 100644
--- a/conf/machine/raspberrypi5.conf
+++ b/conf/machine/raspberrypi5.conf
@@ -15,6 +15,7 @@ MACHINE_EXTRA_RRECOMMENDS += "\

 RPI_KERNEL_DEVICETREE = " \
     broadcom/bcm2712-rpi-5-b.dtb \
+    broadcom/bcm2712-rpi-cm5-cm5io.dtb \
 "

 SDIMG_KERNELIMAGE ?= "kernel_2712.img"
diff --git a/recipes-bsp/bootfiles/rpi-bootfiles.bb b/recipes-bsp/bootfiles/rpi-bootfiles.bb
index b04f24b..ffb9388 100644
--- a/recipes-bsp/bootfiles/rpi-bootfiles.bb
+++ b/recipes-bsp/bootfiles/rpi-bootfiles.bb
@@ -5,14 +5,14 @@ LIC_FILES_CHKSUM = "file://LICENCE.broadcom;md5=c403841ff2837657b2ed8e5bb474ac8d

 inherit deploy nopackages

-RPIFW_DATE ?= "20240319"
-SRCREV = "9f24f4bc2bdd07ffd158cfbb4bce88a2efc4c1f5"
+RPIFW_DATE ?= "20241210"
+SRCREV = "24247756bb3e962f0429b048cec5644e4a4253ab"
 SHORTREV = "${@d.getVar("SRCREV", False).__str__()[:7]}"
-RPIFW_SRC_URI ?= "https://api.github.com/repos/raspberrypi/firmware/tarball/9f24f4bc2bdd07ffd158cfbb4bce88a2efc4c1f5;downloadfilename=raspberrypi-firmware-${SHORTREV}.tar.gz"
+RPIFW_SRC_URI ?= "https://api.github.com/repos/raspberrypi/firmware/tarball/${SRCREV};downloadfilename=raspberrypi-firmware-${SHORTREV}.tar.gz"
 RPIFW_S ?= "${WORKDIR}/raspberrypi-firmware-${SHORTREV}"

 SRC_URI = "${RPIFW_SRC_URI}"
-SRC_URI[sha256sum] = "4b436f8946b139c6a1202375ef55d4848e3bcd8c1a9cb47000e06d7ecec828f7"
+SRC_URI[sha256sum] = "50eceb608a3b3fa6b1f7cea7cd9618a7b910c5c3f541fb9e7c23eadcbdaee706"

 PV = "${RPIFW_DATE}"

diff --git a/recipes-kernel/linux/linux-raspberrypi_6.6.bb b/recipes-kernel/linux/linux-raspberrypi_6.6.bb
index d12a78b..703c5ec 100644
--- a/recipes-kernel/linux/linux-raspberrypi_6.6.bb
+++ b/recipes-kernel/linux/linux-raspberrypi_6.6.bb
@@ -1,8 +1,8 @@
-LINUX_VERSION ?= "6.6.36"
+LINUX_VERSION ?= "6.6.64"
 LINUX_RPI_BRANCH ?= "rpi-6.6.y"
 LINUX_RPI_KMETA_BRANCH ?= "yocto-6.6"

-SRCREV_machine = "769634f344626ed73bcda14c91b567067974d7b2"
+SRCREV_machine = "80533a952218696c0ef1b346bab50dc401e6b74c"
 SRCREV_meta = "733366844f5e114221372929392bf237fc8d823c"

 KMETA = "kernel-meta"
```
config.txt:
```shell
x@xps:~/workspace/code/usbboot$ cat /mntc/config.txt  | grep uart
#init_uart_baud=115200
[... 56 lines cut ...]
-rwxr-xr-x  1 root root 2.6K Dec 14 23:00 config.txt
drwxr-xr-x 22 root root 4.0K Dec 14 19:57 ..
-rwxr-xr-x  1 root root  52K Apr  5  2011 bootcode.bin
-rwxr-xr-x  1 root root 3.2K Apr  5  2011 fixup4cd.dat
-rwxr-xr-x  1 root root 5.4K Apr  5  2011 fixup4.dat
-rwxr-xr-x  1 root root 8.3K Apr  5  2011 fixup4db.dat
-rwxr-xr-x  1 root root 8.3K Apr  5  2011 fixup4x.dat
-rwxr-xr-x  1 root root 3.2K Apr  5  2011 fixup_cd.dat
-rwxr-xr-x  1 root root 7.2K Apr  5  2011 fixup.dat
-rwxr-xr-x  1 root root  11K Apr  5  2011 fixup_db.dat
-rwxr-xr-x  1 root root  11K Apr  5  2011 fixup_x.dat
-rwxr-xr-x  1 root root  25M Apr  5  2011 kernel_2712.img
drwxr-xr-x  2 root root 8.0K Apr  5  2011 overlays
-rwxr-xr-x  1 root root    0 Apr  5  2011 rpi-bootfiles-20241210.stamp
-rwxr-xr-x  1 root root 794K Apr  5  2011 start4cd.elf
-rwxr-xr-x  1 root root 3.6M Apr  5  2011 start4db.elf
-rwxr-xr-x  1 root root 2.2M Apr  5  2011 start4.elf
-rwxr-xr-x  1 root root 2.9M Apr  5  2011 start4x.elf
-rwxr-xr-x  1 root root 794K Apr  5  2011 start_cd.elf
-rwxr-xr-x  1 root root 4.7M Apr  5  2011 start_db.elf
-rwxr-xr-x  1 root root 2.9M Apr  5  2011 start.elf
-rwxr-xr-x  1 root root 3.6M Apr  5  2011 start_x.elf
drwxr-xr-x  3 root root  16K Dec 31  1969 .
x@xps:~/workspace/code/usbboot$
```

gokamura, 2024-12-19:
I saw a similar kernel panic like `SError Interrupt on CPUx, code 0x00000000be000011 -- SError` here. Googling the code 0x00000000be000011 led me to this discussion:
https://hub.mender.io/t/asynchronous-serror-interrupt-rpi5-boot-issue/7275/12

After adding 
`RPI_KERNEL_DEVICETREE_OVERLAYS:append = " overlays/bcm2712d0.dtbo"` to my local.conf, the kernel panic has been fixed. Tested on the scarthgap branch.

gizahNL, 2024-12-20:
@gokamura that indeed did fix it for me! It's booting now.

stu-spp, 2024-12-20:
Yes, I think that was my problem as well. For posterity:

```shell
root@raspberrypi5:~# ls -laht /boot/overlays/
total 24K
drwxr-xr-x    2 root     root        4.0K Apr  5  2011 .
-rwxr-xr-x    1 root     root        1.5K Apr  5  2011 bcm2712d0.dtbo
drwxr-xr-x    3 root     root       16.0K Jan  1  1970 ..
root@raspberrypi5:~# uname -a
Linux raspberrypi5 6.6.64-v8-16k #1 SMP PREEMPT Thu Dec 12 19:45:21 UTC 2024 aarch64 GNU/Linux
root@raspberrypi5:~# ls -laht /boot/
total 46M
drwxr-xr-x   18 root     root        4.0K Dec 20 19:06 ..
-rwxr-xr-x    1 root     root         104 Dec 20 19:05 cmdline.txt
-rwxr-xr-x    1 root     root       79.3K Apr  5  2011 bcm2712-rpi-5-b.dtb
-rwxr-xr-x    1 root     root       79.6K Apr  5  2011 bcm2712-rpi-cm5-cm4io.dtb
-rwxr-xr-x    1 root     root       79.7K Apr  5  2011 bcm2712-rpi-cm5-cm5io.dtb
-rwxr-xr-x    1 root     root       51.2K Apr  5  2011 bootcode.bin
-rwxr-xr-x    1 root     root        2.7K Apr  5  2011 config.txt
-rwxr-xr-x    1 root     root        7.2K Apr  5  2011 fixup.dat
-rwxr-xr-x    1 root     root        5.3K Apr  5  2011 fixup4.dat
-rwxr-xr-x    1 root     root        3.2K Apr  5  2011 fixup4cd.dat
-rwxr-xr-x    1 root     root        8.3K Apr  5  2011 fixup4db.dat
-rwxr-xr-x    1 root     root        8.3K Apr  5  2011 fixup4x.dat
-rwxr-xr-x    1 root     root        3.2K Apr  5  2011 fixup_cd.dat
-rwxr-xr-x    1 root     root       10.1K Apr  5  2011 fixup_db.dat
-rwxr-xr-x    1 root     root       10.1K Apr  5  2011 fixup_x.dat
-rwxr-xr-x    1 root     root       24.1M Apr  5  2011 kernel_2712.img
drwxr-xr-x    2 root     root        4.0K Apr  5  2011 overlays
-rwxr-xr-x    1 root     root           0 Apr  5  2011 rpi-bootfiles-20241210.stamp
-rwxr-xr-x    1 root     root        2.8M Apr  5  2011 start.elf
-rwxr-xr-x    1 root     root        2.2M Apr  5  2011 start4.elf
-rwxr-xr-x    1 root     root      793.7K Apr  5  2011 start4cd.elf
-rwxr-xr-x    1 root     root        3.6M Apr  5  2011 start4db.elf
-rwxr-xr-x    1 root     root        2.9M Apr  5  2011 start4x.elf
-rwxr-xr-x    1 root     root      793.7K Apr  5  2011 start_cd.elf
-rwxr-xr-x    1 root     root        4.6M Apr  5  2011 start_db.elf
-rwxr-xr-x    1 root     root        3.6M Apr  5  2011 start_x.elf
drwxr-xr-x    3 root     root       16.0K Jan  1  1970 .
root@raspberrypi5:~# cat /boot/cmdline.txt
dwc_otg.lpm_enable=0 console=ttyAMA0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait  net.ifnames=0
root@raspberrypi5:~# cat /boot/config.txt | tail
 dtparam=i2c_arm=on
 dtparam=i2c_vc=on
 dtoverlay=uart0
 dtoverlay=uart0_console
 dtparam=uart0=on
 enable_rp1_uart=1
 pciex4_reset=0
 dtoverlay=disable-bt
 dtoverlay=dwc2,dr_mode=host

root@raspberrypi5:~# cat /etc/os-release
ID=poky
NAME="Poky (Yocto Project Reference Distro)"
VERSION="5.1.1 (styhead)"
VERSION_ID=5.1.1
VERSION_CODENAME="styhead"
PRETTY_NAME="Poky (Yocto Project Reference Distro) 5.1.1 (styhead)"
CPE_NAME="cpe:/o:openembedded:poky:5.1.1"
root@raspberrypi5:~#
```

stu-spp, 2024-12-30:
@gizahNL Do you have any insight into why u-boot is not working?

gizahNL, 2024-12-30:
> @gizahNL Do you have any insight into why u-boot is not working?

Spurious input on the debug uart (or at least that's my guestimate) Either connect a uart console on it, or set autoboot timeout to -2.

gizahNL, 2024-12-30:
https://lists.denx.de/pipermail/u-boot/2024-December/576043.html

closed 2025-02-12 by agherzan

bigalnz, 2025-04-05:
I am trying to fix this same issue with a SD card that doesnt boot once put into a new Rpi5 8GB. I have put the SD card in another booting Rpi, mounted it, but cant find local.conf file anywhere - can anyone help?

herrera-luis-alberto, 2025-08-04:
hi all,

I ran into a very similar issue when using buildroot on the Raspberry Pi 5 with the **D0** stepping of the bcm2712. In that case setting BR2_PACKAGE_RPI_FIRMWARE_INSTALL_DTB_OVERLAYS to `yes` solves the issue.

PR #1397 (2025-02-12): Fix boottime kernel panic on RPi CM5
Fixes #1394 

This adds the required devicetree and overlays to the raspberrypi5.conf file to allow booting the RPi CM5 without a kernel panic.

Thanks @gokamura for finding the fix.

U-Boot still doesn't work in my testing, waiting to get the uart headers to see if there is some output there.