agherzan/meta-raspberrypi #1403: Intramfs doesn't gets included (was trying to apply initramfs-module-overlayroot)

brian-geicp, 2025-01-13:
Was trying to apply `initramfs-module-overlayroot` to overlay a partition over the root layer and my understanding is that this require initramfs to be enabled.

So thus I was trying to figure out how to get initramfs image by following the instructions in <https://meta-raspberrypi.readthedocs.io/en/latest/extra-build-config.html#image-with-initramfs> which said to put into local.conf this configuration.

```bb
INITRAMFS_IMAGE = "core-image-testapp-initramfs"
INITRAMFS_IMAGE_BUNDLE = "1"
BOOT_SPACE = "1073741"
INITRAMFS_MAXSIZE = "315400"
IMAGE_FSTYPES:pn-${INITRAMFS_IMAGE} = "${INITRAMFS_FSTYPES}"
```

Where the image bb script is based on `./layers/poky/meta/recipes-core/images/core-image-minimal-initramfs.bb` but where `INITRAMFS_SCRIPTS` and `PACKAGE_INSTALL` is shrunk to:

```bb
INITRAMFS_SCRIPTS ?= "\
                      initramfs-framework-base \
                      initramfs-module-overlayroot \
                     "

PACKAGE_INSTALL = "${INITRAMFS_SCRIPTS}"
```

However while the intramfs cpio archive appears to be present `./build/tmp/deploy/images/test-app/core-image-test-app-initramfs-test-app.cpio.gz` it doesn't seem to be present in the bootup log.

For context, this is my layer stackup so far:

```bb
BBLAYERS ?= " \
  ${TOPDIR}/../layers/poky/meta \
  ${TOPDIR}/../layers/poky/meta-poky \
  ${TOPDIR}/../layers/poky/meta-yocto-bsp \
  ${TOPDIR}/../layers/meta-openembedded/meta-oe \
  ${TOPDIR}/../layers/meta-openembedded/meta-networking \
  ${TOPDIR}/../layers/meta-openembedded/meta-filesystems \
  ${TOPDIR}/../layers/meta-openembedded/meta-python \
  ${TOPDIR}/../layers/meta-virtualization \
  ${TOPDIR}/../layers/meta-raspberrypi \
  ${TOPDIR}/../layers/meta-rauc \
  ${TOPDIR}/../layers/meta-rauc-raspberrypi \
  ${TOPDIR}/../layers/meta-application-test-layer \
  "
```

I added some echo to both init and overlay bash script to see if the initramfs is running the script, but did not see any presence.
I'm still relatively newish to yocto, so I'm not sure where to look in the log to see if  `copy_initramfs()` in the `kernel.bbclass` is actually copying the initramfs image to `./usr` but I can definitely not find the image when I ssh into the device and inspected the `./usr` folder.

Maybe I'm missing something simple here, to which it might be a documentation issue. But in the meantime I hope this will still be helpful to track in case there is actually a bug in the bb script.

lagoon-come-strep, 2025-03-27:
I am also trying to make the same thing on kirkstone. I don't know if you find a solution.

I also tried to add in my local.conf:
```
INITRAMFS_KERNEL_IMAGE="${KERNEL_IMAGETYPE}-initramfs-${MACHINE}.bin"
IMAGE_BOOT_FILES:append = " ${INITRAMFS_KERNEL_IMAGE};${KERNEL_IMAGETYPE}"
```
but it doesn't seems to do anything.

In the documentation `Set this 3 kernel variables (in kernel’s do_configure:prepend in linux-raspberrypi.inc after the line kernel_configure_variable LOCALVERSION “””” )` is not really possible because  linux-raspberrypi.inc doesn't seems to have the given line.

I am also using u-boot with `RPI_USE_U_BOOT = "1"` but I don't know if it change anything because it shall be bundle.

lagoon-come-strep, 2025-04-01:
@dsosunov thank you for your reply.
First, I was able to add the initramfs cpio.gz archive through my local.conf by adding:
```
INITRAMFS_KERNEL_IMAGE="${INITRAMFS_IMAGE}-${MACHINE}"
IMAGE_BOOT_FILES:append = " ${INITRAMFS_KERNEL_IMAGE}.${INITRAMFS_FSTYPES};initramfs.cpio.gz"
```
but I think maybe your solution can also be fine, as my goal was just to get the overlayfs work. For now I wil just give the way I made all things works.

To give more context I use some particular layers:
meta-raspberrypi
meta-swupdate
meta-swupdate-boards
meta-readonly-rootfs-overlay  

The meta-swupdate-boards layer overload, through a .bbappend, the boot.cmd.in file used by meta-raspberrypi.
I fixed the problem here, but it can also found here:
https://github.com/agherzan/meta-raspberrypi/blob/master/recipes-bsp/rpi-u-boot-scr/files/boot.cmd.in

Here the content (actual master content, but I am on kirkstone branch actually):
```
fdt addr ${fdt_addr} && fdt get value bootargs /chosen bootargs
fatload @@BOOT_MEDIA@@ 0:1 ${kernel_addr_r} @@KERNEL_IMAGETYPE@@
if test ! -e @@BOOT_MEDIA@@ 0:1 uboot.env; then saveenv; fi;
@@KERNEL_BOOTCMD@@ ${kernel_addr_r} - ${fdt_addr}
```

We can see that the initramfs is not call anyway.
Here a raw fix (I do not test it) to ensure that initramfs is load & used:
```
fdt addr ${fdt_addr} && fdt get value bootargs /chosen bootargs
fatload @@BOOT_MEDIA@@ 0:1 ${kernel_addr_r} @@KERNEL_IMAGETYPE@@
fatload @@BOOT_MEDIA@@ 0:1 ${ramdisk_addr_r} initramfs.cpio.gz
if test ! -e @@BOOT_MEDIA@@ 0:1 uboot.env; then saveenv; fi;
@@KERNEL_BOOTCMD@@ ${kernel_addr_r} ${ramdisk_addr_r}:${filesize} ${fdt_addr}
```
(!) The name of the initramfs shall be initramfs.cpio.gz

I also need to fix meta-readonly-rootfs-overlay  to get all things work. with following fixes:
https://github.com/marcusfolkesson/meta-readonly-rootfs-overlay/pull/10
(!) The read only root feature shall not be enabled.

djboris9, 2025-10-26:
I had the same issue, but without using U-Boot. The solution from @dsosunov worked for me,