agherzan/meta-raspberrypi #1405: Core-image-base fails with 32 bit architecture and kernel 6.1 (No rule  for broadcom/bcm2708-rpi-zero.dtb)

UmonDaisuke, 2025-01-21:
> Hello,
> 
> please fin below my current status when i try to build  a core-image-base from scrthgap with different kernel and architecture for a pi zero v2:
> 64 bit + kernel 6.6 : ok
> 32 bit + kernel 6.6 : ok
> 64 bit + kernel  6.1: ok
> 32 bit + kernel 6.1: fail
> 
> the problem is that the last one is my target and i m not able to find any workaround to "No rule to make target 'arch/arm/boot/dts/broadcom/bcm2708-rpi-zero.dtb"
> 
> please find below log of the error  that avoid image creation.
> woul you have any  solution?
> Kind regards
> Michael
> 
> 
> ```
DEBUG: Executing shell function do_compile
> NOTE: KBUILD_BUILD_TIMESTAMP: Thu Feb  8 15:07:10 UTC 2024
> NOTE: make -j 16 CC=arm-poky-linux-gnueabi-gcc  -mno-thumb-interwork -marm -fuse-ld=bfd -fcanon-prefix-map  -fmacro-prefix-map=/home/dingo/poky/32/tmp/work-shared/raspberrypi0-2w/kernel-source=/usr/src/debug/linux-raspberrypi/6.1.77+git  -fdebug-prefix-map=/home/dingo/poky/32/tmp/work-shared/raspberrypi0-2w/kernel-source=/usr/src/debug/linux-raspberrypi/6.1.77+git  -fmacro-prefix-map=/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/linux-raspberrypi0_2w-standard-build=/usr/src/debug/linux-raspberrypi/6.1.77+git  -fdebug-prefix-map=/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/linux-raspberrypi0_2w-standard-build=/usr/src/debug/linux-raspberrypi/6.1.77+git  -fdebug-prefix-map=/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot=  -fmacro-prefix-map=/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot=  -fdebug-prefix-map=/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot-native=  -fdebug-prefix-map=/home/dingo/poky/32/tmp/work-shared/raspberrypi0-2w/kernel-source=/usr/src/kernel -fdebug-prefix-map=/home/dingo/poky/32/tmp/work-shared/raspberrypi0-2w/kernel-build-artifacts=/usr/src/kernel LD=arm-poky-linux-gnueabi-ld.bfd   OBJCOPY=arm-poky-linux-gnueabi-objcopy  STRIP=arm-poky-linux-gnueabi-strip HOSTCC=gcc  HOSTCFLAGS=-isystem/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot-native/usr/include -O2 -pipe HOSTLDFLAGS=-L/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot-native/usr/lib                         -L/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot-native/lib                         -Wl,-rpath,/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot-native/lib                         -Wl,-O1 -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/home/dingo/poky/32/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2 -pthread HOSTCPP=gcc  -E HOSTCXX=g++  HOSTCXXFLAGS=-isystem/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot-native/usr/include -O2 -pipe HOSTPKG_CONFIG=pkg-config-native PAHOLE=false -j 16 zImage LOADADDR=0x00008000
>   SYNC    include/config/auto.conf
>   GEN     Makefile
>   GEN     Makefile
>   CALL    /home/dingo/poky/32/tmp/work-shared/raspberrypi0-2w/kernel-source/scripts/checksyscalls.sh
>   Kernel: arch/arm/boot/Image is ready
>   Kernel: arch/arm/boot/zImage is ready
> NOTE: make -j 16 CC=arm-poky-linux-gnueabi-gcc  -mno-thumb-interwork -marm -fuse-ld=bfd -fcanon-prefix-map  -fmacro-prefix-map=/home/dingo/poky/32/tmp/work-shared/raspberrypi0-2w/kernel-source=/usr/src/debug/linux-raspberrypi/6.1.77+git  -fdebug-prefix-map=/home/dingo/poky/32/tmp/work-shared/raspberrypi0-2w/kernel-source=/usr/src/debug/linux-raspberrypi/6.1.77+git  -fmacro-prefix-map=/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/linux-raspberrypi0_2w-standard-build=/usr/src/debug/linux-raspberrypi/6.1.77+git  -fdebug-prefix-map=/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/linux-raspberrypi0_2w-standard-build=/usr/src/debug/linux-raspberrypi/6.1.77+git  -fdebug-prefix-map=/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot=  -fmacro-prefix-map=/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot=  -fdebug-prefix-map=/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot-native=  -fdebug-prefix-map=/home/dingo/poky/32/tmp/work-shared/raspberrypi0-2w/kernel-source=/usr/src/kernel -fdebug-prefix-map=/home/dingo/poky/32/tmp/work-shared/raspberrypi0-2w/kernel-build-artifacts=/usr/src/kernel LD=arm-poky-linux-gnueabi-ld.bfd   OBJCOPY=arm-poky-linux-gnueabi-objcopy  STRIP=arm-poky-linux-gnueabi-strip HOSTCC=gcc  HOSTCFLAGS=-isystem/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot-native/usr/include -O2 -pipe HOSTLDFLAGS=-L/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot-native/usr/lib                         -L/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot-native/lib                         -Wl,-rpath,/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot-native/lib                         -Wl,-O1 -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/home/dingo/poky/32/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2 -pthread HOSTCPP=gcc  -E HOSTCXX=g++  HOSTCXXFLAGS=-isystem/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot-native/usr/include -O2 -pipe HOSTPKG_CONFIG=pkg-config-native PAHOLE=false broadcom/bcm2708-rpi-zero.dtb CC=arm-poky-linux-gnueabi-gcc  -mno-thumb-interwork -marm -fuse-ld=bfd -fcanon-prefix-map  -fmacro-prefix-map=/home/dingo/poky/32/tmp/work-shared/raspberrypi0-2w/kernel-source=/usr/src/debug/linux-raspberrypi/6.1.77+git  -fdebug-prefix-map=/home/dingo/poky/32/tmp/work-shared/raspberrypi0-2w/kernel-source=/usr/src/debug/linux-raspberrypi/6.1.77+git  -fmacro-prefix-map=/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/linux-raspberrypi0_2w-standard-build=/usr/src/debug/linux-raspberrypi/6.1.77+git  -fdebug-prefix-map=/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/linux-raspberrypi0_2w-standard-build=/usr/src/debug/linux-raspberrypi/6.1.77+git  -fdebug-prefix-map=/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot=  -fmacro-prefix-map=/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot=  -fdebug-prefix-map=/home/dingo/poky/32/tmp/work/raspberrypi0_2w-poky-linux-gnueabi/linux-raspberrypi/6.1.77+git/recipe-sysroot-native=  -fdebug-prefix-map=/home/dingo/poky/32/tmp/work-shared/raspberrypi0-2w/kernel-source=/usr/src/kernel -fdebug-prefix-map=/home/dingo/poky/32/tmp/work-shared/raspberrypi0-2w/kernel-build-artifacts=/usr/src/kernel   LD=arm-poky-linux-gnueabi-ld.bfd   OBJCOPY=arm-poky-linux-gnueabi-objcopy  STRIP=arm-poky-linux-gnueabi-strip LOADADDR=0x00008000
>   SYNC    include/config/auto.conf
>   GEN     Makefile
> make[1]: *** No rule to make target 'arch/arm/boot/dts/broadcom/bcm2708-rpi-zero.dtb'.  Stop.
> make: *** [/home/dingo/poky/32/tmp/work-shared/raspberrypi0-2w/kernel-source/Makefile:1471: broadcom/bcm2708-rpi-zero.dtb] Error 2
> ERROR: oe_runmake failed
> WARNING: exit code 1 from a shell command. 
> 
> 
>  
```

 _Originally posted by @UmonDaisuke in [#1365](https://github.com/agherzan/meta-raspberrypi/issues/1365#issuecomment-2601855470)_

UmonDaisuke, 2025-01-21:
i went further into analysis for kernel 6.1 and architecture 32 bits,
```
MACHINE = "raspberrypi0-2w"
PREFERRED_VERSION_linux-raspberrypi = "6.1.%"
```
in fact after removing the bcm2708-rpi-zero.dtb from device tree, i appears that the problem is the next soc and so on.
`KERNEL_DEVICETREE:remove = "broadcom/bcm2708-rpi-zero.dtb ...."`

so this a global rule that is missing for all broadcom soc. 

any idea how it can be solved?

OldManYellsAtCloud, 2025-01-22:
I find it pretty strange that you managed to compile 64bit version for 6.1...

Anyway... I believe the reason for the failure is the changed devicetree folder structure between 6.1 and 6.6. The `RPI_KERNEL_DEVICETREE` variable, that holds the paths, have a bunch of path prefixed with "broadcom" folder: `broadcom/bcm2708-rpi-zero.dtb`, `broadcom/bcm2708-rpi-zero-w.dtb`, etc. But this folder didn't exist in 6.1 yet. For this version, the paths should be just `bcm2708-rpi-zero.dtb`, `bcm2708-rpi-zero-w.dtb`, etc...

UmonDaisuke, 2025-01-22:
many thanks it works thanks to this update. 64 bit version and  6.1 was just for investigation.

closed 2025-01-22 by UmonDaisuke